RROrg/rr #219: 蜗牛星际ABC款

wjz304, 2023-09-16:
就像我是你的数学老师，你说你高考考了0分，问我为啥，

我只能说：我也想知道为啥，
我又看不到试卷，我也不知道你咋写的，我怎么知道？

你们自己讨论吧.

huang7920, 2023-09-23:
大佬帮忙看看蜗牛星际C款找不到IP。
日志最后部分是这个
[image]

alexallen1, 2023-09-24:
没必要阴阳怪气吧，最初几批玩黑群晖都是J1900 蜗牛星际开始。

wjz304, 2023-09-24:
> 没必要阴阳怪气吧，最初几批玩黑群晖都是J1900 蜗牛星际开始。

  你自己说关联的这几个 isuess，安装信息没有，截图日志没有，你能解决？
  我只是类比，希望你们能明白 我看到那些issue的一种状况，
  之所以我不会明说找你要日志截图，因为你提供了日志截图，问题也不一定能解决，到时又怪我浪费你时间。

  我只是菜鸟，我解不了你们的问题，建议去找 @pocopico @jumkey @fbelavenuto @jim3ma @AuxXxilium @PeterSuh-Q3 @RedPill-TTG

PeterSuh-Q3, 2023-09-24:
Is this related to CAI Coin?

[image]

[image]

closed 2024-03-30 by wjz304