RROrg/rr #221: 安装DSM时无硬盘，找不到 virio scsi 硬盘

r0nn, 2023-09-18:
Install ENV:  
 * motherboard:  KVM
 * CPU:  KVM
 * NIC:  KVM intel e1000

Arpl version:  23.9.1

DSM model:  DS3622xs+
DSM version:  7.2

Issue:  
配置为scsi控制器，virtio硬盘，安装配置引导时可以认到硬盘 /dev/vda，但是引导安装DSM的时候检测不到硬盘，配置的时候也选择了virtio, virtio-scsi模块
因为虚拟平台原因无法更改硬盘配置，请问是否有办法支持？谢谢

logs:  
[image]

wjz304, 2023-09-18:
没有

wjz304, 2023-09-18:
不过我到时挺好奇， 有兴趣可以在 安装界面 telnet 进去 取这些log
```
ls -d /dev/sd*
lspci -d ::106
lspci -d ::107
ls -l /sys/class/scsi_host
cat /etc/synoinfo.conf | grep portcfg
cat /etc/synoinfo.conf | grep disks
```

r0nn, 2023-09-18:
安装界面 telnet log 如下：

SynologyNAS> ls -d /dev/sd*
ls: /dev/sd*: No such file or directory
SynologyNAS> lspci -d ::106
-ash: lspci: not found
SynologyNAS> lspci -d ::107
-ash: lspci: not found
SynologyNAS> ls -l /sys/class/scsi_host
lrwxrwxrwx    1 root     root             0 Aug 27 08:28 host0 -> ../../devices/pci0000:00/0000:00:18.7/usb1/1-6/1-6:1.0/host0/scsi_host/host0
SynologyNAS> cat /etc/synoinfo.conf | grep portcfg
esataportcfg="0x00"
internalportcfg="0x1"
usbportcfg="0xf00000"
SynologyNAS> cat /etc/synoinfo.conf | grep disks
maxdisks="26"

wjz304, 2023-09-18:
行吧，忘了不是sd 

Ing
***@***.***

 
安装界面 telnet log 如下：
 
SynologyNAS&gt; ls -d /dev/sd*

 SynologyNAS&gt; lspci -d ::106
 -ash: lspci: not found
 SynologyNAS&gt; lspci -d ::107
 -ash: lspci: not found
 SynologyNAS&gt; ls -l /sys/class/scsi_host
 lrwxrwxrwx    1 root     root             0 Aug 27 08:28 host0 -&gt; ../../devices/pci0000:00/0000:00:18.7/usb1/1-6/1-6:1.0/host0/scsi_host/host0
 SynologyNAS&gt; cat /etc/synoinfo.conf | grep portcfg
 SynologyNAS&gt; cat /etc/synoinfo.conf | grep disks
 maxdisks="26"

r0nn, 2023-09-18:
安装界面 telnet log，没有 lspci:
```
SynologyNAS> ls -d /dev/sd*
ls: /dev/sd*: No such file or directory
SynologyNAS> lspci -d ::106
-ash: lspci: not found
SynologyNAS> ls -l /sys/class/scsi_host/
lrwxrwxrwx    1 root     root             0 Aug 27 09:22 host0 -> ../../devices/pci0000:00/0000:00:18.7/usb1/1-6/1-6:1.0/host0/scsi_host/host0
SynologyNAS> cat /etc/synoinfo.conf | grep portcfg
esataportcfg="0x00"
internalportcfg="0x1"
usbportcfg="0xf00000"
SynologyNAS> cat /etc/synoinfo.conf | grep disks
maxdisks="26"
SynologyNAS> lsmod
Module                  Size  Used by    Tainted: P
usbhid                 26542  0
hid                    91650  1 usbhid
broadwellnk_synobios    94924  0
adt7475                31272  0
nfsv3                  26042  0
nfs                   184216  1 nfsv3
lockd                  79573  2 nfsv3,nfs
grace                   2178  1 lockd
sunrpc                248805  3 nfsv3,nfs,lockd
r8168                 554602  0
i2c_algo_bit            5812  0
dca                     4896  0
vxlan                  33429  0
ip6_udp_tunnel          1967  1 vxlan
udp_tunnel              2419  1 vxlan
vfat                   10511  0
fat                    55795  1 vfat
sg                     26471  0
sha256_generic         10665  0
synorbd                81311  0
synofsbd               29052  0
uas                    13557  0
usb_storage            52967  1 uas
virtio_pci             13902  0
e1000                 100259  0
virtio                  5442  1 virtio_pci
virtio_ring             6903  1 virtio_pci
processor              26502  0
xhci_pci                4820  0
xhci_hcd              110302  1 xhci_pci
ehci_pci                4191  0
ehci_hcd               43205  1 ehci_pci
usbcore               208618  7 usbhid,uas,usb_storage,xhci_pci,xhci_hcd,ehci_pci,ehci_hcd
usb_common              2286  1 usbcore
mv14xx                529314  0
SynologyNAS>
```

loader配置时一些 log:
```
root@arpl:/opt/arpl# lspci
00:00.0 Host bridge: Intel Corporation 440FX - 82441FX PMC [Natoma] (rev 02)
00:01.0 ISA bridge: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II]
00:01.1 IDE interface: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II]
00:01.2 USB controller: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] (rev 01)
00:01.3 Bridge: Intel Corporation 82371AB/EB/MB PIIX4 ACPI (rev 03)
00:02.0 VGA compatible controller: Cirrus Logic GD 5446
00:03.0 Communication controller: Red Hat, Inc. Virtio console
00:0b.0 SCSI storage controller: Red Hat, Inc. Virtio block device
00:12.0 Ethernet controller: Intel Corporation 82540EM Gigabit Ethernet Controller (rev 03)
00:18.0 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #1 (rev 03)
00:18.1 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #2 (rev 03)
00:18.2 USB controller: Intel Corporation 82801I (ICH9 Family) USB UHCI Controller #3 (rev 03)
00:18.7 USB controller: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 (rev 03)
00:19.0 USB controller: NEC Corporation uPD720200 USB 3.0 Host Controller (rev 03)
root@arpl:/opt/arpl# lsmod
Module                  Size  Used by    Not tainted
8021q                  32768  0
ipv6                  544768 42 [permanent]
virtio_blk             28672  0
virtio_console         40960  0
virtio_pci             24576  0
e1000                 131072  0
virtio_pci_modern_dev    20480  1 virtio_pci
virtio                 16384  3 virtio_blk,virtio_console,virtio_pci
virtio_ring            32768  3 virtio_blk,virtio_console,virtio_pci
root@arpl:/opt/arpl# ls -l /sys/class/scsi_host
total 0
lrwxrwxrwx 1 root root 0 Sep 18 04:55 host0 -> ../../devices/pci0000:00/0000:00:18.7/usb1/1-6/1-6:1.0/host0/scsi_host/host0
root@arpl:/opt/arpl# ls -l /dev/disk/by-id
total 0
lrwxrwxrwx 1 root root  9 Sep 18 04:55 usb-SanDisk_Ultra_Fit_4C530001251111115043-0:0 -> ../../sda
lrwxrwxrwx 1 root root 10 Sep 18 04:59 usb-SanDisk_Ultra_Fit_4C530001251111115043-0:0-part1 -> ../../sda1
lrwxrwxrwx 1 root root 10 Sep 18 04:59 usb-SanDisk_Ultra_Fit_4C530001251111115043-0:0-part2 -> ../../sda2
lrwxrwxrwx 1 root root 10 Sep 18 04:59 usb-SanDisk_Ultra_Fit_4C530001251111115043-0:0-part3 -> ../../sda3
lrwxrwxrwx 1 root root  9 Sep 18 04:55 virtio-6135270269397-1 -> ../../vda
```

另外有一个情况是移除virtio磁盘的时候引导很快，但是加了virtio磁盘后引导很慢，需要等待好几分钟
[image]
[image]

wjz304, 2023-09-18:
arpl 下 lspci -vvvnn
 

Ing
***@***.***

 

 SynologyNAS&gt; ls -d /dev/sd* ls: /dev/sd*: No such file or directory SynologyNAS&gt; lspci -d ::106 -ash: lspci: not found SynologyNAS&gt; ls -l /sys/class/scsi_host/ lrwxrwxrwx    1 root     root             0 Aug 27 09:22 host0 -&gt; ../../devices/pci0000:00/0000:00:18.7/usb1/1-6/1-6:1.0/host0/scsi_host/host0 SynologyNAS&gt; cat /etc/synoinfo.conf | grep portcfg esataportcfg="0x00" internalportcfg="0x1" usbportcfg="0xf00000" SynologyNAS&gt; cat /etc/synoinfo.conf | grep disks maxdisks="26" SynologyNAS&gt; lsmod Module                  Size  Used by    Tainted: P usbhid                 26542  0 hid                    91650  1 usbhid broadwellnk_synobios    94924  0 adt7475                31272  0 nfsv3                  26042  0 nfs                   184216  1 nfsv3 lockd                  79573  2 nfsv3,nfs grace                   2178  1 lockd sunrpc                248805  3 nfsv3,nfs,lockd r8168                 554602  0 i2c_algo_bit            5812  0 dca                     4896  0 vxlan                  33429  0 ip6_udp_tunnel          1967  1 vxlan udp_tunnel              2419  1 vxlan vfat                   10511  0 fat                    55795  1 vfat sg                     26471  0 sha256_generic         10665  0 synorbd                81311  0 synofsbd               29052  0 uas                    13557  0 usb_storage            52967  1 uas virtio_pci             13902  0 e1000                 100259  0 virtio                  5442  1 virtio_pci virtio_ring             6903  1 virtio_pci processor              26502  0 xhci_pci                4820  0 xhci_hcd              110302  1 xhci_pci ehci_pci                4191  0 ehci_hcd               43205  1 ehci_pci usbcore               208618  7 usbhid,uas,usb_storage,xhci_pci,xhci_hcd,ehci_pci,ehci_hcd usb_common              2286  1 usbcore mv14xx                529314  0 SynologyNAS&gt;  
loader配置时一些 log:

wjz304, 2023-09-18:
lspci -vvvmm
 

Ing
***@***.***

r0nn, 2023-09-18:
lspci -vvvmm
```
Slot:   00:00.0
Class:  Host bridge
Vendor: Intel Corporation
Device: 440FX - 82441FX PMC [Natoma]
SVendor:        Red Hat, Inc.
SDevice:        Qemu virtual machine
Rev:    02
ProgIf: 00

Slot:   00:01.0
Class:  ISA bridge
Vendor: Intel Corporation
Device: 82371SB PIIX3 ISA [Natoma/Triton II]
SVendor:        Red Hat, Inc.
SDevice:        Qemu virtual machine
ProgIf: 00

Slot:   00:01.1
Class:  IDE interface
Vendor: Intel Corporation
Device: 82371SB PIIX3 IDE [Natoma/Triton II]
SVendor:        Red Hat, Inc.
SDevice:        Qemu virtual machine
ProgIf: 80

Slot:   00:01.2
Class:  USB controller
Vendor: Intel Corporation
Device: 82371SB PIIX3 USB [Natoma/Triton II]
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    01
ProgIf: 00

Slot:   00:01.3
Class:  Bridge
Vendor: Intel Corporation
Device: 82371AB/EB/MB PIIX4 ACPI
SVendor:        Red Hat, Inc.
SDevice:        Qemu virtual machine
Rev:    03
ProgIf: 00

Slot:   00:02.0
Class:  VGA compatible controller
Vendor: Cirrus Logic
Device: GD 5446
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
ProgIf: 00

Slot:   00:03.0
Class:  Communication controller
Vendor: Red Hat, Inc.
Device: Virtio console
SVendor:        Red Hat, Inc.
SDevice:        Virtio console
PhySlot:        3
ProgIf: 00

Slot:   00:0b.0
Class:  SCSI storage controller
Vendor: Red Hat, Inc.
Device: Virtio block device
SVendor:        Red Hat, Inc.
SDevice:        Virtio block device
PhySlot:        11
ProgIf: 00

Slot:   00:12.0
Class:  Ethernet controller
Vendor: Intel Corporation
Device: 82540EM Gigabit Ethernet Controller
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
PhySlot:        18
Rev:    03
ProgIf: 00

Slot:   00:18.0
Class:  USB controller
Vendor: Intel Corporation
Device: 82801I (ICH9 Family) USB UHCI Controller #1
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    03
ProgIf: 00

Slot:   00:18.1
Class:  USB controller
Vendor: Intel Corporation
Device: 82801I (ICH9 Family) USB UHCI Controller #2
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    03
ProgIf: 00

Slot:   00:18.2
Class:  USB controller
Vendor: Intel Corporation
Device: 82801I (ICH9 Family) USB UHCI Controller #3
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    03
ProgIf: 00

Slot:   00:18.7
Class:  USB controller
Vendor: Intel Corporation
Device: 82801I (ICH9 Family) USB2 EHCI Controller #1
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    03
ProgIf: 20

Slot:   00:19.0
Class:  USB controller
Vendor: NEC Corporation
Device: uPD720200 USB 3.0 Host Controller
SVendor:        Red Hat, Inc.
SDevice:        QEMU Virtual Machine
Rev:    03
ProgIf: 30
```

r0nn, 2023-09-18:
lspci -vvvnn
```
00:00.0 Host bridge [0600]: Intel Corporation 440FX - 82441FX PMC [Natoma] [8086:1237] (rev 02)
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0

00:01.0 ISA bridge [0601]: Intel Corporation 82371SB PIIX3 ISA [Natoma/Triton II] [8086:7000]
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-

00:01.1 IDE interface [0101]: Intel Corporation 82371SB PIIX3 IDE [Natoma/Triton II] [8086:7010] (prog-if 80 [ISA Compatibility mode-only controller, supports bus mastering])
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Region 0: I/O ports at 01f0 [size=8]
        Region 1: I/O ports at 03f4
        Region 2: I/O ports at 0170 [size=8]
        Region 3: I/O ports at 0374
        Region 4: I/O ports at c180 [size=16]

00:01.2 USB controller [0c03]: Intel Corporation 82371SB PIIX3 USB [Natoma/Triton II] [8086:7020] (rev 01) (prog-if 00 [UHCI])
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin D routed to IRQ 11
        Region 4: I/O ports at c100 [size=32]
        Kernel driver in use: uhci_hcd

00:01.3 Bridge [0680]: Intel Corporation 82371AB/EB/MB PIIX4 ACPI [8086:7113] (rev 03)
        Subsystem: Red Hat, Inc. Qemu virtual machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B+ ParErr- DEVSEL=medium >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Interrupt: pin A routed to IRQ 9

00:02.0 VGA compatible controller [0300]: Cirrus Logic GD 5446 [1013:00b8] (prog-if 00 [VGA controller])
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster- SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Region 0: Memory at fc000000 (32-bit, prefetchable) [size=32M]
        Region 1: Memory at febf4000 (32-bit, non-prefetchable) [size=4K]
        Expansion ROM at 000c0000 [disabled] [size=128K]

00:03.0 Communication controller [0780]: Red Hat, Inc. Virtio console [1af4:1003]
        Subsystem: Red Hat, Inc. Virtio console [1af4:0003]
        Physical Slot: 3
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
        Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0
        Interrupt: pin A routed to IRQ 11
        Region 0: I/O ports at c080 [size=64]
        Region 1: Memory at febf5000 (32-bit, non-prefetchable) [size=4K]
        Region 4: Memory at fe000000 (64-bit, prefetchable) [size=16K]
        Capabilities: [98] MSI-X: Enable+ Count=2 Masked-
                Vector table: BAR=1 offset=00000000
                PBA: BAR=1 offset=00000800
        Capabilities: [84] Vendor Specific Information: VirtIO: <unknown>
                BAR=0 offset=00000000 size=00000000
        Capabilities: [70] Vendor Specific Information: VirtIO: Notify
                BAR=4 offset=00003000 size=00001000 multiplier=00000004
        Capabilities: [60] Vendor Specific Information: VirtIO: DeviceCfg
                BAR=4 offset=00002000 size=00001000
        Capabilities: [50] Vendor Specific Information: VirtIO: ISR
                BAR=4 offset=00001000 size=00001000
        Capabilities: [40] Vendor Specific Information: VirtIO: CommonCfg
                BAR=4 offset=00000000 size=00001000
        Kernel driver in use: virtio-pci
[... 65 lines cut ...]
        Kernel driver in use: uhci_hcd

00:18.7 USB controller [0c03]: Intel Corporation 82801I (ICH9 Family) USB2 EHCI Controller #1 [8086:293a] (rev 03) (prog-if 20 [EHCI])
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx-
        Status: Cap- 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin D routed to IRQ 11
        Region 0: Memory at febf7000 (32-bit, non-prefetchable) [size=4K]
        Kernel driver in use: ehci-pci

00:19.0 USB controller [0c03]: NEC Corporation uPD720200 USB 3.0 Host Controller [1033:0194] (rev 03) (prog-if 30 [XHCI])
        Subsystem: Red Hat, Inc. QEMU Virtual Machine [1af4:1100]
        Control: I/O+ Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR+ FastB2B- DisINTx+
[... cut ...] SERR- <PERR- INTx-
        Latency: 0, Cache Line Size: 64 bytes
        Interrupt: pin A routed to IRQ 10
        Region 0: Memory at febf0000 (64-bit, non-prefetchable) [size=16K]
        Capabilities: [90] MSI-X: Enable+ Count=16 Masked-
                Vector table: BAR=0 offset=00003000
                PBA: BAR=0 offset=00003800
        Capabilities: [70] MSI: Enable- Count=1/16 Maskable- 64bit+
                Address: 0000000000000000  Data: 0000
        Kernel driver in use: xhci_hcd
```

wjz304, 2023-09-18:
Kernel modules: virtio_pci

wjz304, 2023-10-10:
解不了
还是使用 virtio-scsi-pci 类型把

closed 2023-10-10 by wjz304