RROrg/rr #227: 请问为什么DS3617xs型号没有了？

gexiaopeng, 2023-09-20:
我的蜗牛星际A 目前安装的是DS3617xs -6.1.7 ，运行了3年一直很稳定，现在用arpl-i18n发现没有 DS3617xs了，安装了其他型号比如DS918+,DS3622 ,最后启动以后都找不到IP。

wjz304, 2023-09-20:
有
[image]

closed 2023-09-20 by wjz304

gexiaopeng, 2023-09-20:
难道我下载的镜像文件不对？ 我是这里下载的 https://github.com/wjz304/arpl-i18n/releases

[image]

[image]

gexiaopeng, 2023-09-20:
> 有 [image]

请问为什么我选择型号的界面不一样？