RROrg/rr #230: 群晖7.2u3版本无法识别nvme硬盘(非DT机型)，有什么解决方案

PWMFI, 2023-09-21:


wjz304, 2023-09-21:
Please fill in the following information.

            Install ENV:  
            * motherboard:  
            * CPU:  
            * NIC:  

            Arpl version:  
            * arpl:  
            * addons:  
            * modules:  
            * lkms:  

            DSM: 
            * model:  
            * version:  

            Issue:  

            logs:  

            (Plz review the content of #173, #175, #226 first)  
            ...

PWMFI, 2023-09-25:
Please fill in the following information.

        Install ENV:  
        * motherboard:  dell poweredge 
[image]
[image]
R730
        * CPU:  E5-2698
        * NIC:  i350

        Arpl version:  
        * arpl:  23.9.5
        * addons:  NO
        * modules:  NO
        * lkms:  NO

        DSM: 
        * model:  RS3621
        * version:  7.2

        Issue:  Non-DT models cannot display the drive in the Storage Manager

        logs:  

        (Plz review the content of #173, #175, #226 first)  
        ...

wjz304, 2023-09-25:
DS1621xs+ 支持 2个m.2

PWMFI, 2023-09-25:
> DS1621xs+ 支持 2个m.2

所有型号都试过了，只有带dt型号的机型才能识别nvme，其他所有型号都不行，但是有阵列卡不能用DT机型，esxi安装的群晖，阵列卡和nvme都是直通的

PWMFI, 2023-09-25:
nvme是pcie拆分卡上的

wjz304, 2023-09-25:
ls /sys/block/nvme*

AuxXxilium, 2023-09-25:
ESXi later than 7.0U3g (including 8.0) does not support NVMe Mapping in DSM. PCIe Paths are changed in later versions and not compatible to DSM.

PWMFI, 2023-09-26:
> 高于 7.0U3g（包括 8.0）的 ESXi 不支持 DSM 中的 NVMe 映射。PCIe 路径在更高版本中会更改，并且与 DSM 不兼容。

这个有什么解决方法，虚拟机兼容性能改成6.7来解决这个问题吗

AuxXxilium, 2023-09-26:
no it won't help. only possible option: go back to 7.0u3g. that is why i use a 7.0u3g vm inside my esxi 8.0

FelixLAI85, 2023-10-03:
我也有同样的NVME硬盘不识别的问题，我目前的平台是EXSi8.0U2, DS918+7.2.1-69057，我之前使用ESXi8.0U1c, DS918+7.2版本的时候是可以直通NVME，然后通过修改文件让虚拟机的黑群晖识别nvme作为缓存的。我是根据GXNAS博客的教程操作的，我在他的post下面的comments里找到两个和我想通的情况：
https://wp.gxnas.com/10930.html/comment-page-7#comments
根据我的经历，我觉得不像是ESXi8.0版本的问题，更像是目前的引导版本的问题，有没有可能是之前的某个引导版本添加了启用nvme cache和存储那两个插件导致的pci识别问题？
在目前的引导版本下通过ssh查询nvme的位置会得到类似如下的信息：
P: /devices/pci0000:02/0000:02:04.0/nvme/nvme0/nvme0n1 显示pci0000:02 
但是在我之前可以成功识别nvme的时候，一般应该是pci0000:00/0000:00，是00而不是02.
我猜测是不是pci信息识别方面有一些问题，有没有可能在后续版本修复？

AuxXxilium, 2023-10-03:
it's esxi related, i had multiple dsm installations and all failed to load nvme after update to 8.0. nothing else were changed. tried it multiple times with clean install 7.0u3g and upgrade to 8.0. everytime the same issue. opened a ticket to vmware and they wrote that they changed a layer in esxi to get more performance. so can simple find this with lspci. the new paths are shorter than the old, they miss a .xx. in the middle.

FelixLAI85, 2023-10-03:
> it's esxi related, i had multiple dsm installations and all failed to load nvme after update to 8.0. nothing else were changed. tried it multiple times with clean install 7.0u3g and upgrade to 8.0. everytime the same issue. opened a ticket to vmware and they wrote that they changed a layer in esxi to get more performance. so can simple find this with lspci. the new paths are shorter than the old, they miss a section in the middle.

OK, then I think this should be ESXi's problem. Did they mention any fix can be applied to solve this issue? Thanks.

wjz304, 2023-10-12:
Environmental issues, No further response, this issue is closed

closed 2023-10-12 by wjz304

mm350, 2023-11-19:
> 我也有同样的NVME硬盘无法识别的问题，我目前的平台是EXSi8.0U2, DS918+7.2.1-69057，我之前使用ESXi8.0U1c, DS918+7.2版本的时候是可以直通NVME的，然后通过修改文件让虚拟机的黑群晖识别nvme作为缓存的。我是根据GXNAS博客的教程操作的，我在他的帖子下面的评论里找到了两个和我类似的情况：https: [//wp.gxnas.com /10930.html/comment-page-7#comments](https://wp.gxnas.com/10930.html/comment-page-7#comments) 根据我的经验，我觉得不是ESXi8.0版本的问题，此前是目前的引导版本的问题，有可能是之前添加的某个引导版本了启用nvme缓存和存储这两个插件导致的pci识别问题？ 在目前的引导版本下通过ssh查询nvme的位置会得到类似如下的信息： P: /devices/pci0000:02/0000:02:04.0/ nvme/nvme0/nvme0n1显示pci0000:02 但是在我可以成功识别nvme的时候，一般应该是pci0000:00/0000:00，是00而不是02。 我猜测是不是pci信息识别方面有一些问题，有可能没有后续版本修复？

我也是EXSi8.0U2, DS918+7.2.1-69057，同樣情況。因爲是直通硬盤，換到U盤引導，是能識別的，所以還是EXSi的問題

Sevenzhi, 2023-11-27:
> ，
请问解决了嘛

Sevenzhi, 2023-11-27:
> > > ，
> > > 请问解决了嘛
> 
> esxi新版本问题，换7.0u3g就可以了，不是引导的事

就是引导的事情，这两天翻遍了Google，偶然看到了一个评论解决了，esxi8.0 Update 2使用ds923的黑裙就好了，使用918就是不行，我刚刚安装了923的黑裙后直接自动识别了