RROrg/rr #231: 请求更新8125驱动~ 官网看到最新驱动9.011.01~

captcao, 2023-09-22:
直通8125bg网卡后，大流量就会断流~

请求作者更新8125最新驱动~

captcao, 2023-09-22:
nstall ENV:

motherboard: 华擎 z370m pro4
CPU: i5-8600k
NIC: 8125bg
Arpl version:[arpl-i18n]2394

arpl:[arpl-i18n]2394
addons:默认
modules:全
lkms:
DSM:7.2

model:918
version:7.2
Issue:直通8125bg网卡后，iperf3打流过程会断流~插拔网线数次后会恢复~

logs:

captcao, 2023-09-22:
2023-09-22T09:45:44+08:00 DS_Matt_8600k if_link_down_hook_event[12339]: eth0
2023-09-22T09:45:50+08:00 DS_Matt_8600k if_link_up_hook_event[13078]: eth0
2023-09-22T09:46:16+08:00 DS_Matt_8600k if_link_down_hook_event[14402]: eth0
2023-09-22T09:46:26+08:00 DS_Matt_8600k if_link_up_hook_event[15236]: eth0
2023-09-22T09:49:03+08:00 DS_Matt_8600k if_link_down_hook_event[18744]: eth0
2023-09-22T09:49:13+08:00 DS_Matt_8600k if_link_up_hook_event[19628]: eth0
2023-09-22T09:50:36+08:00 DS_Matt_8600k if_link_down_hook_event[22155]: eth0

wjz304, 2023-09-22:
现在就是

wjz304, 2023-09-22:
`ethtool -k ethN` 
看看

captcao, 2023-09-22:
Features for eth0:
rx-checksumming: on
tx-checksumming: on
        tx-checksum-ipv4: on
        tx-checksum-ip-generic: off [fixed]
        tx-checksum-ipv6: on
        tx-checksum-fcoe-crc: off [fixed]
        tx-checksum-sctp: off [fixed]
scatter-gather: off
        tx-scatter-gather: off
        tx-scatter-gather-fraglist: off [fixed]
tcp-segmentation-offload: off
        tx-tcp-segmentation: off
        tx-tcp-ecn-segmentation: off [fixed]
        tx-tcp6-segmentation: off
udp-fragmentation-offload: off [fixed]
generic-segmentation-offload: off [requested on]
generic-receive-offload: on
large-receive-offload: off [fixed]
rx-vlan-offload: on
tx-vlan-offload: on
ntuple-filters: off [fixed]
receive-hashing: off [fixed]
highdma: on [fixed]
rx-vlan-filter: off [fixed]
vlan-challenged: off [fixed]
tx-lockless: off [fixed]
netns-local: off [fixed]
tx-gso-robust: off [fixed]
tx-fcoe-segmentation: off [fixed]
tx-gre-segmentation: off [fixed]
tx-ipip-segmentation: off [fixed]
tx-sit-segmentation: off [fixed]
tx-udp_tnl-segmentation: off [fixed]
fcoe-mtu: off [fixed]
tx-nocache-copy: off
loopback: off [fixed]
rx-fcs: off
rx-all: off
tx-vlan-stag-hw-insert: off [fixed]
rx-vlan-stag-hw-parse: off [fixed]
rx-vlan-stag-filter: off [fixed]
l2-fwd-offload: off [fixed]
busy-poll: off [fixed]

captcao, 2023-09-22:
[image]

wjz304, 2023-09-22:
> Features for eth0: rx-checksumming: on tx-checksumming: on tx-checksum-ipv4: on tx-checksum-ip-generic: off [fixed] tx-checksum-ipv6: on tx-checksum-fcoe-crc: off [fixed] tx-checksum-sctp: off [...]

这是在arpl下取的 还是在dsm下取的？

wjz304, 2023-09-22:
我要dsm下的

captcao, 2023-09-22:
[image]

captcao, 2023-09-22:
是dsm下的

wjz304, 2023-09-22:
ethtool -K eth0 scatter-gather on
ethtool -K eth0 tso on
ethtool -K eth0 gso on

然后再取

captcao, 2023-09-22:
[image]

captcao, 2023-09-22:
好像这3条命令没起作用

wjz304, 2023-09-22:
嗯，我现在测试机都在测别的东西，
你看看 ethtool --help ，目的是打开 tso/gso 后试下

wjz304, 2023-09-22:
https://www.cnblogs.com/danxi/p/6591885.html

wjz304, 2023-09-22:
另外看一眼bios 把 pci 节能相关的关掉试一下

captcao, 2023-09-22:
那几个功能开不了，不知是不是8125bg不支持。bios我明天试试看~

FelixLAI85, 2023-10-03:
我是ESXi 8.0 u2虚拟的DS918+ DSM7.2.1-69057版本，之前也是用的引导自带的驱动，一旦流量比较大就会断流，要重新插拔一下网线才能正常工作，不确定是驱动版本自身的问题还是其它因素。后来我上网查了一下有另一个github作者有教怎么编译：https://github.com/tabrezm/r8125-synology，虽然版本比最新的要低一些，但是我按照他的命令自己编译安装驱动之后，目前使用两天没有断流过，你可以参考一下。这里有一点要注意一下，要在引导里面去掉加载的8125模块，要不然在安装自己编译的驱动的时候会提示已经存在相同的文件，无法安装。

captcao, 2023-10-04:
> 我是ESXi 8.0 u2虚拟的DS918+ DSM7.2.1-69057版本，之前也是用的引导自带的驱动，一旦流量比较大就会断流，要重新插拔一下网线才能正常工作，不确定是驱动版本自身的问题还是其它因素。后来我上网查了一下有另一个github作者有教怎么编译：https://github.com/tabrezm/r8125-synology 虽然版本比最新的要低一些，但是我按照他的命令自己编译安装驱动之后，目前使用两天没有断流过，你可以参考一下。这里有一点要注意一下，要在引导里面去掉加载的8125模块，要不然在安装自己编译的驱动的时候会提示已经存在相同的文件，无法安装。

好的 谢谢你 我这几天出差 回去试试看～ 确实像是arpl自带驱动的问题

wjz304, 2023-10-08:
> 好的 谢谢你 我这几天出差 回去试试看～ 确实像是arpl自带驱动的问题

旧版本的驱动试了吗？

marboroman, 2023-10-21:
[wjz304](https://github.com/wjz304), could you please include different versions in modules to choose if they solve problems (r8125-1.ko, r8125-2.ko, r8125-3.ko etc.)

wjz304, 2023-10-21:
> [wjz304](https://github.com/wjz304), could you please include different versions in modules to choose if they solve problems (r8125-1.ko, r8125-2.ko, r8125-3.ko etc.)

可以做到，但是很麻烦。因为现在并没有明确的条件判断应该用哪个

wjz304, 2023-10-21:
I still tend to find the root cause of the problem, or find a version where everyone is OK.

marboroman, 2023-10-21:
In my case, I’m not even sure that the network card is to blame, but I can’t get rid of problems with loading the system. If it boots once, it can work for an infinitely long time and stably, but today I had to turn it off and on again and nothing would load. Typically, loading from the outside looks like this: the bootloader receives an IP address, DSM loading begins, the router sees how the device lost its IP and then immediately received it again, the disks in the NAS began to crunch and the LEDs glow, at the same time the monitor goes out (the intel graphics driver is loading ). I started it for about 2 hours, waited until the router lost my network, but for the 2nd time it did not receive an IP address. At the same time, the discs did not begin to crunch and the screen could go out or remain on. I forcibly turned off the NAS by holding the power button and turned it on again. And after 15 attempts it finally booted. This use is very unreliable... I will kill disks and data with such reboots.

wjz304, 2023-10-22:
> In my case, I’m not even sure that the network card is to blame, but I can’t get rid of problems with loading the system. If it boots once, it can work for an infinitely long time and stably, but [...]

Have you changed it to "direct boot"?

captcao, 2023-10-22:
> 旧版本的驱动试了吗？

不好意思，最近忙到没空折腾，今天才闲下来，按照 https://github.com/tabrezm/r8125-synology 编译了r8125.ko，然后在arpl里面module部分替换了原先的驱动。初步测试了iperf3 线程正向打流5分钟，30线程逆向打流5分钟，正常了，不会大流量断流了~
后续正常使用看看会不会出现问题~

现在使用的版本是9.009.02

driver: r8125
version: 9.009.02-NAPI
firmware-version: 
expansion-rom-version: 
bus-info: 0000:03:00.0
supports-statistics: yes
supports-test: no
supports-eeprom-access: no
supports-register-dump: yes
supports-priv-flags: no

wjz304, 2023-10-22:
等RR架构改完， 降版本编个看看吧

FelixLAI85, 2023-10-22:
> 等RR架构改完， 降版本编个看看吧

我试过使用9.010.01和最新的9.011.01编译的驱动，均会遇到大流量断流的情况，只有9.009.02这个可以保持坚挺，目前没遇到过断流，可能还是这个版本的驱动比较稳定。

captcao, 2023-10-22:
> 等RR架构改完， 降版本编个看看吧

谢大佬~

captcao, 2023-10-22:
> > 等RR架构改完， 降版本编个看看吧
> 
> 我试过使用9.010.01和最新的9.011.01编译的驱动，均会遇到大流量断流的情况，只有9.009.02这个可以保持坚挺，目前没遇到过断流，可能还是这个版本的驱动比较稳定。

再次感谢~指点迷津~

captcao, 2023-10-22:
> > 等RR架构改完， 降版本编个看看吧
> 
> 我试过使用9.010.01和最新的9.011.01编译的驱动，均会遇到大流量断流的情况，只有9.009.02这个可以保持坚挺，目前没遇到过断流，可能还是这个版本的驱动比较稳定。

再次感谢~

marboroman, 2023-10-23:
> Have you changed it to "direct boot"?

yes, it didn't make difference.

wjz304, 2023-10-28:
https://github.com/wjz304/rr-modules/releases/tag/23.10.1

在modules中 勾选 r8125_v9.ko 并取消 r8125.ko
```
r8125.ko      --> v9.011.01
r8125_v9.ko --> v9.009.02
```

wjz304, 2023-10-28:
https://github.com/wjz304/rr/issues/289

wjz304, 2023-10-28:
https://github.com/wjz304/rr/issues/207

wjz304, 2023-10-29:
https://github.com/wjz304/rr/issues/147

FelixLAI85, 2023-11-02:
我是ESXi 8.0 u2虚拟的DS918+ DSM7.2.1-69057版本，使用V9版本驱动，还是会断流，用ethtool -i命令查看驱动信息的时候，发现作者提供的驱动版本后缀会带RSS字样，而我自己编译的驱动则是只有NAPI字样，和作者提供的9.011.01的驱动信息一致。现在还是只能上传自己的驱动，每次更新模块的时候又得重新上传一次，有点麻烦。不知道作者能否抽时间check一下两个版本的驱动有啥区别？非常感谢！顺带附上我自己使用的驱动
[9.009.02.zip](https://github.com/wjz304/rr/files/13233139/9.009.02.zip)

wjz304, 2023-11-02:
> 我是ESXi 8.0 u2虚拟的DS918+ DSM7.2.1-69057版本，使用V9版本驱动，还是会断流，用ethtool -i命令查看驱动信息的时候，发现作者提供的驱动版本后缀会带RSS字样，而我自己编译的驱动则是只有NAPI字样，和作者提供的9.011.01的驱动信息一致。现在还是只能上传自己的驱动，每次更新模块的时候又得重新上传一次，有点麻烦。不知道作者能否抽时间check一下两个版本的驱动有啥区别？非常感谢！顺带附上我自己使用的驱动 [9.009.02.zip](https://github.com/wjz304/rr/files/13233139/9.009.02.zip)

试一下 不带v9的，确认是RSS的问题

FelixLAI85, 2023-11-03:
> 试一下 不带v9的，确认是RSS的问题

两个驱动都试过了，不带v9的在iperf3打流测试下，最多撑30秒左右就会断流，带v9的可以稍微长点时间大概也是不到一分钟的样子，我每种都测过两次。最后还是只能换上自己编译的那个，才能坚持不断流。

wjz304, 2023-11-03:
> 两个驱动都试过了，不带v9的在iperf3打流测试下，最多撑30秒左右就会断流，带v9的可以稍微长点时间大概也是不到一分钟的样子，我每种都测过两次。最后还是只能换上自己编译的那个，才能坚持不断流。

行吧，就是只能 009 不带rss功能呗

wjz304, 2023-11-04:
> 两个驱动都试过了，不带v9的在iperf3打流测试下，最多撑30秒左右就会断流，带v9的可以稍微长点时间大概也是不到一分钟的样子，我每种都测过两次。最后还是只能换上自己编译的那个，才能坚持不断流。

https://github.com/wjz304/rr/releases/tag/23.11.1
v9版本关闭了 RSS 的支持

FelixLAI85, 2023-11-05:
> https://github.com/wjz304/rr/releases/tag/23.11.1 v9版本关闭了 RSS 的支持

我试过了，不管是升级到这个版本，还是全新安装这个版本，使用V9驱动都会断流，我还是只能用回自己编译的驱动才行，不知道其他人的情况如何，可能要收集更多case才能找到问题所在吧。

FelixLAI85, 2023-11-12:
> https://github.com/wjz304/rr/releases/tag/23.11.1 v9版本关闭了 RSS 的支持

螃蟹官网最近更新了新版驱动：[r8125-9.012.03.zip](https://github.com/wjz304/rr/files/13328477/r8125-9.012.03.zip)，大佬是否可以尝试添加到驱动库里试试？看到最新版本的SA6400解决了很多问题，我今天尝试了一下，发现了一些惊喜，原来在DS918下不能识别的NVME硬盘可以在SA6400下轻松识别，不过依照教程没办法编译r8125在SA6400下的9.009.02版本驱动，有些蛋疼。。。不过我还是打算继续停留在SA6400上了，尝试一下后续能否解决这个网卡的断流问题。

wjz304, 2023-11-14:
https://github.com/wjz304/rr/releases/tag/23.11.6
@FelixLAI85  try

FelixLAI85, 2023-11-15:
> https://github.com/wjz304/rr/releases/tag/23.11.6 @FelixLAI85 try

Just tried. I was really disappointed that it still cannot work very well! It is a big shame on Realtek! 尝试了好几次，还是不行，依然会断流，对瑞昱这家公司真的是非常失望，在这个驱动上这么小的一个问题，连续几个版本的驱动都无法处理好，可见这个公司的软件实力有多烂，以后但凡是瑞昱的网卡能够避开不买的，坚决不买。目前的情况是，我还是依然需要上传自己编译的9.009.02版本驱动，暂时还算OK，因为在上一个RR版本，我是连上传自己编译的驱动版本也会出现断流情况的，差点以为自己和SA6400要从此无缘，还好这个版本又重新恢复正常。你在SA6400做的很多更新和改进，都是我之前非常需要又无法获得的，目前我已经打算完全迁移到SA6400上了，毕竟5.X的内核还是比老旧的4.X版香多了。相信随着最近这个版本的更新，以后应该会有很多其他玩家和开发者跟进一块充实这个生态，你真的是我见过最勤奋的作者之一，such a great work!

FelixLAI85, 2023-11-16:
update一下，虽然用了9.009.02版本的驱动，凌晨4点的时候，这个网卡还是挂掉了。可能换了内核之后，驱动的运作和DS918又不一样了。我试试看能不能抓取一些日志看看是什么情况。

stkroot, 2023-11-19:
原来是驱动的问题啊，我一直以为是网线问题，也是8125B网卡，smb不定时卡死，速度剧降，插拔一次网线后恢复正常。重打了好几次水晶头都不太行，因为没有网线测试仪，我还以为是我压线钳坏了，😓

FelixLAI85, 2023-11-19:
> 原来是驱动的问题啊，我一直以为是网线问题，也是8125B网卡，smb不定时卡死，速度剧降，插拔一次网线后恢复正常。重打了好几次水晶头都不太行，因为没有网线测试仪，我还以为是我压线钳坏了，😓

是的，网络断了之后，插拔一次网线就可以恢复，就是驱动问题吧。暂时无解，之前在DS918+下，可以使用自己编译的9.009.02版本的驱动。我现在为了核显硬解和NVME，换到SA6400上了，暂时没有找到能够稳定运行不断流的驱动板版本。

wjz304, 2023-11-19:
https://github.com/wjz304/rr-modules/actions/runs/6918788082
可以一试

FelixLAI85, 2023-11-20:
> https://github.com/wjz304/rr-modules/actions/runs/6918788082
> 
> 可以一试

首先要感謝一下，这个版本的9.012.03版本的驱动在SA6400下有很大改善！在iperf3正向打流的情况下基本可以保持不断流。不过在反向模式下只能坚持几秒到最多几十秒就会断流。不过正常使用下暂时没有遇到断流的情况。9.009.02版本依然表现比较差所以没有使用。刚刚更新了pre-release的rr.23.11.7版本，貌似驱动也更新了，不过这个版本的9.012.03驱动不能获取ip,所以暂时无法测试这个版本的驱动运作情况，可能要麻烦您检查一下。另外还有一个问题，在几个版本里都会出现，我顺便反馈一下，就是当只勾选r8125,没有勾选r8125_v9，而且同时勾选了r8169_lk驱动的时候，r8125驱动无法正常加载，因为r8169_lk被优先加载了。

wjz304, 2023-11-20:
是的，r8169_lk 包含 螃蟹所有型号的网卡

wjz304, 2023-11-20:
很久之前是我从中剔除了r8125，但是改名lk的时候没有剔除

FelixLAI85, 2023-11-28:
> 现在23.11.10，8125是该好好选择8125还是8125v9

如果是DS918+的话，可能还是选v9吧。我现在已经转移到SA6400上了。网卡驱动目前无解，我都是用着虚拟的VMXNET3做代替，虽然速度比直通的8125差一些，也勉强够用了。因为目前新版RR里的SA6400有新内核，而且带来了12代以上核显的硬解，以及未来可以期待的一些令人振奋的新功能，比如无线WiFi，我的无线网卡是AX211，期待到时候可以派上用场！权衡利弊之下，就暂时忍受着这个基本不能用的显卡吧，因为SA6400其他的功能真的很棒，哈哈！

FelixLAI85, 2023-11-28:
> > > https://github.com/wjz304/rr-modules/actions/runs/6918788082
> > > 可以一试
> > 
> > 
> > 首先要感謝一下，这个版本的9.012.03版本的驱动在SA6400下有很大改善！在iperf3正向打流的情况下基本可以保持不断流。不过在反向模式下只能坚持几秒到最多几十秒就会断流。不过正常使用下暂时没有遇到断流的情况。9.009.02版本依然表现比较差所以没有使用。刚刚更新了pre-release的rr.23.11.7版本，貌似驱动也更新了，不过这个版本的9.012.03驱动不能获取ip,所以暂时无法测试这个版本的驱动运作情况，可能要麻烦您检查一下。另外还有一个问题，在几个版本里都会出现，我顺便反馈一下，就是当只勾选r8125,没有勾选r8125_v9，而且同时勾选了r8169_lk驱动的时候，r8125驱动无法正常加载，因为r8169_lk被优先加载了。
> 
> 我突然想起 8125 我已经从8169中剔除了，但是 3000 似乎没有剔除，你莫非是3000？ 一直也没取pidvid，

我在ESXi下看到的id是0x10ec，我的主板是铭瑄的B760i,上传个截图你看看吧
[image]

FelixLAI85, 2023-11-28:
> 只是能确定 3000 有问题，没日志依然解不了

有没有一些具体一点的instruction可以让我去抓一下日志？在SSH下？用什么命令抓取？最近比较忙，没有去钻研。。。

fxljay, 2023-11-28:
> > 现在23.11.10，8125是该好好选择8125还是8125v9
> 
> 如果是DS918+的话，可能还是选v9吧。我现在已经转移到SA6400上了。网卡驱动目前无解，我都是用着虚拟的VMXNET3做代替，虽然速度比直通的8125差一些，也勉强够用了。因为目前新版RR里的SA6400有新内核，而且带来了12代以上核显的硬解，以及未来可以期待的一些令人振奋的新功能，比如无线WiFi，我的无线网卡是AX211，期待到时候可以派上用场！权衡利弊之下，就暂时忍受着这个基本不能用的网卡吧，因为SA6400其他的功能真的很棒，哈哈！

我这里很奇怪，威联通往黑裙DS920  2.5G 上行 下行都能跑满    但是黑裙往威联通只能跑到1000M  这是为什么

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304