RROrg/rr #232: [BUG?]型号选1823xs+时，自定义MAC地址不起作用

Moth338, 2023-09-23:
Install ENV:  
 * motherboard:  B560i aorus pro ax
 * CPU:  i5-11500
 * NIC:  i225v

Arpl version:  
 * arpl:  23.9
 * addons:  无
 * modules:  无
 * lkms:  默认

DSM: 
 * model:  DS1823xs+
 * version:  7.2

Issue:  型号选1823xs+时，自定义MAC地址不起作用

logs:  

(Plz review the content of #173, #175, #226 first)  
...
硬改MAC失败（eeupdate提示修改成功，但实际未成功）
在定制中加入洗白MAC时提示软改MAC成功获取到IP，但重启后用synology assistant查找到主机发现MAC地址仍为原地址，且IP与arpl启动DMS时显示的IP不同。
同机采用DS918+型号则软改成功为洗白MAC。sa查找到的MAC也为洗白MAC。
请问我要怎么定制才行？想要8盘的硬盘管理界面。

wjz304, 2023-09-23:
修改mac的过程都是一样的，

只是需要 盘位可以用 storagepanel 插件  https://github.com/wjz304/arpl-i18n/issues/208

Moth338, 2023-09-24:
好的，我试试，谢谢回复。
另外这个1823xs+软改MAC的问题是否能测试一下？我两块主板（技嘉）都不行。

wjz304, 2023-09-24:
> 好的，我试试，谢谢回复。 另外这个1823xs+软改MAC的问题是否能测试一下？我两块主板（技嘉）都不行。

你只试过 1823 和 918吗？ 其他的有没有试过？

wjz304, 2023-09-24:
[image]

cat /proc/cmdline
cat /sys/class/net/*/address

看看吧

Moth338, 2023-09-24:
[image]
[image]
确实是这样，软改不成功，00:11……这个MAC是我硬改的，9009……这个是软改MAC，1823启动画面上的1.159这个IP是不通的。
虚拟机我试过，没问题，就物理机不成功。
不知道是怎么回事。

wjz304, 2023-09-24:
mac 是lkm设置的，虽然我一直想去掉 lkm的这个操作，但是会影响账号相关功能，很麻烦，
没遇到过你这情况，可能和lkm的某些机制有关吧。
我研究研究再说吧。

wjz304, 2023-09-25:
计划干掉 cmdline里 mac 和网卡的关系，让它只作为激活用 和不作为网卡的mac用

Moth338, 2023-09-26:
感谢回复！期待ing……

wjz304, 2023-10-03:
最新的 取消mac的关系，不知道能不能满足你的需求？

ikapc, 2023-10-09:
> 最新的取消mac的关系，不知道能不能满足你的需求？

大佬，请教一下，取消cmdline里的mac和中断的关系后，的版本需要怎么把mac修改进系统呀，现在怎么修改和自定义mac群晖显示的都是物理mac，无法更改了

wjz304, 2023-10-09:
cmdline里mac1 如果是白的就可以洗白了，网卡的mac不影响， 

不知道你改mac是有什么需求？
非要改 在dsm系统下通过 ip/ifconfig 命令自己改就可以了，永久就添加计划任务

ikapc, 2023-10-09:
> cmdline里mac1 如果是白的就可以洗白了，网卡的mac不影响，
> 
> 不知道你改mac是有什么需求？ 非要改 在dsm系统下通过 ip/ifconfig 命令自己改就可以了，永久就添加计划任务

好的，感谢大佬的耐心回复

closed 2023-10-16 by wjz304