RROrg/rr #235: RebootToARPL addon in 23.9.6 is non-working

TheChief5055, 2023-09-26:
`root@Synology:~# /usr/bin/arpl-reboot.sh "config"
-ash: /usr/bin/arpl-reboot.sh: No such file or directory
root@Synology:~# ls -l /usr/bin/*.sh
-rwxr-xr-x 1 root root 38417 Sep 26 08:05 /usr/bin/cpuinfo.sh
-rwxr-xr-x 1 root root 62704 Sep 26 08:05 /usr/bin/hdddb.sh
-rwxr-xr-x 1 root root  2046 Sep 26 08:05 /usr/bin/nvmecache.sh
root@Synology:~#
`

wjz304, 2023-09-26:
确定 插件没有被移除吗？

wjz304, 2023-09-27:
This script is actually in misc addons, misc must be added by default.
Please check if /usr/bin/grub-editenv exists?

If it does not exist, there may be an error that interrupts the installation process of addons. It is recommended to use the latest version of arpl and addons.

TheChief5055, 2023-09-27:
> This script is actually in misc addons, misc must be added by default.

No 'misc' addon in installed addons list, no available to install.

> Please check if /usr/bin/grub-editenv exists?

Doesn't exist. 

PS: my second 918+ xpenology behaves same way — in DSM 7.2 after update ARPL to 23.9.6 it still works and grub-editenv exists in /usr/bin, but after update to DSM 7.2.1 they're gone.

wjz304, 2023-09-27:
> No 'misc' addon in installed addons list, no available to install.
>
>
> Doesn't exist.
>
> PS: my second 918+ xpenology behaves same way — in DSM 7.2 after update ARPL to 23.9.6 it still works and [...]

Misc is added by default and cannot be deleted.
So I don't know why you're in this situation.

wjz304, 2023-09-28:
https://github.com/wjz304/arpl-addons/releases/tag/23.9.7
try it,

TheChief5055, 2023-09-28:
> https://github.com/wjz304/arpl-addons/releases/tag/23.9.7 try it,

Yes, it works! Thanx a lot.

closed 2023-09-28 by TheChief5055