RROrg/rr #237: question

engineinheart, 2023-09-27:
Install ENV:  
 * motherboard:  MSI B150 cafe 10g
 * CPU:  Intel g4560
 * NIC:  8111 onboard

Arpl version:  
 * arpl:  23.9.6
 * addons:  storagepanel
 * modules:  mpt3sas for LSI 2308 and 5212m4 expender

DSM: 
 * model:  DS918+
 * version:  7.2 69057

Issue:  1、storagepanel配置无效，依旧只显示默认的4盘位
           2、所有硬盘的通电时间一致，固件版本一致
 
thanks

1360633524, 2023-09-27:
用hba卡没？
918用hba卡就这样，在6.23直接看不到smart，只有打GXNAS的0.12.1扩展驱动包才可以正确显示smart，并且打了以后硬盘不能休眠不然会损毁(没什么影响，硬盘基本没法休眠)，目前我是7.2，也是所有硬盘信息都一样，smart看不了正确数值很不安全。希望作者能参考下GXNAS的补丁修复下

wjz304, 2023-09-27:
addons version？

crazyb123, 2023-09-28:
[image]
[image]
这个是ds3622xs+ dsm7.2-64570U3的界面
我这还有台ds2422+ 升级到最新版dsm7.2.1-69057，会缺失图中红框内容，不知道是否官方就是这样，还是其他问题引起的，在升级前和ds3622xs+一样，可以查看硬盘smart信息

wjz304, 2023-09-28:
> [image] [image] 这个是ds3622xs+ dsm7.2-64570U3的界面 我这还有台ds2422+ 升级到最新版dsm7.2.1-69057，会缺失图中红框内容，不知道是否官方就是这样，还是其他问题引起的，在升级前和ds3622xs+一样，可以查看硬盘smart信息 ds2422+的addons是采用的最新版23.9.6 ds2422+没采用HBA卡，都是主板上的sata口

[image]

wjz304, 2023-09-28:
> addons version？

@engineinheart

wjz304, 2023-09-28:
[image]

engineinheart, 2023-09-28:
> addons version？

arpl23.9.6默认版本，没有做改动，重启后也不生效

wjz304, 2023-09-28:
> > addons version？
> 
> arpl23.9.6默认版本，没有做改动，重启后也不生效

https://github.com/wjz304/arpl-i18n/issues/237#issuecomment-1738388319

engineinheart, 2023-09-28:
> > > addons version？
> > 
> > 
> > arpl23.9.6默认版本，没有做改动，重启后也不生效
> 
> [#237 (comment)](https://github.com/wjz304/arpl-i18n/issues/237#issuecomment-1738388319)

sorry,没仔细看更新说明，以为arpl更新了addons也会同步更新，storagepanel已解决，谢谢！另外请问所有硬盘的通电时间和固件版本一致是系统底层原因修复不了吗？

wjz304, 2023-10-12:
SMART问题 same as https://github.com/wjz304/arpl-i18n/issues/238

closed 2023-10-12 by wjz304