RROrg/rr #239: 更新到DMS7.2.1（69057）后，除了存储池1，其他存储池全部丢失

theseato, 2023-09-28:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  QEMU Standard PC (Q35 + ICH9, 2009), BIOS 0.0.0 02/06/2015
 * CPU:  11th Gen Intel(R) Core(IM) i5-11400F 0 2.60GHz
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  ARPL-i18n V23.9.6
 * addons:  V23.9.7
 * modules:  V23.9.5
 * lkms:  V23.9.2 （dev）

DSM: 
 * model:  DS3617xs
 * version:  7.2.1（69057）

Issue:  更新到DMS7.2.1（69057）后，除了存储池1，其他存储池全部丢失。

[image]

[image]

[image]

doandat943, 2023-09-28:
@wjz304 Happy Mid Autumn Festival, hope you and your family have good holiday. Thank you for all work <3

category-9431, 2023-09-28:
in proxmox vm, I have a same issue

[image]
[image]

BIG5T, 2023-09-29:
@theseato @category-9431 @wjz304 祝各位中秋节快乐，感谢@wjz304 和所有有贡献的人在此项目的工作。我遇到了同样的情况，已经解决。简单的说是因为原来的SATA端口变成了ESATA，拆机把硬盘数据线拔下来换一个SATA接口即可解决（不行就再换一个接口）。想从根源上解决问题要修改引导文件，等着引导更新吧。
喜欢折腾的朋友接着看下面：
连接SSH运行 cat /etc.defaults/synoinfo.conf ​​​​​​​
文件中查找以下几个配置
esataportcfg="0x10" #这里定义了主板上的哪个端口是esata端口，把16进制转换成2进制，1表示是esata，0表示否。
internalportcfg="0x1ffef" #这里定义了主板上的哪个端口是sata端口，把16进制转换成2进制，1表示是sata，0表示否。同上相反。

0x1ffef转换成二进制后=11111111111101111，我发现我消失的那个硬盘正好是第五个，就是变成0那里。我尝试改回，但是每次重启都会被引导覆盖成0x1ffef，所以无奈只能在主板上直接换一个端口了，虽然不太优雅，但好歹解决了问题。

这两个参数的值是十六进制，转换成二进制后有几位数表示有几个端口，从右往左开始数，每一位代表一个端口，修改成1或者0可以精确定位修改端口模式。据我测试，以上两个参数要同时修改并且相反才生效。例如修改esataportcfg的值为二进制00010，那么internalportcfg必须=11101，因为一个端口只能工作在一个模式上，这个端口不能同时是esata又是sata，所以不能在两个参数里都是1。
sataportcfg=0x2 #二进制00010
internalportcfg=0x1d  #二进制11101
意思是主板有5个sata端口，其中第二个sata端口启用esata模式。有几个端口就要搞几位，少了可能会导致硬盘无法识别。。。。弄的时候用2进制，记得把二进制转换成十六进制再填进去。
这里还可以通过修改usbportcfg参数把USB设备修改成esata，拿来外接USB硬盘盒子当成esata设备使用。esata模式支持NTFS格式的硬盘直接在群晖里使用，这样就能在Windows和群晖下无缝切换了，好处很多就不展开说了，想了解的自行搜索相关资料。

声明：以上是我自行实践得出的个人结论，仅供参考，在测试的时候请自行备份好数据，确保数据、设备和人身安全，本人不对任何人的数据和硬件以及折腾过程中的人身安全负责，参考使用即表示同意本声明。

wjz304, 2023-09-29:
> @theseato @category-9431 @wjz304 祝各位中秋节快乐，感谢@wjz304 和所有有贡献的人在此项目的工作。我遇到了同样的情况，已经解决。简单的说是因为原来的SATA端口变成了ESATA，拆机把硬盘数据线拔下来换一个SATA接口即可解决（不行就再换一个接口）。想从根源上解决问题要修改引导文件，等着引导更新吧。 喜欢折腾的朋友接着看下面： 连接SSH运行 cat [...]

https://github.com/wjz304/arpl-i18n/issues/226#issuecomment-1726628716

theseato, 2023-09-29:
更新到 ARPL-i18n V23.9.7之后，存储池丢失的这个问题依旧存在。

theseato, 2023-09-29:
@BIG5T 
我的参数情况是：
esataportcfg="0x00" 
internalportcfg="0x84ff" （二进制为：1000010011111111），目前存储池丢失的就是硬盘9和硬盘10，因为相对应的二进制数字是“0”。
虽然我可以硬盘11可用，但之后的又是一串0，我的开始盘符是从硬盘8开始的，所以我哪怕把硬盘拆下来换sata口也不够我更换啊。
但只是修改参数，重启之后又被原参数覆盖……所以，现在就尬在这里了，草

BG5T, 2023-09-29:
> @BIG5T 我的参数情况是： esataportcfg="0x00" internalportcfg="0x84ff" （二进制为：1000010011111111），目前存储池丢失的就是硬盘9和硬盘10，因为相对应的二进制数字是“0”。 虽然我可以硬盘11可用，但之后的又是一串0，我的开始盘符是从硬盘8开始的，所以我哪怕把硬盘拆下来换sata口也不够我更换啊。 但只是修改参数，重启之后又被原参数覆盖……所以，现在就尬在这里了，草

你可以进入引导设置界面，选择设置synoinfo，添加synoinfo参数，esataportcfg 值0x00 ，再添加一个参数internalportcfg 值0x1ffff，重启。

BG5T, 2023-09-29:
@theseato 
好了没，你这也太慢了，快搞啊，等会儿我玩游戏去了。

theseato, 2023-09-29:
@BIG5T 修改了，还是不成功。

theseato, 2023-09-29:
@BIG5T 修改了引导设置。重启，进入群晖之后用SSH查看，结果依旧是 internalportcfg="0x84ff" 。

BG5T, 2023-09-29:
@theseato **这是最后的倔强**：
连接ssh,运行 
`sudo -i `
输入密码后会变成
`root@xxx`
这样的前缀，
然后，接着
`cd /dev`
在你已经存在的文件夹里新建一个文件夹用来挂载boot,我假设你存在名为docker的共享文件夹
`mkdir -p /volume1/docker/boot`
将synoboot1 分区挂载到boot
`mount -t vfat synoboot1 /volume1/docker/boot`
然后你进`docker/boot`文件夹用文本编辑器去编辑`user-config.yml`文件（如果编辑不了，复制出来改完粘贴覆盖回去。）
找到synoinfo那一段
按照文本已有格式手动加入以下参数（注意与其它参数对齐）：
```
esataportcfg:0x00
internalportcfg:0xffff
```
保存。
保重。
重启。

BIG5T, 2023-09-30:
@theseato @category-9431 楼上BG5T是我小号。我早上测试了一下，**找到终极完美解决办法**，你先 
1. 进入引导设置界面，重新编译一下引导（目的是让它恢复默认状态）
2. 点高级设置，编辑用户文件，找到synoinfo配置段把这里`internalportcfg: 0x1ffef`的值改成 `0x1ffff` 并确认。

[image]

3. 回到主界面会提示你配置已改重新编译引导，你点` 是`（如果没提示你自己再次点击编译引导），编译完重新启动就修改成功没问题了。

[image]

[image]

wjz304, 2023-09-30:
> @theseato @category-9431 楼上BG5T是我小号。我早上测试了一下，**找到终极完美解决办法**，你先
> 
> 1. 进入引导设置界面，重新编译一下引导（目的是让它恢复默认状态）
> 2. 点高级设置，编辑用户文件，找到synoinfo配置段把这里的值改成  并确认（只是举个例子，也行，自己按上面的方法根据需要自行调整数值）。`internalportcfg: 0x1ffef``0xffff``0x1ffff`
> 
> [image]
> 3. 回到主界面会提示你配置已改重新编译引导，你点（如果没提示你自己再次点击编译引导），编译完重新启动就修改成功没问题了。` 是`
> 4. 每次系统升级后进引导界面检查一下用户配置再启动。我测试的结果是升级后internalportcfg项没有被改变，可以不用管，但是为了安全起见升级完建议先检查完再启动系统，避免掉盘。
> 
> [image] [image]
> @wjz304 建议在新写的引导盘里设置默认synoinfo配置internalportcfg项全为1，或者在引导编译之前先自动检测一下sata端口，然后自动设置，要不然全新升级引导盘后导致磁盘丢失就是一件很蛋疼的事情。我组的raid5因为掉盘后来跑了十几个小时修复才完成，又跑了十几个小时数据清理。幸好我成功了，网上看到很多raid5修复十几小时最后修复失败的例子。

发你你又不看

BIG5T, 2023-09-30:
> 发你你又不看

@wjz304 我看了，写的很全面，很感谢你的辛劳付出，但对于新手略微复杂，而且是英文的，有些朋友应该看不懂，标题太高大上一时间没有联想到跟这个问题有关，直接改成`掉盘硬盘相关问题存储池和设备问题看这里修改`更为友好，因为这几个词都是搜索高频关键词。一般不搞开发的应该没听过啥是Device Tree，更不会想到去搜索这个词相关资料。狗头保命.jpg（我深知言禁交浅言深，挂个狗头保命，如有冒犯还请不要计较，当没看见就好哈）
我个人只是想着能默认开局新手村全是SATA最好不过了，新手应该也不会有那么复杂的场景，有需要改成ESATA的再自行来这里查资料谨慎修改，避免本来用着好好的升个级触不及防来一下掉盘吓死人 :) 
谢谢你在百忙之中还看完回复我这么多次。

theseato, 2023-09-30:
> @wjz304 我看了，写的很全面，很感谢你的辛劳付出，但对于新手略微复杂，而且是英文的，有些朋友应该看不懂，标题太高大上一时间没有联想到跟这个问题有关，直接改成`掉盘硬盘相关问题存储池和设备问题看这里修改`更为友好，因为这几个词都是搜索高频关键词。一般不搞开发的应该没听过啥是Device [...]

@wjz304 没错，@BIG5T 兄弟说的很对。其实我也看了那个Device Tree的帖子，额，看得一脸懵逼，直接被劝退了，嘤嘤嘤