RROrg/rr #240: BUG 23.9.7以后，修改MAC地址不生效了？

VincentShin, 2023-09-30:
RT，修改后进群晖还是原始地址，路由器显示也是原始地址

VincentShin, 2023-09-30:
Install ENV: (You can find it in the boot interface.)

DMI:OEM ITX-SC2/ITX-SC2, BIOS 5.12 04/23/2023
CPU:INTEL(R) Core(TM) i3-8100T CPU @ 3.10GHz
NIC:没看到这个参数
Arpl version: v23.9.7

arpl:i18n v23.9.7
addons:添加了真实cpu信息、storagepanel两个插件
modules:i915,R8125 2.5G 网卡驱动
lkms:prod
DSM:

model:7.2.1
version:(69057)
Issue:ARPL升级23.9.7以后，修改MAC地址不生效了，修改后进群晖还是原始地址，路由器显示也是原始地址

> [image]
>
> Please fill in the following information.
>
> Install ENV: (You can find it in the boot interface.)
>
> * DMI:
> * CPU:
> * NIC:
>
> Arpl version: (You can find it in the update menu.)
>
> * arpl:
> * [...]

RainThings, 2023-09-30:
[image]
一看就是没有好好看Release的更新说明，mac地址现在只跟洗白的部分有关了，虽然你看着没变，但是需要洗白才能过的操作还是可以正常执行的（登账号，AME解码包安装，QC）

VincentShin, 2023-09-30:
> [image] 一看就是没有好好看Release的更新说明，mac地址现在只跟洗白的部分有关了，虽然你看着没变，但是需要洗白才能过的操作还是可以正常执行的（登账号，AME解码包安装，QC）

之前看到了，就是这表述确实没看明白。开机后不敢登账号。现在登上是正常能用的，谢谢。

closed 2023-09-30 by VincentShin