RROrg/rr #242: BUG

hxhwing0522, 2023-10-02:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  N3160
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  arpl-i18n-23.9.7.img
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  
你好。
我把鏡像寫入U盤後，插入機子後，沒法順利進入編譯選單。(反而用arpl-zh_CN版卻能正常進入選單。)
試了很多種方法，把硬盤拔掉，只留一塊並格式化。
最後還是顯示 
end kernel panic - not sncing: VFS: unable to mount root fs on unknown-block(1,0)
end kernel panic - not sncing: No working init found. Try passing init= option, kernel] 

logs:  

(Plz review the content of #173, #175, #226 first)  
...
[image]

[image]

wjz304, 2023-10-02:
用 DG 打开 arpl.img 在 /boot/grub/grub.cfg  107行的未知 加 root=/dev/ram  试下
[image]

hxhwing0522, 2023-10-02:
笑死 我大半天換了N個方法刷了N次都不行  
結果現在一刷又可以了 : ) 
這是欺負我嗎?  不告狀都不給我好好工作 哈哈哈

> 用 DG 打开 arpl.img 在 /boot/grub/grub.cfg 107行的位置 加 root=/dev/ram 试下 [image]

wjz304, 2023-10-02:
可能是写盘的问题

closed 2023-10-05 by wjz304