RROrg/rr #244: Please add support for network card NetLink BCM57780 Gigabit Ethernet

adixus, 2023-10-02:
Please...help....with this driver. I found this but I don't know how to use it https://deviceinbox.com/id/ven-14E4-dev-1692.html   or maybe this: https://linux-hardware.org/?id=pci:14e4-1692-1028-0400

Thx!

wjz304, 2023-10-02:
module -> tg3.ko

adixus, 2023-10-02:
not working :(

wjz304, 2023-10-02:
Get log.
refer to https://github.com/wjz304/arpl-i18n/issues/173#issue-1851945793

wjz304, 2023-10-02:
1. logs is a folder,  `ls /mnt/p1/`
2. The function of dbgutils is to capture the startup log and save it to /mnt/p1. It has nothing to do with the Internet. I want to check the log to find the error information related to the network card loading.

adixus, 2023-10-02:
[image]

this is how it appears when I try to install. I have no option to activate that addon

adixus, 2023-10-02:
[image]

I managed this. What can I do next?

wjz304, 2023-10-03:
Oh, you didn’t make it clear. It turns out that you didn’t get the IP of tg3 through arpl.

wjz304, 2023-10-03:
> [image]
> 
> I managed this. What can I do next?

Execute
```
dmesg | grep tg3
ls /sys/class/net/
ifconfig
```

adixus, 2023-10-03:
[image]

wjz304, 2023-10-05:
```
rmmod tg3
modprobe tg3
ifconfig
```

wjz304, 2023-10-05:
It seems that this is an old problem of tg3, and no solution has been found yet.

If the network card can work after rmmod/modprobe, you can compile and install and change to direct startup in the advanced menu, and try to link to DSM, because the arpl problem will be ignored after direct startup, and you only care about whether dsm can be driver it.

kyonefq, 2023-10-06:
> ```
> rmmod tg3
> modprobe tg3
> ifconfig
> ```

[image]
没有驱动起来，请问能增加bcm57780的驱动吗？

wjz304, 2023-10-08:
> > ```
> > rmmod tg3
> > modprobe tg3
> > ifconfig
> > ```
> 
> [image] 没有驱动起来，请问能增加bcm57780的驱动吗？

23.10.0 更新了 buildroot，试下吧，不过大概率依然不ok

kyonefq, 2023-10-13:
> > > ```
> > > rmmod tg3
> > > modprobe tg3
> > > ifconfig
> > > ```
> > 
> > 
> > [image] 没有驱动起来，请问能增加bcm57780的驱动吗？
> 
> 23.10.0 更新了 buildroot，试下吧，不过大概率依然不ok

还是不行，可以装PVE，用虚拟网卡然后arpl引导安装

wjz304, 2023-10-15:
https://github.com/wjz304/arpl-i18n/actions/runs/6522571545
@kyonefq @adixus  test it

adixus, 2023-10-17:
@wjz304 after boot DSM lose connection with router...
[image]

wjz304, 2023-10-17:
> [image]

看来现在 arpl 的问题解决了

> @wjz304 after boot DSM lose connection with router... [image]

DSM 的问题，依然需要获取日志来确认
https://github.com/wjz304/arpl-i18n/issues/173

wjz304, 2023-10-18:
> @wjz304 logs folder https://s.go.ro/x5oyowdi | password: 516553

[    6.666605] tg3.c:v3.137 (May 11, 2014)
[    6.692561] kvm: VM_EXIT_LOAD_IA32_PERF_GLOBAL_CTRL does not work properly. Using workaround
[    6.695242] libphy: tg3 mdio bus: probed
[    6.695482] tg3 0000:02:00.0: No PHY devices
[    6.695761] tg3 0000:02:00.0: Problem fetching invariants of chip, aborting

wjz304, 2023-10-19:
> @wjz304 and what does that mean? :)

I am trying to fix this problem and will give you a beta version around tomorrow

wjz304, 2023-10-19:
> https://github.com/wjz304/arpl-i18n/actions/runs/6522571545 @kyonefq @adixus test it

https://github.com/wjz304/arpl-modules/actions/runs/6577695194
Please use this version of arpl and upload this modules

adixus, 2023-10-19:
@wjz304 and how do I upload the modules?

adixus, 2023-10-19:
@wjz304 I found it and it works now

wjz304, 2023-10-20:
OK，The next version will be released

wjz304, 2023-10-20:
https://github.com/wjz304/arpl-i18n/releases/tag/23.10.4

kyonefq, 2023-10-20:
感谢大神！我这边也可以了，可以引导安装了。

closed 2023-10-20 by wjz304