RROrg/rr #245: Error with real sn/mac dva1622-dva3219

jlvlnzl, 2023-10-04:
hi, i'm having error "Unable to perform operation. Contact synology Technical Support" (AME-QuickConnect) with previously real sn/mac working on dva1622-dva3219.
I'm using lastest release 23.9.7, updated from 23.9.4.

Thanks a lot

jlvlnzl, 2023-10-04:
DMI: VMware6.7
CPU:i5-8500t
NIC:vmxnet3
Arpl version: 23.9.7

arpl:23.9.7
addons:23.9.7
modules:23.9.5
lkms:23.9.2
DSM:DSM 7.2.1-69057

model:dva1622
version:DSM 7.2.1-69057
Issue:  having error "Unable to perform operation. Contact synology Technical Support" (AME-QuickConnect) with previously real sn/mac working on dva1622-dva3219.
I'm using lastest release 23.9.7, updated from 23.9.4.

logs:no log was created by this error

wjz304, 2023-10-05:
[image]
Is it an upgrade from an old version?

jlvlnzl, 2023-10-05:
hi, from 23.9.4 to 23.9.7. Mac1 was filled statically and still does.

wjz304, 2023-10-05:
After the upgrade, have you confirmed that mac1 is your mac1?
The error seems to occur when the mac does not correctly

jlvlnzl, 2023-10-05:
Yes, ping shows correct mac address on ARP table. SN is ok but same error. I tried installing Arc loader with same Mac/SN and it's working correctly..

wjz304, 2023-10-05:
> "filled statically", "correct mac" 

To be honest, English is not my native language and I don’t quite understand the true meaning of these words.

Because I blocked the process of setting cmdline's mac to the physical network card, arp should display the real mac address of the network card.

Of course, I only used 918’s sn/mac for testing. It does not affect account-related functions.
I don't have the sn/mac of DAV1622 so I don't know if there's anything special about it.
To be verified.

wjz304, 2023-10-05:
The current Arc loader does not blocked cmdline's mac1 set to the physical network card, and  so there is no problem with it.

Currently you can use arc, or fall back to arpl-i18n v23.9.6,
I want to verify this question.

wjz304, 2023-10-05:
If it is convenient for you, you can give me the result of `cat /proc/cmdline` (note that sn/mac uses * instead),

wjz304, 2023-10-12:
No further response, this issue is closed

closed 2023-10-12 by wjz304

blessendor, 2024-08-24:
> ![image](https://private-user-images.githubusercontent.com/5615843/272752896-c8633d9e-5009-4658-a9f8-3764e7985db7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3L [...]

Is any way to use a real mac addresses during boot and install DSM with RR loader? I need to use a real mac, generated by Hetzner hosting and use reserved IP address.

wjz304, 2024-08-24:
> > ![图像](https://private-user-images.githubusercontent.com/5615843/272752896-c8633d9e-5009-4658-a9f8-3764e7985db7.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjQ1MTA3NzQsIm5iZiI6MTcyNDUxMDQ3NCwicGF0aCI6Ii81NjE1ODQzLzI3Mjc1Mjg5Ni1jODYzM2Q5ZS01MDA5LTQ2NTgtYTlmOC0zNzY0ZTc5ODVkYjcucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDgyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDA4MjRUMTQ0MTE0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9ZDA3OTQwOWZlZmZiZjY2MDBjYjlhMDBlOGI0YjVhNzg4YzcxNWVjZDJkZThmZGQwNGU3MDc3ZGIxNGFjNDkwNyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.h20ZDw2m9bSK_yDiU3PMhGwDU6VG6Uwx6audyJbRoQE)它是从旧版本升级的吗？
> 
> 有什么方法可以在引导过程中使用真实的 mac 地址，并通过 RR 加载程序安装 DSM？我需要使用由 Hetzner 托管生成的真实 mac 并使用保留的 IP 地址。

Your question is not related to this issue. Please create a new issue.