RROrg/rr #246: DS918+ Model Not Recognizing M.2 NVMe SSD as Usable Storage Volume on Dell Latitude 5400

doomsday616, 2023-10-05:
Install ENV: (You can find it in the boot interface.)
 * DMI: Dell Latitude 5400
 * CPU: i3-8145U @2.10GHz Dual-Core
 * NIC: (Please specify your Network Interface Card model here)

Arpl version: (You can find it in the update menu.)
 * arpl: (Please specify the arpl version here)
 * addons: (Please specify any addons here)
 * modules: (Please specify any modules here)
 * lkms: (Please specify any lkms here)

DSM:
 * model: DS918+
 * version: 9.2

Issue:
I have installed an M.2 NVMe SSD (1TB) on my Dell Latitude 5400 laptop, which has 32GB of RAM and is powered by an i3-8145U @2.10GHz Dual-Core CPU. The SSD is correctly installed and is recognized in the ARPL boot interface. However, in the usable disk interface, my M.2 SSD is not highlighted in green. I have set the motherboard to AHCI, did not use any RAID cards, and used Rufus to write the arpl.img. I am also using the latest version of ARPL.

I have a couple of questions:
1. Do I need to load custom drivers for my M.2 SSD to be recognized as a usable disk? If yes, what are the steps to load custom drivers?
2. Is there any specific formatting required for the M.2 SSD before installing DSM?

logs:
(If there are any relevant logs, please include them here.)

(Plz review the content of #173, #175, #226 first)
...

wjz304, 2023-10-05:
[image]
安装这两个addons，（hdddb 安装后需要重启一次）

doomsday616, 2023-10-05:
> [image] 安装这两个addons，（hdddb 安装后需要重启一次）

你好，我已经安装了这两个addons，依然提示无法检测到硬盘
[
[image]
[image]
[image]
](url)

doomsday616, 2023-10-05:
你好，我安装了这2个插件，依然提示无法检测到硬盘
[image]
[image]
[image]

wjz304, 2023-10-05:
哦，首次安装 不支持纯ssd，需要有 sata 先装上

hdddb 是硬板数据库 也需要安装
[image]

doomsday616, 2023-10-05:
原来如此，感谢回复，我想再和您确认一下，请问第一次一定要hhd，还是说，sata的ssd也可以。
不过值得一提的是，我在另一台Latitude 5480上，只装了一块256G的nvme ssd，成功装上系统了，目前我还不理解为什么这块1TB的不行……

wjz304, 2023-10-05:
sata的ssd也可以。

doomsday616, 2023-10-05:
非常抱歉再次打扰您，我刚刚确认了一下，我的笔记本电脑（Latitude 5400）并没有SATA接口，只有M.2，请问是不是就没办法安装DSM了

wjz304, 2023-10-05:
目前我没研究，你可以看jim大的博客，

doomsday616, 2023-10-05:
您好，我成功让电脑识别到我的硬盘了，但是现在一直说文件已损毁
[image]

wjz304, 2023-10-30:
纯m2安装，目前不在计划范围内