RROrg/rr #250: esxi直通i350T4，启动arpl时网口reset

jetlins, 2023-10-07:
环境：esxi6.7 u3，虚拟openwrt，虚拟群晖918+。
配置：i3-9100，i350T4四口千兆直通，SATA控制器直通。
群晖版本：7.2 64570，群晖虚机分配i350t4的一个网口，openwrt分配两个网口，
问题：更新arpl引导到23.9.6（后经测试23.8.3及以上版本），只要启动arpl的设置页面，openwrt就会掉线，无法上网，查看openwrt日志，arpl在启动时可能会将整块i350T4的四个端口进行reset，并在不断的循环启动。
重新部署了一台群晖虚机测试，群晖的网卡使用未直通的vmnic网卡，arpl引导网络就一切正常。
更换不同作者的openwrt固件测试故障依旧。

jetlins, 2023-10-07:
[image]

jetlins, 2023-10-07:
[image]

jetlins, 2023-10-07:
[image]

wjz304, 2023-10-07:
虚拟机配置文件添加
pciPassthru0.msiEnabled = “FALSE“

grub 加 
pci=nomsi pci=pcie_scan_all

wjz304, 2023-10-07:
点错了

jetlins, 2023-10-07:
问题已解决，感谢！
如下：
修改 \p1\boot\grub\grub.cfg
[image]

jetlins, 2023-10-07:
esxi主机，编辑-虚拟机选项-高级-配置参数
添加如下参数[image]

wjz304, 2023-10-08:
23.10.0

closed 2023-10-08 by wjz304