RROrg/rr #251: 请问下载里内有个叫 update.zip  有什么作用，文件包没有说明文件，我也不知道怎么用

fk998877, 2023-10-09:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-09:
[image]
[image]
[image]

closed 2023-10-09 by wjz304

liangwenzhong, 2023-10-16:
> [image] [image] [image]

请问一定要按发布的版本更新？还是可以跳版本上传更新？例如23.9.6可以不可以直接上传23.10.3的update.zip？