RROrg/rr #254: 开机有时候网卡顺序会变，双网卡1G+2.5G，怎么固定网卡顺序

xemaco, 2023-10-10:
之前用的6.2.3系统，其中pcie插的2.5G网卡8125是mac1，板载千兆8169是mac2，一直都是不会变化顺序的
后来为了升级7.2，使用arpl自己编译了引导，一切都正常，然而发现个问题，每次启动，网卡顺序都会变动
有时候开机启动两块网卡位置会换，但是不是每次启动都会换，不知道啥问题，比如这次开机，群晖局域网1千兆，局域网2是2.5g，下次再开机，就变成了1号2.5g，2号千兆。。。就很无语。不知道有没有办法固定？不清楚是引导的问题还是系统的问题，请问有人遇到过吗？

还有汇报一个问题，我装1819+，920+，918+都没问题，但是只要装3622xs+就会所有硬盘报错，提示未认证还是不兼容啥的。已经按网上教程关闭了硬盘兼容性检测，也看到arpl里面配置是关闭的，但只要安装3622xs+就不管用。
以上都是dsm7.2系统里遇到的，其他7.x系统没有测试

wjz304, 2023-10-10:
一直是是 23.10.1 吗? 还是 23.9.7 之前的版本升上来的? 
因为23.9.7的修改对这个问题有些许的影响

wjz304, 2023-10-10:
嗯, 有解但是比较麻烦，我还在构思如何实现。

xemaco, 2023-10-10:
好的，坐等大佬解决，我还以为是系统问题，疯狂各种重装搞了两天了。主要是需要固定网卡ip，一旦断电路由器没有点就没了ip，还有一个nas需要依靠这个ip触发ups自动关机。一台ups连两个nas就只能这样了。

wjz304, 2023-10-10:
你的问题很好解决吧，从路由侧设置静态ip就好了

xemaco, 2023-10-10:
没用啊，停电以后路由就断电了，路由器并不在UPS上。我想弄个脚本让NAS2 ping Nas1的IP。比如5分钟运行一次，运行就ping20次，超过一半次数ping不通就自动关机。但是不知道该怎么写脚本，不会搞。。。

wjz304, 2023-10-10:
猜不到你的接法，NAS1 和NAS2直连？ 如果都接在路由上 路由器断电了 谁做数据交换？

xemaco, 2023-10-10:
所以路由断电了路由端设置的静态IP没有作用了，只能靠主机端的设置。我是两个NAS和UPS放在一块。nas1上两个网卡，一个2.5G网卡连了路由器，不过其实也是走了一个2.5G交换机，作为数据主要传输网卡，在群晖里这张卡是dhcp。。另一个千兆网卡和nas2的千兆网卡都接在NAS这边的一个千兆交换机上，这个千兆交换机也接了UPS。路由就算断电了，他俩在同一个交换机的两个网卡都是手动设置的固定ip同一个网关还是能连上。而网卡顺序变动会导致两张卡的ip错乱或者冲突。

xemaco, 2023-10-10:
请问老版本是不是不会有这个问题？降级到老版本引导只能重刷优盘吗？有没有办法进入引到里面敲命令降级

wjz304, 2023-10-10:
也会乱序，但是mac是固定的，就是 eth0 不管是2.5g还是1g的 都是那个mac

wjz304, 2023-10-10:
更新菜单里 手动上传 旧版版的update.zip 就可以了，但是不建议. 因为各个组件之间版本都有依赖关系，除非你都降到对应版本，或者 dd命令也可以，

xemaco, 2023-10-10:
是的，我也发现乱序没乱mac，好像引到里面设置的mac1并不是绑定eth0物理设备.以前用6.2系统，mac1 mac2得跟物理网卡顺序对应不然会影响唤醒，用ARPL我也试着调整mac1和2的地址发现其实没啥用

wjz304, 2023-10-10:
/etc.defaults/sysconfig/network-scripts/ifcfg-ethN
在对应编号的的配置文件里加 HWADDR=00:11:22:33:44:55   （你网卡的mac）
试试吧，

wjz304, 2023-10-10:
> 是的，我也发现乱序没乱mac，好像引到里面设置的mac1并不是绑定eth0物理设备.以前用6.2系统，mac1 mac2得跟物理网卡顺序对应不然会影响唤醒，用ARPL我也试着调整mac1和2的地址发现其实没啥用

23.9.7 就是解除了他俩的绑定关系，其实也是为固定顺序做准备，在不改变原有驱动方案的情况下 唯一的办法就是根据mac进行排序

xemaco, 2023-10-11:
> /etc.defaults/sysconfig/network-scripts/ifcfg-ethN 在对应编号的的配置文件里加 HWADDR=00:11:22:33:44:55 （你网卡的mac） 试试吧，

貌似没有用，我把eth0设定为2.5G网卡mac，eth1设定为千兆网卡mac，在arpl引导里面也是这么现实的，eth0是2.5G网卡，eth1是千兆网卡。但是重启之后群晖系统里面显示局域网2是2.5G，局域网1是千兆

xemaco, 2023-10-11:
> 嗯，等着吧，

嗯，谢谢大佬帮我捣鼓，坐等更新，也不是很紧急的问题

wjz304, 2023-10-11:
基本搞定了，我再测试一下，明天看看吧

wjz304, 2023-10-12:
https://github.com/wjz304/arpl-i18n/releases/tag/23.10.2

xemaco, 2023-10-12:
感谢大佬！

xemaco, 2023-10-12:
直接update可以吗，还是要重写入U盘

wjz304, 2023-10-12:
打开pre后再更就可以

wjz304, 2023-10-12:
？好使吗

xemaco, 2023-10-12:
大佬，我升级之后是不是应该重新编译一下，新功能才能生效

wjz304, 2023-10-12:
嗯
 

Ing
***@***.***

wjz304, 2023-10-12:
插件里要加 sortnetif 啊 
 

Ing
***@***.***

xemaco, 2023-10-12:
选中插件-确定，输入null
或者
输入macs然后输入网卡1，网卡2的MAC是吗

xemaco, 2023-10-12:
测试下来应该是没问题了，有用的，谢谢大佬，非常感谢！

顺便想探讨一下，假如eth0-2.5G地址是001122334455，eth1-千兆的是001122aabbcc
那么如果参数留空，自然是eth0-2.5G被识别成局域网1，假如ip是1.1.1.1，eth1千兆是局域网2，1.1.1.2
但是如果我把参数写成了001122aabbcc,001122334455，那么进入群晖系统的局域网1是2.5G网卡但是获取了1.1.1.2，还是局域网1是千兆网卡获取了1.1.1.2？
再就是网卡顺序插件的参数mac地址和真实地址要一样吗？跟自定义mac地址又是啥关系呢。。。
我说的可能有点多有点乱。。

xemaco, 2023-10-12:
> 你气死我吧

大佬别气，我临时抱佛脚百度了一下学了学。因为之前全都是没加过任何额外功能。实在抱歉。。。其实我也就这两天才开始学着用ARPL😅

xemaco, 2023-10-12:
重启了两三次，设定了静态地址一直没问题。之前只要设定静态IP网卡顺序就乱了，乱了就算了还总是变成一样的IP。我是参数直接留空

xemaco, 2023-10-12:
好的，谢谢大佬。这边又测试了几遍，没有问题了

xemaco, 2023-10-12:
突然发现其实多网卡也不用在添加cmdlines参数来规定网卡数量netif_num和填写mac1   mac2这些参数了吧，反正这些都只是参数，进入系统都是驱动加载的真实硬件信息

wjz304, 2023-10-12:
> 突然发现其实多网卡也不用在添加cmdlines参数来规定网卡数量netif_num和填写mac1 mac2这些参数了吧，反正这些都只是参数，进入系统都是驱动加载的真实硬件信息

23.9.7 之后可以这么理解，之前有关系

xemaco, 2023-10-12:
嗯嗯，怪不得有一次我只定义了一个网卡参数进了系统依然是两个网卡都正常使用，多谢大佬解答

closed 2023-10-15 by wjz304