RROrg/rr #257: BUG - Not auto detect number of lan when define sn/mac

doandat943, 2023-10-11:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:   Intel Core i3 Skylake 6100U
 * NIC:  r8169 (lan integrated), AX88179 (lan adapter)

Arpl version: (You can find it in the update menu.)  
 * arpl:  23.10.1
 * addons:  
 * modules:  
 * lkms:  prod

DSM: 
 * model:  DS918+
 * version:  7.2 Update 1
 
Issue: in older version when i define mac, i can enter mac address one by one, but in lastest version, it become define sn/mac, and i can only show SN and mac1. So i need define netif_num = 2 and mac2 = ...  
    
ogs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-11:
What is the reason why you need to define mac2?

doandat943, 2023-10-11:
> What is the reason why you need to define mac2?

im using it to change mac address of lan adapter

wjz304, 2023-10-11:
I want to know the reason why you want to change your Mac?
Because I had a lot of effort to cancel their binding relationship,

wjz304, 2023-10-11:
[image]

wjz304, 2023-10-11:
netif_num=1 does not affect the startup of eth2 eth3..., but the USB network card may need to be manually enabled after the first installation.

doandat943, 2023-10-11:
> I want to know the reason why you want to change your Mac? Because I had a lot of effort to cancel their binding relationship,

i want using real serial/mac to sign in Synology Account

wjz304, 2023-10-11:
> > I want to know the reason why you want to change your Mac? Because I had a lot of effort to cancel their binding relationship,
> 
> i want using real serial/mac to sign in Synology Account

It only checks mac1,

doandat943, 2023-10-11:
thank for your help

closed 2023-10-11 by doandat943