RROrg/rr #259: 作者大大好，很抱歉用提Issue方式问兼容问题，（QQ群加不去进，已更新新版，自身问题）

yanyisio, 2023-10-12:
作者好：我最近打算重配一台物理黑群，计划使用以下硬件，但不知道能不能使用arpl-i18n来编译
主板：超微MBD-X13SAE-F
CPU：I5-12600K
内存：三星 16G DDR5 4800
HBA卡：SAS 3008/2008
网卡：PCI-E X1 2.5G RTL8125BGASM1806主控

望能回复，感谢！

Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  I5-12600K
 * NIC:  Intel® Ethernet i225LM

Arpl version: (You can find it in the update menu.)  
 * arpl:   23.10.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DVA3221
 * version:   7.2

Issue:  

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-12:
[image]

closed 2023-10-12 by wjz304

yanyisio, 2023-10-12:
> [image]

感谢回复！

12代的CPU核显方案是不多，所以我使用GTX1650 独显。
这样就放心，感谢！