RROrg/rr #260: 用 arpl-i18n安装群晖系统 ，DSM 7.2版本不能降级到6.2版本

gexiaopeng, 2023-10-12:
我的VM虚拟机用多种型号安装了DSM7.0,然后在arpl-i18n配置中设置了降级，并且选择6.2版本编译，重启以后 Synology Assistant 搜索显示可转移，安装DSM_DS1621xs+_25556.pat重启以后，Synology Assistant 搜索仍然是可转移，也就是无法降级到6.2版本。
（PS：按我的操作步骤，DSM从7.2可以降级到7.0）

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

wjz304, 2023-10-12:
6.2 目前不在维护计划

wjz304, 2024-04-19:
RR 24.4.5 后不再支持 6.2

closed 2024-04-19 by wjz304