RROrg/rr #264: 群晖引导开机后登录界面502 bad gateway

Xia-Renjie, 2023-10-13:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  23.10.1
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  918+
 * version:  7.2 69057

Issue:  升级引导之后，打开群晖首页，页面能打开，但是登录框无法加载，调取网页调试信息，显示登录API 502 bad gateway。使用手机浏览器访问就一直显示网页加载中。使用第三方连接群晖显示登录失败。
但是群晖上运行的其他网页服务如Transmission都能正常打开使用。

logs:  nginx的error.log
2023/10/13 21:17:15 [error] 2134#2134: *255 upstream prematurely closed connection while reading response header from upstream, client: 192.168.199.162, server: _, request: "GET /webapi/auth.cgi?account=admin&passwd=xia988425%40&otp_code=&version=4&api=SYNO.API.Auth&method=login&session=FileStation&enable_device_token=no&enable_sync_token=yes&isIframeLogin=yes HTTP/1.1", upstream: "scgi://unix:/run/synoscgi.sock:", host: "192.168.199.127:5000", referrer: "http://192.168.199.127:5000"

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-13:
有nvme？ 加了 nvmecache 插件？

Xia-Renjie, 2023-10-13:
> 有nvme？ 加了 nvmecache 插件？

一开始想加nvme，但是出错之后我又用新版arpl加上默认设置引导了，还是同样的问题

wjz304, 2023-10-13:
nvmecache 插件 加没加过吧

wjz304, 2023-10-13:
开ssh没？

Xia-Renjie, 2023-10-13:
> 开ssh没？

没加nvme，有ssh

wjz304, 2023-10-13:
```
ls /bin/*.sh
```

Xia-Renjie, 2023-10-13:
> ls /bin/*.sh

通过重装DSM解决了，还是感谢你的耐心回答

closed 2023-10-13 by Xia-Renjie