RROrg/rr #266: Problem, wol do not in realtek/r8169

spammoogle, 2023-10-13:
Hello, I just installed version 7.2 of DS3622xs+ with arpl-i18n 23.10.1 and everything has worked correctly. The problem is that the "realtek/r8169" network card, when we turn it off, the light does not blink and does not respond to any magic packet. Wake on LAN do not work.

The MAC used is the real one on the card. BIOS configured correctly.
$ sudo ethtool eth0 
Settings for eth0:
         Supported ports: [TP MII]
         ....
         Speed: 1000Mb/s
         Duplex: Full
         Port: MII
         PHYAD: 0
         Transceiver: internal
         Auto-negotiation: on
         Supports Wake-on: pumbg
         Wake-on: g
         Current message level: 0x00000033 (51)
                                drv probe ifdown ifup
         Link detected: yes

I previously had another version of NAS installed (DS1621+) and it worked correctly.
Otherwise the network card works correctly.

Can anybody help me ?

[image]

wjz304, 2023-10-13:
try add wol addons

spammoogle, 2023-10-13:
I upload addon wol
[image]

But, addon not found into menu
[image]

wjz304, 2023-10-13:
Just return to the main menu and enter again. 
There is a small bug that it has not been refreshed.

wjz304, 2023-10-13:
Oh, sorry, I forgot. This addons will be added by default, so it will not be displayed.
Do you only have one network card?

spammoogle, 2023-10-13:
If I only have one network card.
I have tried putting another one on pcie and canceling the one on the motherboard, but it didn't work either.

wjz304, 2023-10-13:
I haven't test wol of all models, I need to look into that.

wjz304, 2023-10-14:
https://github.com/wjz304/arpl-addons/actions/runs/6515751887

Download this attachment, then manually update it in the update menu, then recompile and try testing

spammoogle, 2023-10-14:
Thank you very much for your help.

I have done it but it hasn't worked either.
We will continue investigating...

wjz304, 2023-10-14:
[image]

wjz304, 2023-10-14:
I installed a new DS3622xs+ on a physical machine. It couldn't wake up without making any modifications. After opening WOL in the control panel, it worked.

spammoogle, 2023-10-14:
OKAY. I have also reinstalled new DS3622xs+ on a physical machine.
The result is the same. Wol don't work. Activating it from the control panel doesn't work either.

Later, in days, I will try a new installation of the DS3622xs+ on a different motherboard. Maybe the problem is in some incompatibility like my current ASUS motherboard.

Thanks again for your interest. Any new news about this issue I will report here.

Greetings from Catalonia

wjz304, 2023-10-14:
OK

wjz304, 2023-10-28:
https://github.com/wjz304/rr-modules/releases/tag/23.10.1

更新了 r8168/r8169 的驱动，可以更新试试

paidong, 2023-10-29:
> https://github.com/wjz304/rr-modules/releases/tag/23.10.1
> 
> 更新了 r8168/r8169 的驱动，可以更新试试

我在物理机上用23.10.4版本也发现这个问题，启动DSM后用ssh登陆，用ethtool查wol为g，但关机后WOL无法开机。如果先启动到配置模式，这是用ethtool查wol为d，先运行ethtools -s eth0 wol g，再运行menu.sh，最后选择“启动”，这样启动后关机WOL就正常（这样操作后关机，通过路由可以看到物理机有10M的连接，否则会显示未连接），请问是否是跟先启动arpl再启动DSM有关？

wjz304, 2023-10-29:
> > https://github.com/wjz304/rr-modules/releases/tag/23.10.1
> > 更新了 r8168/r8169 的驱动，可以更新试试
> 
> 我在物理机上用23.10.4版本也发现这个问题，启动DSM后用ssh登陆，用ethtool查wol为g，但关机后WOL无法开机。如果先启动到配置模式，这是用ethtool查wol为d，先运行ethtools -s eth0 wol g，再运行menu.sh，最后选择“启动”，这样启动后关机WOL就正常（这样操作后关机，通过路由可以看到物理机有10M的连接，否则会显示未连接），请问是否是跟先启动arpl再启动DSM有关？

我23.9.7之后我好像误删除了一段代码，我加回去 下个版本试试吧

paidong, 2023-10-30:
谢谢！期待下个版本

wjz304, 2023-10-30:
> 谢谢！期待下个版本

https://github.com/wjz304/rr/releases/tag/23.10.7

paidong, 2023-10-30:
这么快！效率太高了，等周末回去试试，应该没问题了

spammoogle, 2023-11-04:
Hi, 
I have installed the latest version xxxx.
WOL now works!!! Thank you very much.
I imagine it was driver problems?

Another thing:
When we boot the USB and want to do the Build. Some writing errors occur.
Send photo
[image]

wjz304, 2023-11-04:
> Hi, I have installed the latest version 23.11.1 WOL now works!!! Thank you very much. I imagine it was driver problems?
> 
> Another thing: When we boot the USB and want to do the Build. Some writing errors occur. Send photo [image]

It seems that the p1 partition of the USB flash drive is in a read-only state. It should not be read-only.
What software was used to create the USB flash drive?

spammoogle, 2023-11-04:
BalenaEtcher, but it is true that I did not delete it before doing the Flash.
I'll try to create it again.

wjz304, 2023-11-05:
其他问题单独创建issue吧

closed 2023-11-05 by wjz304