RROrg/rr #268: arpl能添加支持群晖FS3017的型号吗cpu是Intel Xeon E5-2620 v3 x 2

Joyist2021, 2023-10-16:
刚好我用的是dell R73xd cpu是Intel Xeon E5-2697 v3双路

 [FS3017群晖官网查到的信息](https://kb.synology.cn/zh-cn/DSM/tutorial/What_kind_of_CPU_does_my_NAS_have)

FS3017	Intel Xeon E5-2620 v3 x 2	6	12	✓	Grantley	DDR4 ECC RDIMM 64GB

Joyist2021, 2023-10-16:
型号:  SA6400
版本:  7.2(69057u1)
LKM:   prod
DMI:   QEMU Standard PC (Q35 + ICH9, 2009), BIOS unknown 2/2/2022
CPU:   Intel(R) Xeon(R) CPU E5-2697 v3 @ 2.60GHz
MEM:   7940 MB
Cmdline:
withefi earlyprintk syno_hw_version=SA6400 console=ttyS0,115200n8 netif_num=1 synoboot2 pid=0x0001 earlycon=uart8250,io,0x3f8,115200n8 mac1=001132d744e0 sn=2270UMRBY59CW SMBusHddDynamicPower=1 vid=0x46f4 loglevel=15 skip_vender_mac_interfaces=0,1,2,3,4,5,6,7 panic=5 vender_format_version=2 log_buf_len=32M root=/dev/md0 syno_ttyS1=serial,0x2f8 syno_ttyS0=serial,0x3f8
检测到 1 个网卡.
检查连接.
eth0 connected.
获取 IP.(仅供参考)
eth0(virtio_net): 在浏览器中访问 http://192.168.1.208:5000 链接 DSM.

虚拟机中完美启动

wjz304, 2023-10-16:
kernel v3 的型号不想加了

ikapc, 2023-10-25:
> kernel v3 的型号不想加了

大佬，可以加一个DS218+的型号吗

wjz304, 2023-11-04:
> > kernel v3 的型号不想加了
> 
> 大佬，可以加一个DS218+的型号吗

太老了，没合适的理由不加

closed 2023-11-04 by wjz304