RROrg/rr #270: 非DT机型，怎么支持热拔插呢？

naoki66, 2023-10-17:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  pve8.02
 * CPU:  12700
 * NIC:  x540

Arpl version: (You can find it in the update menu.)  
 * arpl:  [23.10.3](https://github.com/wjz304/arpl-i18n/releases/tag/23.10.3)
 * addons:   
  misc: ""
  acpid: ""
  reboottoarpl: ""
  cpuinfo: ""
  setrootpw: ""
  codecpatch: ""
  addincards: ""
  sortnetif: 0011328f6263,00e0ed01f8ec,00e0ed01f8ed
  hdddb: ""
  storagepanel: TOWER_8_Bay 1X2
  nvmevolume: ""
  dbgutils: ""
  powersched: ""

 * modules:  
  support_disk_compatibility: no
  support_memory_compatibility: no
  support_led_brightness_adjustment: no
  support_bde_internal_10g: no
  support_oob_ctl: no
  support_syno_hybrid_raid: ""
  supportraidgroup: ""
  maxlanport: "8"
  netif_seq: 0 1 2 3 4 5 6 7
  HddHotplug: "1"
  internalportcfg: "0x0000000FFF"
  esataportcfg: "0x0000000000"
  usbportcfg: "0x000003F000"
  maxdisks: "18"
addons:
  misc: ""
  acpid: ""
  reboottoarpl: ""
  cpuinfo: ""
  setrootpw: ""
  codecpatch: ""
  addincards: ""
  sortnetif: 0011328f6263,00e0ed01f8ec,00e0ed01f8ed
  hdddb: ""
  storagepanel: TOWER_8_Bay 1X2
  nvmevolume: ""
  dbgutils: ""
  powersched: ""
modules:
  8139cp: ""
  acpi-cpufreq: ""
  aic94xx: ""
  alx: ""
  aqc111: ""
  asix: ""
  atkbd: ""
  atl1c: ""
  atl1e: ""
  atlantic: ""
  auxiliary: ""
  ax88179_178a: ""
  b44: ""
  be2net: ""
  bitblit: ""
  bnx2: ""
  bnx2x: ""
  bnxt_en: ""
  button: ""
  cdc_ncm: ""
  cdrom: ""
  cfbcopyarea: ""
  cfbfillrect: ""
  cfbimgblt: ""
  ch: ""
  cpufreq_conservative: ""
  cpufreq_ondemand: ""
  cpufreq_performance: ""
  crc-itu-t: ""
  cxgb3: ""
  cxgb4: ""
  cxgb4vf: ""
  cxgb: ""
  dvb-core: ""
  dvb-usb-dib0700: ""
  dvb-usb-dvbsky: ""
  dvb-usb: ""
  dvb_usb_v2: ""
  e1000e: ""
  e1000: ""
  efifb: ""
  fbcon: ""
  fbdev: ""
  fb: ""
  fb_sys_fops: ""
  font: ""
  hpsa: ""
  hv_netvsc: ""
  i40e: ""
  i8042: ""
  iavf: ""
  ib_core: ""
  ice: ""
  igb: ""
  igbvf: ""
  igc: ""
  intel_auxiliary: ""
  ipv6: ""
  irqbypass: ""
  iscsi_tcp: ""
  it87: ""
  ixgbe: ""
  ixgbevf: ""
  ixgb: ""
  jc42: ""
  jme: ""
  kvm-amd: ""
  kvm-intel: ""
  kvm: ""
  libfc: ""
  libiscsi: ""
  libiscsi_tcp: ""
  libphy: ""
  m88ds3103: ""
  mdio: ""
  megaraid_mbox: ""
  megaraid_mm: ""
  megaraid_sas: ""
  mii: ""
  mlx4_core: ""
  mlx4_en: ""
  mlx4_ib: ""
  mlx5_core: ""
  mlx5_ib: ""
  mlx_compat: ""
  mlxsw_core: ""
  mlxsw_pci: ""
  mmc_block: ""
  mmc_core: ""
  mpt3sas: ""
  mptbase: ""
  mptctl: ""
  mptsas: ""
  mptscsih: ""
  mptspi: ""
  mtk-sd: ""
  nct6775: ""
  netxen_nic: ""
  processor: ""
  qcaspi: ""
  qede: ""
  qed: ""
  qla3xxx: ""
  qlcnic: ""
  qlge: ""
  r8125: ""
  r8152: ""
  r8153_ecm: ""
  r8169: ""
  rc-core: ""
  realtek: ""
  scsi_transport_fc: ""
  scsi_transport_spi: ""
  sdhci: ""
  sg: ""
  si2157: ""
  si2168: ""
  skge: ""
  sky2: ""
  softcursor: ""
  sp2: ""
  sr_mod: ""
  syscopyarea: ""
  sysfillrect: ""
  sysimgblt: ""
  tg3: ""
  thermal: ""
  ts2020: ""
  uhci-hcd: ""
  usbnet: ""
  ushc: ""
  vesafb: ""
  vga16fb: ""
  vgastate: ""
  via-sdmmc: ""
  virtio_input: ""
  virtio: ""
  virtio_mmio: ""
  virtio_net: ""
  virtio_pci: ""
  virtio_ring: ""
  virtio_scsi: ""
  vmw_pvscsi: ""
  vmxnet3: ""
  vub300: "" 
 * lkms:  

DSM: 
 * model:  DS3622xs
 * version:  7.2  69057

Issue:  
手动在synoinfo下添加
  HddHotplug: "1"
但似乎并不生效。

另外 storagepanel: TOWER_8_Bay 1X2
实际启动后仅有8个SATA，看不懂nvme，但是nvme可以识别，仅能用作缓存，不能作为存储空间

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-17:
cmdline，bios设置，很多相关，自行测试

closed 2023-10-17 by wjz304

naoki66, 2023-10-19:
cmdline，添加 HddHotplug: "1"
bios设置，开启Hotplug支持。
DSM的日志也能看到很多相关内容，但是死活起不来，日志COMRESET failed (errno=-16)。

[   93.955847] ata15: device plugged sstatus 0x133
[   93.956191] ata15: exception Emask 0x10 SAct 0x0 SErr 0x4050000 action 0xe frozen
[   93.956666] ata15: irq_stat 0x00400040, connection status changed
[   93.957050] ata15: SError: { PHYRdyChg CommWake DevExch }
[   93.957404] ata15: hard resetting link
[   96.819991] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P     
[   96.820682] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[   99.730223] ata15: link is slow to respond, please be patient (ready=0)
[  102.984864] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P     
[  102.985619] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[  103.964174] ata15: COMRESET failed (errno=-16)
[  103.964494] ata15: COMRESET fail, set COMRESET fail flag
[  103.964988] ata15: hard resetting link
[  109.153931] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P     
[  109.154631] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[  109.737177] ata15: link is slow to respond, please be patient (ready=0)
[  110.094173] ata15: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
[  110.094621] ata15: link reset sucessfully clear error flags
[  110.648469] ata15.00: disabling queued TRIM support
[  110.648818] ata15.00: ATA-9: WDC WD140EMFZ-11A0XXX, 81.00A81, max UDMA/133
[  110.649251] ata15.00: 27344764928 sectors, multi 0: LBA48 NCQ (depth 31/32), AA
[  110.649798] ata15.00: SN:Z2GHXXXT
[  110.758196] ata15.00: disabling queued TRIM support
[  110.758504] ata15.00: configured for UDMA/133
[  110.758775] ata15.00: Write Cache is enabled
[  110.759068] ata15: EH complete
[  110.759532] scsi 14:0:0:0: Direct-Access     WDC      WD140EMFZ-11A0XXX        0A81 PQ: 0 ANSI: 5
[  110.760243] <redpill/scsi_notifier.c:66> Probing SCSI device using sd_probe_shim
[  110.760711] <redpill/scsi_notifier.c:86> Triggering SCSI_EVT_DEV_PROBING notifications
[  110.761323] <redpill/scsi_notifier.c:96> Calling original sd_probe()
[  110.761749] <redpill/scsi_notifier.c:100> Triggering SCSI_EVT_DEV_PROBED notifications - sd_probe() exit=0
[  110.761765] sd 14:0:0:0: [sdn] 27344764928 512-byte logical blocks: (14.0 TB/12.7 TiB)
[  110.761765] sd 14:0:0:0: [sdn] 4096-byte physical blocks
[  110.761799] sd 14:0:0:0: [sdn] Write Protect is off
[  110.761800] sd 14:0:0:0: [sdn] Mode Sense: 00 3a 00 10
[  110.761813] sd 14:0:0:0: [sdn] Write cache: enabled, read cache: enabled, supports DPO and FUA
[  110.764539] sd 14:0:0:0: Attached scsi generic sg1 type 0
[  110.809445]  sdn: sdn1 sdn2 sdn3 sdn4 sdn5
[  110.810070] sd 14:0:0:0: [sdn] Attached SCSI disk
------------------------------------------------------------
另外就是日志一直在跑
[  150.439276] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[  156.608810] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P     
[  156.609608] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[  162.755567] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P     
[  162.756321] <redpill/smart_shim.c:355> ATA_CMD_ID_ATA confirmed SMART support - noop
[  168.906174] <redpill/scsi_disk_serial.c:17> scsi host no: 0, device id: 0, name: sda, serial: S1SUNSAG145560P