RROrg/rr #271: BUG

pickleqin, 2023-10-17:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  Intel 1581
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  23.10.3

DSM: 
 * model:  DS3622xs
 * version:   7.2.1

Issue:  

对于使用HBA卡lsi  3008(接sata硬盘)无法调整硬盘顺序，总是在固定位置，已经设置了SataPortMap和DsikIdxMap。这是引导本身就不支持还是一个问题。

wjz304, 2023-10-17:
没懂具体问题

pickleqin, 2023-10-23:
Lis 3008 HBA 直通卡在pve系统通过硬件直通给群晖，接入的硬盘无法通过SataPortMap和DsikIdxMap调整磁盘在群晖系统的磁盘序号。

OldLi1997, 2023-10-27:
> Lis 3008 HBA 直通卡在pve系统通过硬件直通给群晖，接入的硬盘无法通过SataPortMap和DsikIdxMap调整磁盘在群晖系统的磁盘序号。

使用ds918或者dva3221非dt机型可以调

wjz304, 2023-10-28:
https://github.com/wjz304/rr/issues/258

wjz304, 2023-10-30:
https://github.com/wjz304/rr/issues/291

wjz304, 2023-11-12:
This problem will not be solved,
The issue is closed

closed 2023-11-12 by wjz304