RROrg/rr #272: 你好，请问一下，这个在12代或者13代上大小核CPU上能正常运行吗？

atiko00, 2023-10-17:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  
 * version:  

Issue:  

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-18:
等我有钱了， 买个13代，试试告诉你

a8223492, 2023-10-18:
@wjz304  14代出了，可以买14代了！！

wjz304, 2023-10-18:
我连11代都买不起，还14代

atiko00, 2023-10-19:
好吧，我用一台正在跑服务的12900T ES不显的机器试了一下，能跑，但是CPU识别有問題，只认出来4核8线程……

wjz304, 2023-10-19:
> 好吧，我用一台正在跑服务的12900T ES不显的机器试了一下，能跑，但是CPU识别有問題，只认出来4核8线程…… 然后我在bios里把小核关闭，重装了一下，还是一样，只能认出来4核8线程，

不同型号不一样

atiko00, 2023-10-19:
> > 好吧，我用一台正在跑服务的12900T ES不显的机器试了一下，能跑，但是CPU识别有問題，只认出来4核8线程…… 然后我在bios里把小核关闭，重装了一下，还是一样，只能认出来4核8线程，
> 
> 不同型号不一样

嗯，我是打了那个cpuinfo的补丁，不同型号的引导识别的频率都不一样，DS918+识别的频率是801mHz，DS920+识别的频率是5.6gHz，不知道什么原因，可能是我用的是ES不显的CPU，

wjz304, 2023-10-19:
cpuinfo的补丁只是改显示，跟系统没什么关系

jim3ma, 2023-11-02:
SA6400 可以识别所有核心，只是大小核调度可能还不完美

wjz304, 2023-11-05:
https://github.com/wjz304/rr/issues/321

closed 2024-03-30 by wjz304