RROrg/rr #275: BUG高负载内核崩溃，出现异常关机

PAINCLOWN, 2023-10-18:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  
 * NIC:  

Arpl version: (You can find it in the update menu.)  
 * arpl:  最新
 * addons:  最新
 * modules:  最新
 * lkms:  最新
[image]

DSM: 
 * model:  ds918+ ds920+
 * version:  7.2.1

Issue:  
使用PVE8.0.4 虚拟机跑dsm7.2.1 会打印如下日志
不管怎么重装 只要负载高，就会随机kp ，群晖显示就是异常关机
[image]

logs:  
[ 5108.689573] IPVS: Creating netns size=2104 id=25
[ 5109.229900] unable to execute userspace code (SMEP?) (uid: 0)
[ 5109.231021] BUG: unable to handle kernel paging request at ffffffff81068a80
[ 5109.231021] IP: [<ffffffff81068a80>] search_exception_tables+0x0/0x40
[ 5109.231021] PGD 37ca0d067 PUD 37ca0e063 PMD 37c2001e1 
[ 5109.231021] Thread overran stack, or stack corrupted
[ 5109.231021] Oops: 0019 [#1] SMP 
[ 5109.231021] Modules linked in: xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables xt_REDIRECT xt_nat nf_nat_redirect xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nfnetlink xfrm_user xfrm_algo fuse xt_addrtype ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_filter ip_tables x_tables vfat fat vhost_scsi(O) vhost(O) udf isofs tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) synoacl_vfs(O) 8021q snd_hda_intel snd_hda_codec snd_hda_core snd_pcm snd_timer snd soundcore uas usb_storage leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper geminilake_synobios(O) usblp r8168(O) dm_crypt sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave zram cpufreq_performance cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel loop btrfs raid6_pq xor zstd_compress zstd_decompress ecryptfs xxhash sha256_generic synorbd(O) synofsbd(O) etxhci_hcd virtio_net(OE) hid_generic kvm_intel kvm irqbypass sr_mod(OE) cdrom(OE) virtio_pci(OE) virtio(OE) virtio_ring(OE) i915 video backlight i2c_algo_bit iosf_mbi drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm drm_panel_orientation_quirks cfbfillrect cfbimgblt cfbcopyarea fb fbdev(OE) rtc_cmos(OE) button processor usbhid hid xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common [last unloaded: xt_LOG]
[ 5109.231021] CPU: 1 PID: 13751 Comm: PM2 v5.3.0: God Tainted: P           OE   4.4.302+ #69057
[ 5109.231021] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
[ 5109.231021] task: ffff88020ef30000 ti: ffff88020ed5c000 task.ti: ffff88020ed5c000
[ 5109.231021] RIP: 0010:[<ffffffff81068a80>]  [<ffffffff81068a80>] search_exception_tables+0x0/0x40
[ 5109.231021] RSP: 0000:ffff88020ed43cc0  EFLAGS: 00010082
[ 5109.231021] RAX: 0000000000000000 RBX: ffff88020ed43df8 RCX: 000000000000000b
[ 5109.231021] RDX: ffffffff81068a80 RSI: 0000000000000019 RDI: ffffffff81068a80
[ 5109.231021] RBP: ffff88020ed43cd0 R08: 0000000000030001 R09: 00007ffca13f14d8
[ 5109.231021] R10: 00007ffca13f14c8 R11: 00007ffca13f15d0 R12: 0000000000000019
[ 5109.231021] R13: 000000000000000b R14: ffffffff81068a80 R15: ffff88020ef30000
[ 5109.231021] FS:  00007f8fa9cf7c18(0000) GS:ffff88037dc80000(0000) knlGS:0000000000000000
[ 5109.231021] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[ 5109.231021] CR2: ffffffff81068a80 CR3: 00000001ec9aa000 CR4: 00000000001606f0
[ 5109.231021] DR0: 0000000000000000 DR1: 0000000000000000 DR2: 0000000000000000
[ 5109.231021] DR3: 0000000000000000 DR6: 00000000fffe0ff0 DR7: 0000000000000400
[ 5109.231021] Stack:
[ 5109.231021]  ffffffff8103e964 ffff88020ed43df8 ffff88020ed43d28 ffffffff8103d53f
[ 5109.231021]  0000000000000000 0000000000000000 00007fcd00030001 00007fcd79265000
[ 5109.231021]  0000000000000019 ffffffff81068a80 ffff88020ef30000 ffff88020ed43df8
[ 5109.231021] Call Trace:
[ 5109.231021]  <UNK> 
[ 5109.231021] Code: 24 48 39 d8 75 e0 4c 89 f7 e8 4d bb 51 00 48 89 d8 5b 41 5c 41 5d 41 5e 5d c3 31 db eb e8 66 2e 0f 1f 84 00 00 00 00 00 0f 1f 00 <48> be f0 a8 58 81 ff ff ff ff 55 48 89 fa 48 89 e5 53 48 89 fb 
[ 5109.231021] RIP  [<ffffffff81068a80>] search_exception_tables+0x0/0x40
[ 5109.231021]  RSP <ffff88020ed43cc0>
[ 5109.231021] CR2: ffffffff81068a80
[ 5109.231021] ---[ end trace b0e76ae4bb55c65c ]---
[ 5109.831758] BUG: unable to handle kernel paging request at 000000000001ae10
[ 5109.832302] IP: [<ffffffff81089a4d>] queued_spin_lock_slowpath+0xbd/0x160
[ 5109.832302] PGD 0 
[ 5109.832302] Oops: 0002 [#2] SMP 
[ 5109.832302] Modules linked in: xt_ipvs ip_vs_rr ip_vs xt_mark iptable_mangle br_netfilter bridge stp aufs macvlan veth xt_conntrack nf_conntrack_ipv6 nf_defrag_ipv6 ip6table_filter ip6_tables xt_REDIRECT xt_nat nf_nat_redirect xt_recent xt_iprange xt_limit xt_state xt_tcpudp xt_multiport xt_LOG nfnetlink xfrm_user xfrm_algo fuse xt_addrtype ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat nf_nat_ipv4 nf_nat nf_conntrack_ipv4 nf_defrag_ipv4 nf_conntrack iptable_filter ip_tables x_tables vfat fat vhost_scsi(O) vhost(O) udf isofs tcm_loop(O) iscsi_target_mod(O) target_core_user(O) target_core_ep(O) target_core_multi_file(O) target_core_file(O) target_core_iblock(O) target_core_mod(O) syno_extent_pool(PO) rodsp_ep(O) synoacl_vfs(O) 8021q snd_hda_intel snd_hda_codec snd_hda_core snd_pcm snd_timer snd soundcore uas usb_storage leds_lp3943 aesni_intel glue_helper lrw gf128mul ablk_helper geminilake_synobios(O) usblp r8168(O) dm_crypt sg dm_snapshot dm_bufio crc_itu_t crc_ccitt psnap p8022 llc hfsplus md4 hmac sit tunnel4 ipv6 flashcache_syno(O) flashcache(O) syno_flashcache_control(O) dm_mod arc4 crc32c_intel cryptd ecb aes_x86_64 authenc des_generic ansi_cprng cts md5 cbc cpufreq_powersave zram cpufreq_performance cpufreq_stats vxlan ip6_udp_tunnel udp_tunnel ip_tunnel loop btrfs raid6_pq xor zstd_compress zstd_decompress ecryptfs xxhash sha256_generic synorbd(O) synofsbd(O) etxhci_hcd virtio_net(OE) hid_generic kvm_intel kvm irqbypass sr_mod(OE) cdrom(OE) virtio_pci(OE) virtio(OE) virtio_ring(OE) i915 video backlight i2c_algo_bit iosf_mbi drm_kms_helper syscopyarea sysfillrect sysimgblt fb_sys_fops drm drm_panel_orientation_quirks cfbfillrect cfbimgblt cfbcopyarea fb fbdev(OE) rtc_cmos(OE) button processor usbhid hid xhci_pci xhci_hcd uhci_hcd ehci_pci ehci_hcd usbcore usb_common [last unloaded: xt_LOG]
[ 5110.010964] CPU: 0 PID: 16404 Comm: pip3 Tainted: P      D    OE   4.4.302+ #69057
[ 5110.010964] Hardware name: QEMU Standard PC (Q35 + ICH9, 2009), BIOS 3.20230228-4 06/06/2023
[ 5110.010964] task: ffff880100536800 ti: ffff880101340000 task.ti: ffff880101340000
[ 5110.010964] RIP: 0010:[<ffffffff81089a4d>]  [<ffffffff81089a4d>] queued_spin_lock_slowpath+0xbd/0x160
[ 5110.010964] RSP: 0018:ffff880101343a28  EFLAGS: 00010002
[ 5110.010964] RAX: 000000000001ae10 RBX: ffff88020ed54018 RCX: 00000000000003b4
[ 5110.010964] RDX: ffff88020ed54030 RSI: ffff88037dc1ae00 RDI: 0000000000040000
[ 5110.010964] RBP: ffff880101343a30 R08: 0000000000000001 R09: 0000000000000000
[ 5110.010964] R10: ffff8802b30836c0 R11: 0000000000000005 R12: 0000000000000000
[... 72 lines cut ...]
[ 5110.125204]  [<ffffffff8107ff17>] rebalance_domains+0x227/0x2e0
[ 5110.125204]  [<ffffffff81080112>] run_rebalance_domains+0x142/0x1d0
[ 5110.125204]  [<ffffffff81050e2d>] __do_softirq+0xdd/0x330
[ 5110.125204]  [<ffffffff810511b3>] irq_exit+0x73/0xa0
[ 5110.125204]  [<ffffffff81031bf3>] smp_apic_timer_interrupt+0x43/0x60
[ 5110.125204]  [<ffffffff8158585b>] apic_timer_interrupt+0x9b/0xa0
[ 5110.125204]  <EOI> 
[ 5110.125204]  [<ffffffff8100bebc>] ? default_idle+0x2c/0xf0
[ 5110.125204]  [<ffffffff8100cc40>] arch_cpu_idle+0x10/0x20
[ 5110.125204]  [<ffffffff8108564b>] default_idle_call+0x2b/0x30
[ 5110.125204]  [<ffffffff8108592b>] cpu_startup_entry+0x2db/0x320
[ 5110.125204]  [<ffffffff8102fad3>] start_secondary+0x133/0x140
[ 5110.125204] Code: ff ff c9 c3 66 66 2e 0f 1f 84 00 00 00 00 00 66 90 48 8b 47 08 48 63 48 10 48 8b 87 40 07 00 00 48 8b 50 60 48 8b 82 b8 00 00 00 <48> 03 04 cd 20 4a 8d 81 48 01 30 48 8b 52 48 48 85 d2 75 e5 c3 
[ 5110.125204] RIP  [<ffffffff81088dfa>] cpuacct_charge+0x1a/0x30
[ 5110.125204]  RSP <ffff88037dd83cc8>
[ 5110.125204] CR2: fffffffff9254a20
[ 5110.125204] ---[ end trace b0e76ae4bb55c65e ]---
[ 5110.125204] Kernel panic - not syncing: Fatal exception in interrupt
[ 5112.114858] Shutting down cpus with NMI
[ 5112.114858] Kernel Offset: disabled
[ 5112.114858] Rebooting in 3 seconds..
[ 5112.114858] ACPI MEMORY or I/O RESET_REG.

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-18:
vm 的配置看一下

PAINCLOWN, 2023-10-18:
args: -serial file:/var/log/104-console.log
bios: ovmf
boot: order=sata0
cores: 4
cpu: host
hostpci0: 0000:00:02,pcie=1
hostpci1: 0000:01:00,pcie=1
ide2: none,media=cdrom
machine: q35
memory: 12288
meta: creation-qemu=8.0.2,ctime=1697425592
name: DSMn
net0: virtio=00:11:32:02:7F:AE,bridge=vmbr0
numa: 0
ostype: l26
sata0: local:104/vm-104-disk-0.raw,size=1G
scsihw: virtio-scsi-single
smbios1: uuid=579e030e-8b07-497d-9dd0-368db0f3535e
sockets: 1
vmgenid: 06497d83-54e9-4ea1-b439-0eed783cae3b

> vm 的配置看一下

PAINCLOWN, 2023-10-18:
> vm 的配置看一下

[image]
[image]
[image]

wray-lee, 2023-10-20:
同样的情况

wjz304, 2023-10-20:
机器总内存多少？

PAINCLOWN, 2023-10-20:
> 机器总内存多少？

32G，分配给虚拟机后物理机还有几个G的富余

wjz304, 2023-10-20:
看日志似乎和 aufs.ko 有关，
卸载它看看吧，突然卸载 可能会影响某些功能，如果卸载后死机等就直接重启。
```
lsmod | grep aufs   #查看是否加载 aufs 驱动
rmmod aufs   # 卸载 aufs 驱动
```

wjz304, 2023-10-20:
或者试试其他型号

anorl, 2023-10-20:
我有同样的问题，尝试了DS918+ DS920+，都一样，经常显示NAS异常关机，但在PVE中显示又是正常的。
一开始怀疑是我的直通的硬盘的问题，我换了一块直通硬盘，问题也依旧。
我的黑群配置是2核CPU，也试过4核，同样问题。
内存是2G-8G,也试过4G，同样问题。

wjz304, 2023-10-20:
另外 可以尝试更新微码试试，反正应该没啥坏处

PAINCLOWN, 2023-10-20:
> rmmod aufs

 lsmod | grep aufs 没有这个驱动
root@DSM:~# rmmod aufs
rmmod: ERROR: Module aufs is not currently loaded

微码的话 我尝试更新一下看看 感觉应该不是这个原因

zhangbaiquan, 2023-10-24:
同遇到，esxi8 上 N5105 黑裙4C12G 最初直接使用在arpl-i18n-23.9.7 遇到，后升级到 arpl-i18n-23.10.3，仍然出现，不太会找楼主那种日志，遇到了也反馈下。

alipig, 2023-10-24:
exsi 7 也是这版本安装7.2.1 出现这问题
试过几次，发现启动玩，不访问web是不会异常关机的。我用ssh都可以正常使用，只要不访问web其他手机APP都能正常使用

一访问web就直接出现群晖nginx那错误页面，然后就异常关机

wjz304, 2023-10-24:
你们的高负载是不是都是用qb测试的？

alipig, 2023-10-28:
我 exsi里面看CPU占用率，升级7.2.1后，启动时CPU占用非常高，等启动完，不访问 群晖登录界面就可以用ssh登录一直使用，无论什么时候访问web登录界面，就出现404错误，然后群晖就异常关机了

Promix953, 2023-10-28:
> 你们的高负载是不是都是用qb测试的？

一样的问题，在qb中下大包，频繁异常重启
PVE8，内核最新
N5105（微码和BIOS最新）
qb 4.6.0
内存16G，群晖分8G

试过docker资源限制无效，增减虚拟机分配的内存无效，换硅脂提高散热无效，清理插拔内存和sata数据线无效，跳过背板直插无效
现在尝试qb只绑定IPv4，正在测试能坚持多久
之前PVE8 + qb 4.5.5貌似没遇见这问题，只绑IPv4再不行的话下步准备降级qb再试试

PAINCLOWN, 2023-10-30:
> 另外 可以尝试更新微码试试，反正应该没啥坏处

目前已经稳定运行两天，随机kp的情况暂时没有出现，只能参考一下

所做操作：
1.更新pve下的微码
2.dsm从shr（raid6）转为raid10 （非常麻烦和危险）

wray-lee, 2023-10-30:
> > 另外 可以尝试更新微码试试，反正应该没啥坏处
> 
> 目前已经稳定运行两天，随机kp的情况暂时没有出现，只能参考一下
> 
> 所做操作： 1.更新pve下的微码 2.dsm从shr（raid6）转为raid10 （非常麻烦和危险）

看下能否稳定运行一周，之前更新过微码好像也是稳定几天之后还是kp,但没更改过raid类型

PAINCLOWN, 2023-11-02:
> > > 另外 可以尝试更新微码试试，反正应该没啥坏处
> > 
> > 
> > 目前已经稳定运行两天，随机kp的情况暂时没有出现，只能参考一下
> > 所做操作： 1.更新pve下的微码 2.dsm从shr（raid6）转为raid10 （非常麻烦和危险）
> 
> 看下能否稳定运行一周，之前更新过微码好像也是稳定几天之后还是kp,但没更改过raid类型

目前已经稳定运行五天，没有再出现问题

wray-lee, 2023-11-02:
```bash
2023-11-02T10:45:22.982Z In(05) vcpu-1 - AHCI-VMM: sata0:1: Halted the port due to command abort.
2023-11-02T10:45:22.982Z In(05) vcpu-0 - AHCI-VMM:Aborting the command: 0xb0
2023-11-02T10:45:22.982Z In(05) vcpu-0 - AHCI-VMM: sata0:2: Halted the port due to command abort.
2023-11-02T10:46:19.317Z Wa(03) vcpu-0 - MONITOR PANIC: vcpu-1:VMM fault 14: src=MONITOR rip=0xfffffffffc04e146 regs=0xfffffffffc407e80 LBR stack=0xfffffffffc407b20
2023-11-02T10:46:19.317Z In(05) vcpu-2 - Exiting vcpu-2
2023-11-02T10:46:19.317Z In(05) vcpu-3 - Exiting vcpu-3
2023-11-02T10:46:19.317Z In(05) vcpu-1 - Exiting vcpu-1
2023-11-02T10:46:19.317Z In(05) vcpu-0 - Core dump with build build-21930508
2023-11-02T10:46:19.319Z In(05) vcpu-0 - Writing monitor `vmmcores.gz`
2023-11-02T10:46:19.319Z Wa(03) vcpu-0 - Dumping core for vcpu-0
2023-11-02T10:46:19.319Z In(05) vcpu-0 - VMK Stack for vcpu 0 is at 0x453890c13000
2023-11-02T10:46:19.319Z In(05) vcpu-0 - Beginning monitor coredump
2023-11-02T10:46:19.420Z Wa(03) mks - Panic in progress... ungrabbing
2023-11-02T10:46:19.420Z In(05) mks - MKS: Release starting (Panic)
2023-11-02T10:46:19.420Z In(05) mks - MKS: Release finished (Panic)
2023-11-02T10:46:19.612Z In(05) vcpu-0 - End monitor coredump
2023-11-02T10:46:19.612Z Wa(03) vcpu-0 - Dumping core for vcpu-1
2023-11-02T10:46:19.612Z In(05) vcpu-0 - VMK Stack for vcpu 1 is at 0x453892f93000
2023-11-02T10:46:19.612Z In(05) vcpu-0 - Beginning monitor coredump
2023-11-02T10:46:19.901Z In(05) vcpu-0 - End monitor coredump
2023-11-02T10:46:19.901Z Wa(03) vcpu-0 - Dumping core for vcpu-2
2023-11-02T10:46:19.901Z In(05) vcpu-0 - VMK Stack for vcpu 2 is at 0x45388a593000
2023-11-02T10:46:19.901Z In(05) vcpu-0 - Beginning monitor coredump
2023-11-02T10:46:20.195Z In(05) vcpu-0 - End monitor coredump
2023-11-02T10:46:20.196Z Wa(03) vcpu-0 - Dumping core for vcpu-3
2023-11-02T10:46:20.196Z In(05) vcpu-0 - VMK Stack for vcpu 3 is at 0x453891113000
2023-11-02T10:46:20.196Z In(05) vcpu-0 - Beginning monitor coredump
2023-11-02T10:46:20.483Z In(05) vcpu-0 - End monitor coredump
2023-11-02T10:46:21.477Z Wa(03) vcpu-0 - A core file is available in "/vmfs/volumes/65266f4c-d33be7d8-deea-7085c237dd5a/Nas/vmx-zdump.002"
2023-11-02T10:46:21.477Z In(05) vcpu-0 - Msg_Post: Error
2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.log.error.unrecoverable] VMware ESX unrecoverable error: (vcpu-0)
2023-11-02T10:46:21.477Z In(05)+ vcpu-0 - vcpu-1:VMM fault 14: src=MONITOR rip=0xfffffffffc04e146 regs=0xfffffffffc407e80 LBR stack=0xfffffffffc407b20
2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.haveLog] A log file is available in "/vmfs/volumes/65266f4c-d33be7d8-deea-7085c237dd5a/Nas/vmware.log".  
2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.requestSupport.withoutLog] You can request support.  
2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.requestSupport.vmSupport.vmx86] 
2023-11-02T10:46:21.477Z In(05)+ vcpu-0 - To collect data to submit to VMware technical support, run "vm-support".
2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.response] We will respond on the basis of your support entitlement.
2023-11-02T10:46:21.477Z In(05) vcpu-0 - ----------------------------------------
2023-11-02T10:46:21.478Z In(05) vmx - VigorTransportProcessClientPayload: opID=vim-cmd-48-4348-434a seq=179: Receiving Bootstrap.MessageReply request.
2023-11-02T10:46:21.478Z In(05) vmx vim-cmd-48-4348-434a VigorTransport_ServerSendResponse opID=vim-cmd-48-4348-434a seq=179: Completed Bootstrap request.
```

Esxi下kp的日志如上

PAINCLOWN, 2023-11-04:
> ```shell
> 2023-11-02T10:45:22.982Z In(05) vcpu-1 - AHCI-VMM: sata0:1: Halted the port due to command abort.
> 2023-11-02T10:45:22.982Z In(05) vcpu-0 - AHCI-VMM:Aborting the command: 0xb0
> 2023-11-02T10:45:22.982Z In(05) vcpu-0 - AHCI-VMM: sata0:2: Halted the port due to command abort.
> 2023-11-02T10:46:19.317Z Wa(03) vcpu-0 - MONITOR PANIC: vcpu-1:VMM fault 14: src=MONITOR rip=0xfffffffffc04e146 regs=0xfffffffffc407e80 LBR stack=0xfffffffffc407b20
> 2023-11-02T10:46:19.317Z In(05) vcpu-2 - Exiting vcpu-2
> 2023-11-02T10:46:19.317Z In(05) vcpu-3 - Exiting vcpu-3
> 2023-11-02T10:46:19.317Z In(05) vcpu-1 - Exiting vcpu-1
> 2023-11-02T10:46:19.317Z In(05) vcpu-0 - Core dump with build build-21930508
> 2023-11-02T10:46:19.319Z In(05) vcpu-0 - Writing monitor `vmmcores.gz`
> 2023-11-02T10:46:19.319Z Wa(03) vcpu-0 - Dumping core for vcpu-0
> 2023-11-02T10:46:19.319Z In(05) vcpu-0 - VMK Stack for vcpu 0 is at 0x453890c13000
> 2023-11-02T10:46:19.319Z In(05) vcpu-0 - Beginning monitor coredump
> 2023-11-02T10:46:19.420Z Wa(03) mks - Panic in progress... ungrabbing
> 2023-11-02T10:46:19.420Z In(05) mks - MKS: Release starting (Panic)
> 2023-11-02T10:46:19.420Z In(05) mks - MKS: Release finished (Panic)
> 2023-11-02T10:46:19.612Z In(05) vcpu-0 - End monitor coredump
> 2023-11-02T10:46:19.612Z Wa(03) vcpu-0 - Dumping core for vcpu-1
> 2023-11-02T10:46:19.612Z In(05) vcpu-0 - VMK Stack for vcpu 1 is at 0x453892f93000
> 2023-11-02T10:46:19.612Z In(05) vcpu-0 - Beginning monitor coredump
> 2023-11-02T10:46:19.901Z In(05) vcpu-0 - End monitor coredump
> 2023-11-02T10:46:19.901Z Wa(03) vcpu-0 - Dumping core for vcpu-2
> 2023-11-02T10:46:19.901Z In(05) vcpu-0 - VMK Stack for vcpu 2 is at 0x45388a593000
> 2023-11-02T10:46:19.901Z In(05) vcpu-0 - Beginning monitor coredump
> 2023-11-02T10:46:20.195Z In(05) vcpu-0 - End monitor coredump
> 2023-11-02T10:46:20.196Z Wa(03) vcpu-0 - Dumping core for vcpu-3
> 2023-11-02T10:46:20.196Z In(05) vcpu-0 - VMK Stack for vcpu 3 is at 0x453891113000
> 2023-11-02T10:46:20.196Z In(05) vcpu-0 - Beginning monitor coredump
> 2023-11-02T10:46:20.483Z In(05) vcpu-0 - End monitor coredump
> 2023-11-02T10:46:21.477Z Wa(03) vcpu-0 - A core file is available in "/vmfs/volumes/65266f4c-d33be7d8-deea-7085c237dd5a/Nas/vmx-zdump.002"
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - Msg_Post: Error
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.log.error.unrecoverable] VMware ESX unrecoverable error: (vcpu-0)
> 2023-11-02T10:46:21.477Z In(05)+ vcpu-0 - vcpu-1:VMM fault 14: src=MONITOR rip=0xfffffffffc04e146 regs=0xfffffffffc407e80 LBR stack=0xfffffffffc407b20
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.haveLog] A log file is available in "/vmfs/volumes/65266f4c-d33be7d8-deea-7085c237dd5a/Nas/vmware.log".  
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.requestSupport.withoutLog] You can request support.  
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.requestSupport.vmSupport.vmx86] 
> 2023-11-02T10:46:21.477Z In(05)+ vcpu-0 - To collect data to submit to VMware technical support, run "vm-support".
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - [msg.panic.response] We will respond on the basis of your support entitlement.
> 2023-11-02T10:46:21.477Z In(05) vcpu-0 - ----------------------------------------
> 2023-11-02T10:46:21.478Z In(05) vmx - VigorTransportProcessClientPayload: opID=vim-cmd-48-4348-434a seq=179: Receiving Bootstrap.MessageReply request.
> 2023-11-02T10:46:21.478Z In(05) vmx vim-cmd-48-4348-434a VigorTransport_ServerSendResponse opID=vim-cmd-48-4348-434a seq=179: Completed Bootstrap request.
> ```
> 
> Esxi7.0 Update 3下kp的日志如上, 值得一提的是`AHCI-VMM: sata0:1: Halted the port due to command abort.`和`AHCI-VMM: sata0:2: Halted the port due to command abort.` 会在群晖开机之后反复出现，直到发生kp

建议带上具体的配置和dsm的内核日志
不然不好判断

wjz304, 2023-11-20:
https://github.com/wjz304/rr/issues/369

alipig, 2023-11-24:
> 我 exsi里面看CPU占用率，升级7.2.1后，启动时CPU占用非常高，等启动完，不访问 群晖登录界面就可以用ssh登录一直使用，无论什么时候访问web登录界面，就出现404错误，然后群晖就异常关机了

昨天刚更新的RR版本，正常升级DSM到7.2.1

wjz304, 2024-03-30:
复现请重新打开

closed 2024-03-30 by wjz304