RROrg/rr #277: 白群能使用nvme扩展卡来实现ds3622等非内置nvme插槽机型的nvme硬盘在存储空间管理员首页显示，有没有办法能让黑群也这样显示nvme。

PWMFI, 2023-10-19:
[image]

PWMFI, 2023-10-19:
目前Esxi直通了拆分卡上的两条nvme，能识别单独的两条nvme

naoki66, 2023-10-19:
addons选中 nvmevolume试试看

PWMFI, 2023-10-19:
> addons选中 nvmevolume试试看

这个只能用nvme创建存储池，并不能让nvme在存储空间管理器首页里面显示

naoki66, 2023-10-20:
> > addons选中 nvmevolume试试看
> 
> 这个只能用nvme创建存储池，并不能让nvme在存储空间管理器首页里面显示

https://github.com/007revad/Synology_enable_M2_volume

PWMFI, 2023-10-20:
> > > addons选中 nvmevolume试试看
> > 
> > 
> > 这个只能用nvme创建存储池，不能让nvme在存储空间管理器首页里面显示
> 
> https://github.com/007revad/Synology_enable_M2_volume

非常感谢

PWMFI, 2023-10-20:
还有一个问题，黑群的硬盘顺序是错乱的，DiskIdxMap怎么改都不行，有什么好的方法吗，用Tincore的编译程序试过了，给出的参数也是错误的

PWMFI, 2023-10-20:
> > > addons选中 nvmevolume试试看
> > 
> > 
> > 这个只能用nvme创建存储池，并不能让nvme在存储空间管理器首页里面显示
> 
> https://github.com/007revad/Synology_enable_M2_volume

这个好像不能在非内置nvme插槽的机型在首页显示nvme硬盘

PWMFI, 2023-10-20:
有没有插件能让pci的m.2的缓存盘在存储空间管理器首页显示出来

naoki66, 2023-10-20:
> 还有一个问题，黑群的硬盘顺序是错乱的，DiskIdxMap怎么改都不行，有什么好的方法吗，用Tincore的编译程序试过了，给出的参数也是错误的

https://github.com/wjz304/rr/issues/278

closed 2023-10-22 by PWMFI