RROrg/rr #278: 请教Sata_remap的用法

naoki66, 2023-10-19:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  PVE
 * CPU:  PVE
 * NIC:  PVE

Arpl version: (You can find it in the update menu.)  
 * arpl:  https://github.com/wjz304/arpl-i18n/releases/tag/23.10.3
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS3622xs
 * version:  69057

Issue:  

logs:  

(Plz review the content of #173, #175, #226 first)  
...
[image]

我的引导盘写法
cmdline:
 SataPortMap: "068"
  DiskIdxMap: "18001906"

[image]
现在遇到的问题是第三个SATA控制器，物理端口只有4个，但是系统显示有8个（38-45）。
目前群辉系统盘号出现1、2、3、4、5、6、11、12、13、14.中间的7 、8、9、10不连续。后面搜索到
https://gugucomputing.wordpress.com/2018/11/11/experiment-on-sata_args-in-grub-cfg/
Add boot argument sata_remap to remap data port sequence.
For example, sata_remap=0>4:4>0 means swap the first disk name
and the 5th. The following is the remap result.
ata1 – sde
ata2 – sdb
ata3 – sdc
ata4 – sdd
ata5 – sda
尝试增加cmdline:
  Sata_remap: "11>7:12>8:13>9:14>10:7>11:8>12:9>13:10>14"
但是并没有如愿以偿的解决7 、8、9、10不连续的情况。
所以想请教一下，没搞懂sata_remap的参数意义，大于符号＞是交换？冒号：是分隔？

naoki66, 2023-10-19:
系统识别为11、12、13、14盘，缺失7、8、9、10.
设置参数  Sata_remap: "a>6:b>7:c>8:d>9:6>a:7>b:8>c:9>d"
启动后无法识别控制器上的全部硬盘、
参考这个的设置
https://xpenology.com/forum/topic/52094-how-to-config-sataportmap-sata_remap-and-diskidxmap/?do=findComment&comment=445073

wjz304, 2023-10-20:
所以我不建议设置 cmdline里那几个，当然只是不建议，

closed 2023-11-04 by wjz304