RROrg/rr #281: Установка загрузчика на SSD диск.

longtongue91, 2023-10-20:
Здравствуйте, если возможность установки ARPL на SSD SATA? И если в этом практически смысл? Будет ли какие-нибудь баги?

wjz304, 2023-10-20:
Not currently possible,

I'm considering adding this feature, so stay tuned.

longtongue91, 2023-10-20:
> Not currently possible,
> 
> I'm considering adding this feature, so stay tuned.

У вас же написано, что последняя версия ARPL  23.10.4? Или будет что-то новое?

longtongue91, 2023-10-20:
> [image]
> [image]
> 

Прошу прощения, Я не очень умный. Спасибо вам за ответы!

wjz304, 2023-10-20:
Nah. I added it later

closed 2023-10-20 by longtongue91

longtongue91, 2023-10-24:
Оказалось ларчик просто открывался. Я записал образ ARPL сразу на жёсткий диск с помощью balenaEtcher. Работает в штатном режиме.

wjz304, 2023-10-25:
I'm sorry, I made a mistake at the beginning. I thought you wanted to use m.2 ssd for bootloader. so I said no.
m.2 bootloaderwill be supported in the near future.

Sata has always been possible, without distinguishing between sata hdd and sata ssd, and there is no problem.