RROrg/rr #285: 9900T 核显支持问题

FlankerZzzz, 2023-10-21:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  DMI指的是？
 * CPU:  9900T
 * NIC:  RTL8125

Arpl version: (You can find it in the update menu.)  
 * arpl:  10.3
 * addons:  10.3
 * modules:  9.5
 * lkms:  9.2

DSM: 
 * model:  918+
 * version:  7.2.1

Issue:  
8600T换CPU9900T，更换后无法video station无法播放MKV和硬解，尝试过重装（重装完后ls -l /dev/dri无返回）和https://wp.gxnas.com/13344.html/comment-page-2 这个方法替换.ko文件，替换后ls -l /dev/dri 有返回cat /sys/kernel/debug/dri/0/i915_frequency_info | grep HW 返回yes，但是依然不行，看日志有报错，dump文件请见附件两个样本

logs:  

2023-10-22T04:24:24+08:00 FlankerDSM coredump[1818]: Process ffmpeg41[1817](/volume5/@appstore/CodecPack/pack/bin/ffmpeg41) dumped core on signal [6]. Core file [/volume1/@ffmpeg41.synology_apollolake_918+.69057.core.gz]. Cmdline [/var/packages/CodecPack/target/bin/ffmpeg41 -ss 0.000 -vaapi_device /dev/dri/renderD128 -hwaccel vaapi -hwaccel_output_format vaapi -noautorotate -i /volume3/影音库/影片/星际迷航/奇异新世纪Strange New World/第一季/_STCN__星际迷航.新世界.S01E01.新世界.WEBRip.2160P.SDR.10bit.HEVC.AC-3.CHS-ENG.mkv -r 24 -vcodec h264_vaapi -vf format=nv12|vaapi,hwupload,scale_vaapi=w=1920:h=784,scale_vaapi=format=nv12 -vsync 2 -bf 0 -vb 15000000 -acodec libmp3lame -ab 128K -ac 2 -f ssegment -pix_fmt yuv420p -segment_format mpegts -segment_list_type m3u8 -hls_seek_time 0 -segment_time 5 -segment_time_delta 0.000 -segment_start_number 00000 -avoid_negative_ts 0 -break_non_keyframes 1 -max_muxing_queue_size 1024 -map 0:0 -map 0:1 /tmp/VideoStation/HLS/bbbf671cbce421a853ef45eec7cfea93_YrfmM6Aj/sli
2023-10-22T04:24:24+08:00 FlankerDSM coredump[1818]: ce-%05d.ts ]
2023-10-22T04:24:24+08:00 FlankerDSM coredump[1818]: Core file [/volume1/@ffmpeg41.synology_apollolake_918+.69057.core.gz] size [307550]

环境信息
宿主机上
root@flanker:/etc/modprobe.d# uname -r
6.2.16-15-pve

grub.cfg 
GRUB_CMDLINE_LINUX_DEFAULT="quiet intel_iommu=on iommu=pt"

pve-blacklist.conf
  4 #block AMD driver
  5 blacklist radeon 
  6 blacklist amdgpu
  7 #block NVIDIA driver
  8 blacklist nouveau
  9 blacklist nvidia
 10 blacklist nvidiafb
 11 #intel
 12 blacklist i915
 13 blacklist snd_hda_intel
 14 #attach
 15 options vfio_iommu_type1 allow_unsafe_interrupts=1

/etc/modules
vfio
vfio_iommu_type1
vfio_pci
vfio_virqfd

i915.conf
options i915 enable_guc=3
[@ffmpeg41.synology_apollolake_918+.69057.core.gz](https://github.com/wjz304/rr/files/13061879/%40ffmpeg41.synology_apollolake_918%2B.69057.core.gz)
[@ffmpeg41.synology_apollolake_918+.69057.core-1.gz](https://github.com/wjz304/rr/files/13061880/%40ffmpeg41.synology_apollolake_918%2B.69057.core-1.gz)

以下为群晖环境内情况
root@FlankerDSM:/volume1# uname -r
4.4.302+
lspci -v
        0000:06:1b.0 Class 0300: Device 8086:3e98 (rev 02)
Subsystem: Device 8086:2015
Flags: bus master, fast devsel, latency 0, IRQ 63
Memory at c0000000 (64-bit, non-prefetchable) [size=16M]
Memory at 7070000000 (64-bit, prefetchable) [size=256M]
I/O ports at 9100 [size=64]
Capabilities: [40] Vendor Specific Information: Len=0c <?>
Capabilities: [ac] MSI: Enable+ Count=1/1 Maskable- 64bit-
Capabilities: [d0] Power Management version 2
Kernel driver in use: i915

root@FlankerDSM:/volume1# ls -l /dev/dri/
total 0
drwxr-xr-x 2 root root              80 Oct 22 04:10 by-path
crw------- 1 root root        226,   0 Oct 22 04:10 card0
crw-rw---- 1 root videodriver 226, 128 Oct 22 04:10 renderD128

root@FlankerDSM:/volume1# cat /sys/kernel/debug/dri/0/i915_frequency_info 
Video Turbo Mode: yes
HW control enabled: yes
SW control enabled: no
PM IER=0x00000000 IMR=0xffffffff ISR=0x00000000 IIR=0x00000000, MASK=0x00003ffe
pm_intrmsk_mbz: 0x80000000
GT_PERF_STATUS: 0x00000000
Render p-state ratio: 0
Render p-state VID: 0
Render p-state limit: 255
RPSTAT1: 0x0a808000
RPMODECTL: 0x00000d92
RPINCLIMIT: 0x00002c88
RPDECLIMIT: 0x00004fb0
RPNSWREQ: 350MHz
CAGF: 350MHz
RP CUR UP EI: 52 (69us)
RP CUR UP: 52 (69us)
RP PREV UP: 0 (0us)
Up threshold: 95%
RP CUR DOWN EI: 53 (70us)
RP CUR DOWN: 54 (72us)
RP PREV DOWN: 0 (0us)
Down threshold: 85%
Lowest (RPN) frequency: 350MHz
Nominal (RP1) frequency: 350MHz
Max non-overclocked (RP0) frequency: 1200MHz
Max overclocked frequency: 1200MHz
Current freq: 350 MHz
Actual freq: 350 MHz
Idle freq: 350 MHz
Min freq: 350 MHz
Boost freq: 1200 MHz
Max freq: 1200 MHz
efficient (RPe) frequency: 350 MHz
Current CD clock frequency: 337500 kHz
Max CD clock frequency: 675000 kHz
Max pixel clock frequency: 675000 kHz

root@FlankerDSM:/volume1# dmesg |grep i915
[   25.114150] i915 0000:06:1b.0: BAR 6: can't assign [??? 0x00000000 flags 0x20000000] (bogus alignment)
[   25.122966] [drm] Initialized i915 1.6.0 20171222 for 0000:06:1b.0 on minor 0
[   25.132480] [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
[   25.847229] i915 0000:06:1b.0: fb0: inteldrmfb frame buffer device

root@FlankerDSM:/volume1# lsmod | grep i915
i915                 1314764  0 
drm_kms_helper        123302  1 i915
drm                   313498  3 i915,drm_kms_helper
iosf_mbi                4298  1 i915
fb                     35077  2 i915,drm_kms_helper
video                  27424  1 i915
backlight               6170  2 i915,video
i2c_algo_bit            5876  1 i915
button                  5161  1 i915

期待您的指导！

(Plz review the content of #173, #175, #226 first)  
...
[@ffmpeg41.synology_apollolake_918+.69057.core.gz](https://github.com/wjz304/rr/files/13061882/%40ffmpeg41.synology_apollolake_918%2B.69057.core.gz)
[@ffmpeg41.synology_apollolake_918+.69057.core-1.gz](https://github.com/wjz304/rr/files/13061883/%40ffmpeg41.synology_apollolake_918%2B.69057.core-1.gz)

FlankerZzzz, 2023-10-21:
[image]

FlankerZzzz, 2023-10-22:
我用jellyfin试了一下似乎是可以硬解的？难道是官方的vs不支持吗，大佬有没有什么类似intel_gpu_top的工具可以确定一下是不是真的调用硬解了呢？另外大佬，i915.enable_guc=2 这样的参数我要怎么加才能生效啊？

OldLi1997, 2023-10-27:
你是物理机还是虚拟机直通？