RROrg/rr #290: not found /usr/bin/rr-reboot.sh

yanghanbin, 2023-10-23:
arpl version :23.10.3

sudo -i  # 输入密码
/usr/bin/rr-reboot.sh "config"

wjz304, 2023-10-23:
/usr/bin/arpl-reboot.sh "config"

wjz304, 2023-10-23:
现在时改版期

yanghanbin, 2023-10-23:
> /usr/bin/arpl-reboot.sh "config"

-sh: /usr/bin/arpl-reboot.sh: No such file or directory

wjz304, 2023-10-28:
自己写一个进去也是可以的

loader-reboot.sh /arpl-reboot.sh
```shell
#!/usr/bin/env bash

function use() {
  echo "Use: ${0} junior|config"
  exit 1
}

# Sanity checks
if [ ${USER} != "root" ]; then
  exec sudo $0 $@
fi
[ -z "${1}" ] && use
[ "${1}" != "junior" -a "${1}" != "config" ] && use
echo "Rebooting to ${1} mode"
echo 1 >/proc/sys/kernel/syno_install_flag
mount /dev/synoboot1 /mnt
GRUBPATH="$(dirname $(find /mnt/ -name grub.cfg | head -1))"
ENVFILE="${GRUBPATH}/grubenv"
[ ! -f "${ENVFILE}" ] && grub-editenv ${ENVFILE} create

grub-editenv ${ENVFILE} set next_entry="${1}"
umount /mnt
[ -x /usr/syno/sbin/synopoweroff ] &&
  /usr/syno/sbin/synopoweroff -r ||
  reboot

```

closed 2023-10-28 by wjz304