RROrg/rr #292: The ARRAY variable mirrors is missing.

PeterSuh-Q3, 2023-10-24:
https://github.com/wjz304/rr/blob/main/files/initrd/opt/rr/menu.sh#799

The ARRAY variable mirrors is missing.

If there is no declaration of mirrors, it will not work even if the fastest mirror is found.

Perhaps LINE 799 ~ 801 should be modified as follows.

     mirrors=("global.synologydownload.com" "global.download.synology.com" "cndl.synology.cn")
     fastest=$(_get_fastest ${mirrors[@]})

wjz304, 2023-10-24:
它们并没有区别

PeterSuh-Q3, 2023-10-25:
I don't think you understood what I meant.
The declaration of these mirrors is not visible in your script.
mirrors=("global.synologydownload.com" "global.download.synology.com" "cndl.synology.cn")

If you want to test it like your script, the declaration of the mirrors variable should be omitted.
If omitted, an error will occur.

This is the result of using the current menu.sh script without changing it.
Since there is no mirrors variable declaration, it prints nothing.

[image]

PeterSuh-Q3, 2023-10-25:
You have tested in the form of To-be I have told you.

AS-IS is like this.

     fastest=$(_get_fastest "global.synologydownload.com" "global.download.synology.com" "cndl.synology.cn")
     mirror="$(echo ${PATURL} | sed 's|^http[s]*://\([^/]*\).*|\1|')"
     if echo "${mirrors[@]}" | grep -wq "${mirror}" && [ "${mirror}" != "${fastest}" ]; then
       echo "$(printf "$(TEXT "Based on the current network situation, switch to %s mirror to downloading.")" "${fastest}")"
       PATURL="$(echo ${PATURL} | sed "s/${mirror}/${fastest}/")"
       OLDPATURL="https://${fastest}/download/DSM/release/7.0.1/42218/DSM_DS3622xs%2B_42218.pat"
     fi

wjz304, 2023-10-25:
Oh, I checked the code and I understand what you mean. 
This is a bug. Tthank you.

closed 2023-10-25 by PeterSuh-Q3