RROrg/rr #294: 能否在群晖中直接更新arpl的新版本

cglzyz, 2023-10-25:
现在需要先下载arpl的新版update.zip,再重启群晖进入重装设置模式,更新arpl后再更新群晖系统,操作麻烦,很多操作还要去NAS实体机器上操作.
如果能在群晖运行中直接更新arpl,直接操作synoboot修改文件,那么就完全不需要再去实体NAS上做任何操作了,群晖升级新版也更方便,希望能采纳.

yanghanbin, 2023-10-26:
sudo -i # 输入密码
/usr/bin/rr-reboot.sh "config"
或者
/usr/bin/arpl-reboot.sh "config"

可以进入到arpl引导界面，通过Ip+端口访问就可以升级了，但是最近的版本，我是找不着这文件了，只能空了手动去物理机上操作下了
我是从很早之前的版本一步步迭代上来的，一直都没插过键盘和显示器就完成了升级

wjz304, 2023-10-26:
Please fill in the following information.

Install ENV: (You can find it in the boot interface.)

DMI:
CPU:
NIC:
RR version: (You can find it in the update menu.)

RR:
addons:
modules:
lkms:
DSM:

model:
version:
Issue:

logs:

stkroot, 2023-10-31:
控制面板，计划任务有个arpl脚本（如果没有可能需要在arpl下安装插件 rebootttoarpl）

点运行脚本就可以自动重启到Arpl了，然后 http://IP:7681 访问Arpl控制台

Miaooo92, 2023-11-05:
> 控制面板，计划任务有个arpl脚本（如果没有可能需要在arpl下安装插件 rebootttoarpl）
> 
> 点运行脚本就可以自动重启到Arpl了，然后 http://IP:7681 访问Arpl控制台

目前的引导没有这个插件，请问应该如何安装呢？

stkroot, 2023-11-05:
> > 控制面板，计划任务有个arpl脚本（如果没有可能需要在arpl下安装插件 rebootttoarpl）
> > 点运行脚本就可以自动重启到Arpl了，然后 http://IP:7681 访问Arpl控制台
> 
> 目前的引导没有这个插件，请问应该如何安装呢？

可能新版的 rr 代码还没跟上，我用的 arpl-i18n 没升级rr
不追新，稳定为主

stkroot, 2023-11-05:
我刚试了下，最新版的rr有这个插件，插件默认就是开启的，运行也正常

yourber, 2023-11-15:
> 控制面板，计划任务有个arpl脚本（如果没有可能需要在arpl下安装插件 rebootttoarpl）
> 
> 点运行脚本就可以自动重启到Arpl了，然后 http://IP:7681 访问Arpl控制台

这个好~

galaxy-eye, 2023-11-22:
> sudo -i # 输入密码 /usr/bin/rr-reboot.sh "config" 或者 /usr/bin/arpl-reboot.sh "config"
> 
> 可以进入到arpl引导界面，通过Ip+端口访问就可以升级了，但是最近的版本，我是找不着这文件了，只能空了手动去物理机上操作下了 我是从很早之前的版本一步步迭代上来的，一直都没插过键盘和显示器就完成了升级

请问下怎么升级呢，和安装的时候一样选择需要升级的版本，然后编译引导然后启动吗？这样的话数据盘的数据还在吗？编译完启动之后还需要在系统里手动更新对应的pat文件吗？
谢谢解答~

JohnED512, 2023-11-25:
> > sudo -i # 输入密码 /usr/bin/rr-reboot.sh "config" 或者 /usr/bin/arpl-reboot.sh "config"
> > 可以进入到arpl引导界面，通过Ip+端口访问就可以升级了，但是最近的版本，我是找不着这文件了，只能空了手动去物理机上操作下了 我是从很早之前的版本一步步迭代上来的，一直都没插过键盘和显示器就完成了升级
> 
> 请问下怎么升级呢，和安装的时候一样选择需要升级的版本，然后编译引导然后启动吗？这样的话数据盘的数据还在吗？编译完启动之后还需要在系统里手动更新对应的pat文件吗？ 谢谢解答~

在群晖里升级完成后自动重启进入BIOS选择编译引导然后启动就可以了

galaxy-eye, 2023-11-27:
> 在群晖里升级完成后自动重启进入BIOS选择编译引导然后启动就可以了

感谢~下次有升级我去试试

wjz304, 2024-03-30:
rr-manager

closed 2024-03-30 by wjz304