RROrg/rr #296: BUG

livebelive, 2023-10-27:
我想请教一下，我是技嘉B760i D5的板子做的主机，机箱8盘位，技嘉板载只有2个sata，我用了一个ASM1166做了一个M2转6口sata的扩展，机器型号我用的是DS1821+（因为图形是8盘位，有强迫症）。
现在问题是,不管怎么调整SataPortMap和DiskIdxMap，都存在以下两个问题：
1：硬盘不会按照指定的号码显示，他只会根据目前的盘数自动从1排序....我希望的是把相应的sata口的硬盘序号固定下来，查了一下资料，说是DS918+是可以固定的，但是DS818+的图形显示是4盘位，我想直到为什么DS1821+不能固定硬盘序号？
2：我现在硬盘没有接满，现在的问标是，开机前插的盘在运行中热插拔一切正常，但如空接口在系统运行时插入硬盘，就无法识别硬盘，一定要开机前插上，这个是黑群晖的问题还是我选的型号的问题呢？有解决的办法吗？

wjz304, 2023-10-28:
好吧， 我之后做一个 DT型号全硬盘的插件吧

> Присоединяюсь к обсуждению. До сих пор так и не разобрался с этим - [#225 (comment)](https://github.com/wjz304/rr/issues/225#issue-1902903232). Здесь мне советовали разобраться с древом устройств. Но я так и не смог понять как мне добавить пустующие слоты. [image] На скриншоте видно, что отображаются только слоты в которые вставлены диски - 3 штуки, а пустые не видны. У меня 6 отсеков под диски. Нужно ли заполнять все отсеки дисками? Или как это реализовать? P.S. И будет ли работать это после обновления или переустановки DSM?

livebelive, 2023-10-29:
@longtongue91

livebelive, 2023-10-29:
@longtongue91 
可以参考我下面的内容 [我是通过修改DTB文件来固定接口和序号的，这样既能解决硬盘序号自动从1排序的问题，又能解决热插拔，教程参考](https://www.mi-d.cn/8607)
/dts-v1/;

/ {
	compatible = "Synology";
	model = "synology_v1000_1821+";
	version = <0x01>;

	internal_slot@1 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:03.0";
			ata_port = <0x03>;
		};
	};

	internal_slot@2 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:03.0";
			ata_port = <0x04>;
		};
	};

	internal_slot@3 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x02>;
		};
	};

	internal_slot@4 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x03>;
		};
	};

	internal_slot@5 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x04>;
		};
	};

	internal_slot@6 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x05>;
		};
	};

	internal_slot@7 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x06>;
		};
	};

	internal_slot@8 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x07>;
		};
	};

	internal_slot@9 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:05.0";
			ata_port = <0x00>;
		};
	};

	usb_slot@1 {

		usb2 {
			usb_port = "1-1";
		};

		usb3 {
			usb_port = "1-1";
		};
	};

	usb_slot@2 {

		usb2 {
			usb_port = "1-2";
		};

		usb3 {
			usb_port = "1-2";
		};
	};

	usb_slot@3 {

		usb2 {
			usb_port = "1-3";
		};

		usb3 {
			usb_port = "1-3";
		};
	};

	usb_slot@4 {

		usb2 {
			usb_port = "1-4";
		};

		usb3 {
			usb_port = "1-4";
		};
	};

	usb_slot@5 {

		usb2 {
			usb_port = "2-1";
		};

		usb3 {
			usb_port = "2-1";
		};
	};

	usb_slot@6 {

		usb2 {
			usb_port = "2-2";
		};

		usb3 {
			usb_port = "2-2";
		};
	};

	usb_slot@7 {

		usb2 {
			usb_port = "2-3";
		};

		usb3 {
			usb_port = "2-3";
		};
	};

	usb_slot@8 {

		usb2 {
			usb_port = "2-4";
		};

		usb3 {
			usb_port = "2-4";
		};
	};
};

wjz304, 2023-11-04:
楼上正解  新版本以增加 HDD sort 选项。

closed 2023-11-04 by wjz304