RROrg/rr #297: nvme_dev_port_check.c:23 Failed to get slot informtion of nvme0n1

artemus717, 2023-10-27:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  E3 v3
 * NIC:  Intel 217

RR version: (You can find it in the update menu.)  
 * RR:  23.6.8
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  DS920+
 * version:  DSM 7.2-64570 Update 3

Issue:  
系统无法识别NVME盘。使用的是PCI-E卡转接的NVME。Shell中已经能看到NVME信息。试了其他的工具也无法正常打开。
[image]

logs:  

2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_model_spec_get.c:81 Fail to get fdt property of power_limit
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_model_spec_get.c:164 Fail to get power limit of nvme0n1
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_slot_info_get.c:37 Failed to get model specification
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_dev_port_check.c:23 Failed to get slot informtion of nvme0n1
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_model_spec_get.c:81 Fail to get fdt property of power_limit
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_model_spec_get.c:164 Fail to get power limit of nvme0n1
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_slot_info_get.c:37 Failed to get model specification
2023-10-27T21:30:46+08:00 DS920 synostgd-disk[12451]: nvme_dev_port_check.c:23 Failed to get slot informtion of nvme0n1

(Plz review the content of #173, #175, #226 first)  
...

artemus717, 2023-10-27:
PCI-E转接卡的NVME是在系统安装完成后再插上的。

wjz304, 2023-10-27:
`udevadm info --query property --name /dev/nvme0n1`

artemus717, 2023-10-27:
ash-4.4# udevadm info --query property --name /dev/nvme0n1
DEVNAME=/dev/nvme0n1
DEVPATH=/devices/pci0000:00/0000:00:17.0/0000:13:00.0/nvme/nvme0/nvme0n1
DEVTYPE=disk
MAJOR=259
MINOR=0
PHYSDEVBUS=pci
PHYSDEVDRIVER=nvme
PHYSDEVPATH=/devices/pci0000:00/0000:00:17.0/0000:13:00.0
SUBSYSTEM=block
SYNO_ATTR_SERIAL=PHBT829405YL016D
SYNO_DEV_DISKPORTTYPE=UNKNOWN
SYNO_INFO_PLATFORM_NAME=geminilake
SYNO_KERNEL_VERSION=4.4
SYNO_SUPPORT_USB_PRINTER=yes
SYNO_SUPPORT_XA=no
TAGS=:systemd:
USEC_INITIALIZED=543644

artemus717, 2023-10-27:
我正在升级ARPL、addon、modules

artemus717, 2023-10-27:
Upgraded to latest 23.10.4 and latest addon modules, resolved.

closed 2023-10-27 by artemus717