RROrg/rr #299: Intel 7600T kernel Panic - Not Syncing

Tk95100, 2023-10-29:
Status: My system is Optiplex 7050 bare metal (my second 7050 unit), CPU Intel 7600T, 8GB DDR4.
arpl version arpl-i18n-23.10.4.img

After initial boot setup, system crash at loading latest DSM 7.2.1-update 1 in menu.sh. Kernel panic - not syncing.  I've found a possible cause/solution on zendesk, but do not know how to add to grub.

Root Cause quote zendesk "CPU is likely using sophisticated C-state logic of modern CPUs. It can cause these issues when intel_idle driver doesn’t work properly with modern h/w C-states. Similar problems are reported with different distributions."

I think being a low power 7600T with high C-states may have cause kernel panic.

Possible Solution quote zendeks "So basically we need to set up intel_idle module to use a simplified C-state logic. In order to do that, add intel_idle.max_cstate=1 entry to the kernel command line in /etc/default/grub.
For instance: GRUB_CMDLINE_LINUX="biosdevname=0 crashkernel=auto nomodeset rd.auto=1 consoleblank=0 intel_idle.max_cstate=1"
Then please regenerated GRUB configuration: grub2-mkconfig -o /boot/grub2/grub.cfg
You will need to reboot your server to apply the new configuration."

wjz304, 2023-10-29:
[image]
[image]
[image]
[image]
[image]

Tk95100, 2023-10-30:
This works. Thank you for quick response.

closed 2023-10-31 by wjz304