RROrg/rr #302: 如果启动时硬盘位没硬盘，进入群晖后，再插入硬盘不识别了，但单独测试热插拔又都是正常的。

stkroot, 2023-10-29:
在 ARPL 下检测硬盘端口那里，热插拔硬盘都正常（无论之前是否插入硬盘，都可以正常识别）

1. 如果启动群晖前，有插入硬盘。启动后热插拔也正常。
2. 但是如果启动群晖前未插入硬盘，启动群晖后再插入，就没法识别到硬盘了

这样的话，热插拔就完全没意义了，因为必须开机就插满硬盘，否则没插的那个盘位就废了？是不是有什么Bug？

stkroot, 2023-10-29:
现在热插拔功能要正常使用，开机就必须先插满硬盘，开机后再拔出都行，否则热插拔功能就无效

能否改一下，让 ARPL 启动到群晖时，自动模拟空盘位有硬盘？

livebelive, 2023-10-30:
这个你要么用他的自动排序功能，要么关掉自动排序手动编辑好DTS文件上传，要么换DS918+，直接通过DiskIdxMap来定义。我觉得编辑DTS是最简单的。我上传我的给你参考下，你可以进到sata列表里面参考你的接口情况，那个页面可以热插拔的，拔插之后等一会，退出重新列出sata就能看到了。
`/dts-v1/;

/ {
	compatible = "Synology";
	model = "synology_v1000_1821+";
	version = <0x01>;

	internal_slot@1 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:03.0";
			ata_port = <0x04>;
		};
	};

	internal_slot@2 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:03.0";
			ata_port = <0x05>;
		};
	};

	internal_slot@3 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x02>;
		};
	};

	internal_slot@4 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x03>;
		};
	};

	internal_slot@5 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x04>;
		};
	};

	internal_slot@6 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x05>;
		};
	};

	internal_slot@7 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x06>;
		};
	};

	internal_slot@8 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:04.0";
			ata_port = <0x07>;
		};
	};

	internal_slot@9 {
		protocol_type = "sata";

		ahci {
			pcie_root = "22:05.0";
			ata_port = <0x00>;
		};
	};

	usb_slot@1 {

		usb2 {
			usb_port = "1-1";
		};

		usb3 {
			usb_port = "1-1";
		};
	};

	usb_slot@2 {

		usb2 {
			usb_port = "1-2";
		};

		usb3 {
			usb_port = "1-2";
		};
	};

	usb_slot@3 {

		usb2 {
			usb_port = "1-3";
		};

		usb3 {
			usb_port = "1-3";
		};
	};

	usb_slot@4 {

		usb2 {
			usb_port = "1-4";
		};

		usb3 {
			usb_port = "1-4";
		};
	};

	usb_slot@5 {

		usb2 {
			usb_port = "2-1";
		};

		usb3 {
			usb_port = "2-1";
		};
	};

	usb_slot@6 {

		usb2 {
			usb_port = "2-2";
		};

		usb3 {
			usb_port = "2-2";
		};
	};

	usb_slot@7 {

		usb2 {
			usb_port = "2-3";
		};

		usb3 {
			usb_port = "2-3";
		};
	};

	usb_slot@8 {

		usb2 {
			usb_port = "2-4";
		};

		usb3 {
			usb_port = "2-4";
		};
	};
};
`

livebelive, 2023-10-30:
@stkroot 用DTS文件的好处是，你可以把接口和硬盘序号定死，而且只要文件里面定义的接口，都支持热插拔，不管开机之前是否有硬盘接通。

stkroot, 2023-10-30:
> @stkroot 用DTS文件的好处是，你可以把接口和硬盘序号定死，而且只要文件里面定义的接口，都支持热插拔，不管开机之前是否有硬盘接通。

你理解错了，我的排序功能正常。
我安装的是 DS3622xs+ ，没有排序问题。我也测试过，即使中间的硬盘不插，排序也不会混乱。

现在问题是热插拔不完全正常：
1. 如果启动前插满硬盘，热插拔就正常。后续这个盘位再拔出，更换硬盘，都可以正常识别
2. 如果启动前有插口没插硬盘，开机后再插入，就不能发现新硬盘

感觉是 ARPL 启动后，再次重启引导群晖时写死了“某个盘位无硬盘”，所以进入群晖后，即使再插入硬盘也不认盘
因为在 ARPL 时热插拔也正常。

stkroot, 2023-10-30:
比如我的群晖有5个盘位，开机前只插入1，2，3，5盘位。等群晖完全启动后，排序是没问题的。
但这时只有1，2，3，5盘位，可以热插拔更换硬盘；4号盘位就废了，插入新硬盘没反应。

wjz304, 2023-10-30:
> > @stkroot 用DTS文件的好处是，你可以把接口和硬盘序号定死，而且只要文件里面定义的接口，都支持热插拔，不管开机之前是否有硬盘接通。
> 
> 你理解错了，我的排序功能正常。 我安装的是 DS3622xs+ ，没有排序问题。我也测试过，即使中间的硬盘不插，排序也不会混乱。
> 
> 现在问题是热插拔，不完全正常：
> 
> 1. 如果启动前插满硬盘，热插拔就正常。后续这个盘位再拔出，更换硬盘，都可以正常识别
> 2. 如果启动前某个盘位没插硬盘，启动到群晖之后再插入硬盘，就不能识别这个新硬盘了
> 
> 感觉是 ARPL 启动后，再次重启引导群晖时写死了“某个盘位无硬盘”，所以进入群晖后，即使再插入硬盘也不认盘。
> 
> 因为在 ARPL 时热插拔也正常，所有大概率是ARPL 到群晖的第二次重启中有问题。

synoinfo 添加 maxdisks=26

wjz304, 2023-10-30:
https://github.com/wjz304/rr/issues/226#issuecomment-1726628716

按你预期的填，获取到 4个参数 填 synoinfo里

stkroot, 2023-10-30:
谢谢，热插拔正常了