RROrg/rr #303: 从ARPL-i18n切换到RR无法使用【尝试恢复已安装DSM的系统】

koihano, 2023-10-30:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Colorful-BATTLE-AX B360M-E PRO V21
 * CPU:  Intel Core i3-8100
 * NIC:  RTL8111（NIC1）RTL8125（NIC2）

RR version: (You can find it in the update menu.)  
 * RR:  23.10.6
 * addons:  lsiutil
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DS918+
 * version:  7.2.1-69057u1

Issue:  Switching from ARPL-i18n to RR cannot be used [Try to recovery a DSM installed system]（从ARPL-i18n切换到RR无法使用【尝试恢复已安装DSM的系统】）

logs:  none

(Plz review the content of #173, #175, #226 first)  

...

[image]

livebelive, 2023-10-30:
@koihano 其实可以不用 尝试恢复已安装DSM的系统 这个的 你直接启动 然后进到WEB后 有一个恢复 点一下 就能恢复之前的系统了

wjz304, 2023-10-30:
不过这是一个问题。
在 RR shell 下执行 `blkid` 看一下

wjz304, 2023-10-30:
https://github.com/wjz304/rr/actions/runs/6688885261

closed 2023-10-30 by wjz304