RROrg/rr #304: 板载RTL8111+PCIE RTL8125BG 安装阶段失联

koihano, 2023-10-30:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  Colorful-BATTLE-AX B360M-E PRO V21
 * CPU:  Intel Core i3-8100
 * NIC:  RTL8111H（NIC1）RTL8125BG（NIC2）

RR version: (You can find it in the update menu.)  
 * RR:  23.10.6
 * addons:  lsiutil,dbgutils
 * modules:  default
 * lkms:  prod

DSM: 
 * model:  DS918+ to DS3622xs+
 * version:  7.2.1-69057u1

Issue: 
此前使用arpl-i18n版本直至23.10.4均正常，现因上一个issue故进行重装，在RR界面分配IP正常，但安装阶段失联，尝试过以下操作：
1.选用r8169+r8125驱动
2.选用r8168+r8125_v9驱动
3.切换lkms为test
4.切换lkms为dev
5.单插8111网卡启动系统（8125没拔掉，8111在DSM中为NIC1）

logs:  
[check_rootfs_health.log](https://github.com/wjz304/rr/files/13200625/check_rootfs_health.log)
[dmesg.log](https://github.com/wjz304/rr/files/13200626/dmesg.log)
[junior_reason.log](https://github.com/wjz304/rr/files/13200627/junior_reason.log)
[juniorexpansionpack.log](https://github.com/wjz304/rr/files/13200628/juniorexpansionpack.log)
[linuxrc.syno.log](https://github.com/wjz304/rr/files/13200629/linuxrc.syno.log)
[messages.log](https://github.com/wjz304/rr/files/13200630/messages.log)
[upgrade_sh.log](https://github.com/wjz304/rr/files/13200631/upgrade_sh.log)
（为方便上传已将所有日志文件后缀名改为.log

wjz304, 2023-10-30:
在吗？ 我想远程看一下

koihano, 2023-10-30:
> 在吗？ 我想远程看一下

可以的。如何联系？

wjz304, 2023-10-30:
群里@我

wjz304, 2023-10-30:
https://github.com/wjz304/rr/issues/306

koihano, 2023-10-31:
> #306

已升级到23.10.8并进行操作：

1.取消 r8168 和r8169 的勾选：路由器DHCP可识别，但不分配IP（显示0.0.0.0）
2.启用官方模块优先：没有任何反应
3.将1+2同时执行：启动成功进入安装界面，但提示DSM已重置而不是从DS918+转移

koihano, 2023-11-01:
23.10.9已经可以在默认配置不改动下正常使用了，本题可关闭

closed 2023-11-01 by koihano