RROrg/rr #306: 升级23.10.6无法引导，启动DSM后失联

tonyk520, 2023-10-30:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  ACER XC-701
 * CPU:  N3700
 * NIC:  RTL8169

RR version: (You can find it in the update menu.)  
 * RR:  23.10.6
 * addons:  default
 * modules:  ipv6,R8168
 * lkms:  prod

DSM: 
 * model: ds3622xs+
 * version:  7.2.1-69057u1

Issue:  从23.10.4脚本转换成RR，升级成23.10.6，启动DSM后无法获取IP失联，重启可以进入RR，重新编译后启动再次失联，然后重新写U盘伟23.10.6同样情况，再次重写盘为23.10.4后重新编译并恢复系统后正常引导，查看changelog中有更新过网卡驱动“4.优化 r8152/r8168/r8169 驱动 (RR and DSM).”，怀疑和此有关。

logs:  
...

wjz304, 2023-10-30:
获取网卡 pidvid

tonyk520, 2023-10-30:
0000:02:00.0 Class 0200: Device 10ec:8168 (rev 11)

Ardentwheel, 2023-10-30:
一样的问题，升级后联系不上了，网卡只亮黄灯

wjz304, 2023-10-30:
在 modules 中取消 r8168 和r8169 的勾选 或者打开 官方模块优先

Ardentwheel, 2023-10-31:
> 在 modules 中取消 r8168 和r8169 的勾选 或者打开 官方模块优先

23.10.8 

取消 r8168 和r8169 的勾选 网卡只亮黄灯
官方模块优先 网卡只亮黄灯

wjz304, 2023-11-02:
try 23.11.0（r8169_lk）

tonyk520, 2023-11-03:
> try 23.11.0（r8169_lk）

今早重新写盘23.11.0，并且试了取消r8168、r8169，只选择r8169_lk，编译后启动仍然会失联，优先使用官方版本选中也一样，然后又换回了23.10.4引导，编译时发现个不一样的地方不知道是否有关，23.10.4模块默认选择了ipv6、mii、r8169，而23.11.0默认没有选中mii，而在启动好的群晖中lsmod查询内核确实加载了mii，
r8168                 554971  0
pgdrv                   4368  0
r8169                  70624  0
mii                     4035  1 r8169
由于测试的是一台公用存数据的黑群晖，暂时没有条件马上再次测试重新刷23.11.0编译固件

jimmyk1m, 2023-11-04:
验证 23.11.0 已修复

tonyk520, 2023-11-04:
> 验证 23.11.0 已修复

我这里测试还是不行

jimmyk1m, 2023-11-04:
> > 验证 23.11.0 已修复
> 
> 我这里测试还是不行

试下配置全删，自动重新生成一遍，不要动任何配置。不过貌似mac地址的设置是不生效的

tonyk520, 2023-11-06:
> > > 验证 23.11.0 已修复
> > 
> > 
> > 我这里测试还是不行
> 
> 试下配置全删，自动重新生成一遍，不要动任何配置。不过貌似mac地址的设置是不生效的

今天试了一下23.11.1和23.11.2版本，按您说的，选择型号--选择版本--不动其他直接选择编译，确实可以了，然后又特意试了下增改插件没问题，自定义SN/MAC也没问题，但是改模块如果像原来那样选一次自动加载检测到的模块会失联，看来问题还是出在选择的模块中缺少必要的驱动，目前我是先全选，然后去掉明确知道不会用到的对应驱动模块，不确定的都保留，编译后是没问题的，能用就好，懒得去试自己的板子到底8168最少需要选择啥了（23.11.1时候迷迷糊糊试过一次好像仅选择了ipv6、mii、R8168、R8169、R8169_lk、Realtek_PHY以及其他几个公用的模块是可以启动的，当时看到又出了23.11.2就又重新写盘刷了新版默认编译，所以记不清了，有想自己选择模块的仅供参考），如果不需要留给其他和我一样问题的人看，issue应该可以关闭了。

wjz304, 2023-11-06:
有依赖，r8169 依赖mii

closed 2023-11-06 by wjz304

fxljay, 2023-11-27:
pcie 8125b到底该选r8125驱动还是r8125 v9驱动

wjz304, 2023-11-27:
> pcie 8125b到底该选r8125驱动还是r8125 v9驱动

假如，我说假如啊，我告诉你用不带后缀的，那你觉得留着 v9 的意义是什么？