RROrg/rr #307: 23.10.8硬盘识别问题

yanghanbin, 2023-10-30:
畅网N5105 6sata板子
1 X ssd，5X3.5T(raid5)，2Xnvme

当前情况是
ls /sys/block/sata*  
能识别到6块硬盘，但是系统只识别到了5块硬盘，其中编号2的硬盘缺失
[image]

启动日志：
2023-10-30T23:28:35+08:00 NAS synostgtask[13413]: task/entity/disk_entity.cpp:129 invalid disk entity for [{"path":"/dev/sata2","serial":"{}"}]
2023-10-30T23:28:35+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:35+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:35+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/entity/partition_entity.cpp:124 failed to push part of json value [{"disk":{"path":"/dev/sata2","serial":"{}"},"part_no":5}]
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/entity/disk_entity.cpp:129 invalid disk entity for [{"path":"/dev/sata2","serial":"{}"}]
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/subtask/disk_init.cpp:174 Failed to init disk init subtask with [{"disks":[{"path":"/dev/sata2","serial":"{}"}],"disks_inited":true,"space_info":{"path":"/dev/vg2","uuid":"OMW2ho-CcyG-stuA-dKBJ-BxdU-LE47-PPs3s7"}}]
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/subtask/subtask_factory.cpp:109 bad json subtask [{"disks":[{"path":"/dev/sata2","serial":"{}"}],"disks_inited":true,"space_info":{"path":"/dev/vg2","uuid":"OMW2ho-CcyG-stuA-dKBJ-BxdU-LE47-PPs3s7"}}]
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/task_caretaker.cpp:200 empty subtask pointer with [{"data":{"disks":[{"path":"/dev/sata2","serial":"{}"}],"disks_inited":true,"space_info":{"path":"/dev/vg2","uuid":"OMW2ho-CcyG-stuA-dKBJ-BxdU-LE47-PPs3s7"}},"type":"disk_initialize"}]
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgtask[13413]: task/entity/disk_entity.cpp:129 invalid disk entity for [{"path":"/dev/sata2","serial":"{}"}]
2023-10-30T23:28:36+08:00 NAS synostgd-disk[13842]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgd-disk[13842]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:36+08:00 NAS synostgd-disk[13842]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:40+08:00 NAS syno_hdd_util[14977]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2
2023-10-30T23:28:40+08:00 NAS synodisk[14983]: sata_dev_port_check.c:16 Failed to get slot informtion of sata2

wjz304, 2023-10-30:
取已下信息
```
cat /sys/block/sata*/device/syno_block_info
ls -l /sys/class/scsi_host
```
获取已下文件
/etc.defaults/model.dtb

yanghanbin, 2023-10-30:
cat /sys/block/sata*/device/syno_block_info
pciepath=00:17.0
ata_port_no=1
driver=ahci
pciepath=00:1c.1,00.0
ata_port_no=0
driver=ahci
pciepath=00:1c.1,00.0
ata_port_no=1
driver=ahci
pciepath=00:1c.1,00.0
ata_port_no=2
driver=ahci
pciepath=00:1c.1,00.0
ata_port_no=3
driver=ahci
pciepath=00:1c.1,00.0
ata_port_no=4
driver=ahci

ls -l /sys/class/scsi_host
total 0
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host0 -> ../../devices/pci0000:00/0000:00:17.0/ata1/host0/scsi_host/host0
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host1 -> ../../devices/pci0000:00/0000:00:17.0/ata2/host1/scsi_host/host1
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host2 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/ata3/host2/scsi_host/host2
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host3 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/ata4/host3/scsi_host/host3
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host4 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/ata5/host4/scsi_host/host4
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host5 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/ata6/host5/scsi_host/host5
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host6 -> ../../devices/pci0000:00/0000:00:1c.1/0000:02:00.0/ata7/host6/scsi_host/host6
lrwxrwxrwx 1 root root 0 Oct 31 07:32 host7 -> ../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/host7/scsi_host/host7

cat /etc.defaults/model.dtb
(d      Synology
300:1c.2,00.0=nvFssdcacheusb_slot@1usb2P1-1usb3P1-1usb_slot@2usb2P1-2usb3P1-2usb_slot@3usb2P1-3usb3P1-3usb_slot@4usb2P1-4usb3P1-4usb_slot@5usb2P1-5usb3P1-5usb_slot@6usb2P1-6usb3P1-6usb_slot@7usb2P1-7usb3P1-7usb_slot@8usb2P1-8usb3P1-8usb_slot@9usb2P2-1usb3P2-1usb_slot@10usb2P2-2usb3P2-2usb_slot@11usb2P2-3usb3P2-3usb_slot@12usb2P2-4usb3P2-4usb_slot@13usb2P2-5usb3P2-5usb_slot@14usb2P2-6usb3P2-6  compatiblemodelversionpower_limitprotocol_typepcie_rootata_portport_typeusb_port

wjz304, 2023-10-31:
https://github.com/wjz304/rr-addons/releases/tag/23.10.7
更新插件

yanghanbin, 2023-10-31:
已经恢复正常了，漫长的raid修复之路

closed 2023-10-31 by yanghanbin