RROrg/rr #308: 更新到RR之后出现上传到NAS会掉速的问题

wsmayi, 2023-10-31:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  畅网P5  /    其他
 * CPU:  N100    /    N3160
 * NIC:  i226   /       8111H

RR version: (You can find it in the update menu.)  
 * RR:  10.6
 * addons:  默认
 * modules:  默认
 * lkms:  默认

DSM: 
 * model:  SA6400 / DS423+
 * version:  7.2

Issue:  
用两台不同配置机子都试过了，出现了同样的问题。从黑群晖走SMB3复制文件到电脑，速度基本正常波动没什么问题。
但从电脑走SMB3上传大文件到黑群晖，稳定的最高速度大概只能跑网卡的50%，安卓手机（手机到路由器速率2882Mbps，猫头鹰文件app，支持SMB3）更惨，最高只有35MB/s左右，稳定的速度则不到30MB/s。

PS：
更新到RR之前用的是几个月之前的arpl-i18n版本，上传速度就很正常，
电脑上传能稳定跑220MB/S（2.5G）或者100MB/S（千兆）。
安卓手机上传能稳定跑70MB/s-90MB/s（千兆）以及150MB/s（2.5G）.
两台黑裙都是用的SSD盘来测速，硬盘速度不是瓶颈。
另外就是黑裙里安装HomeBox来内网测速，上传速度也飘来飘去不稳定，稳定最高速也只能跑网卡速度的50%左右。

logs:  

(Plz review the content of #173, #175, #226 first)  
...

kt212878, 2023-11-01:
之前速度正常的arpl-i18n版本号多少？

wsmayi, 2023-11-01:
> 之前速度正常的arpl-i18n版本号多少？

重新刷了找不到原来的记录了····应该是6-7月份的····

closed 2023-11-04 by wsmayi