RROrg/rr #309: BUG

Ardentwheel, 2023-10-31:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  
 * CPU:  J3455
 * NIC:  

RR version: (You can find it in the update menu.)  
 * RR:  23.10.8
 * addons:  
 * modules:  
 * lkms:  

DSM: 
 * model:  SA6400
 * version:  7.2

Issue:  
使用新版本后网卡一直亮黄灯，连不上

logs:  
[disk-by-id.log](https://github.com/wjz304/rr/files/13212280/disk-by-id.log)
[disk-scsi_host.log](https://github.com/wjz304/rr/files/13212281/disk-scsi_host.log)
[dmesg.log](https://github.com/wjz304/rr/files/13212283/dmesg.log)
[journalctl.log](https://github.com/wjz304/rr/files/13212284/journalctl.log)
[junior_reason.log](https://github.com/wjz304/rr/files/13212289/junior_reason.log)
[juniorexpansionpack.log](https://github.com/wjz304/rr/files/13212290/juniorexpansionpack.log)
[linuxrc.syno.log](https://github.com/wjz304/rr/files/13212291/linuxrc.syno.log)
[lsmod.log](https://github.com/wjz304/rr/files/13212292/lsmod.log)
[lspci.log](https://github.com/wjz304/rr/files/13212293/lspci.log)

AlleNPooh, 2023-10-31:
你的log是怎么拿到的？

wjz304, 2023-11-04:
从日志看还没启动完成，

wjz304, 2023-11-05:
3455 最少等15分钟 再说吧