RROrg/rr #313: BUG

chess-01, 2023-10-31:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  esxi7.0.2
 * CPU:  n5105
 * NIC:  i226-v

RR version: (You can find it in the update menu.)  
 * RR:  23.10.9
 * addons:  默认
 * modules:  默认
 * lkms:  默认

DSM: synology
 * model:  sa6400
 * version: 7.2-69057 

引导盘位置：sata0:3
虚拟硬盘位置：sata0:0

Issue: 一个虚拟E1000e网卡一个直通网卡，在arpl环境都能获取IP，加载dsm后直通网卡找不到IP，虚拟E1000e网卡可以进安装界面但提示找不到硬盘。
[image]

logs:  
[logs.tar.gz](https://github.com/wjz304/rr/files/13217040/logs.tar.gz)

(Plz review the content of #173, #175, #226 first)  
...

chess-01, 2023-10-31:
尝试变换虚拟硬盘位置到sata0:2和sata0:4都提示找不到硬盘

wjz304, 2023-10-31:
esxi 的问题晓得了，待解决

vinson0231, 2023-11-01:
SA6400还不成熟，除非你有非要用10代以后核显的需求，否则短时间内不见建议用这个型号，目前虚拟机上的BUG还非常多，个人建议如果有显卡需求还是用11代以前的平台，或者搞张英伟达的便宜独显直通。

koreakid, 2023-11-01:
> * RR:  23.10.9

RR:23.11.0还是一样没有修复esxi，找不到硬盘的问题。

livebelive, 2023-11-04:
有人测试 23.11.1修复了嘛？

wjz304, 2023-11-04:
没针对性修复，因为我不用esxi，所以我没办法测试。你们试试吧

livebelive, 2023-11-04:
ok我那我下来试一下 之前是失联 后面发现是直通网卡 和 VMXNET3虚拟网卡不行 后添加了E1000虚拟网卡可以连上了 但是找不到硬盘 我现在测试下网卡和硬盘是否修复了

livebelive, 2023-11-04:
老样子没变化，VMXNET3虚拟网卡，和直通网卡可以获取IP，但进不了WEB
硬盘还是无法识别
ESXI8.0
SA6400
不懂得看日志...只能反馈下结果了

dalianfei, 2023-11-04:
同，准备先物理上了，慢慢等更新

livebelive, 2023-11-04:
@dalianfei 你可以先用旧版ARPL做嘛 等更新

wjz304, 2023-11-04:
[image]

wjz304, 2023-11-04:
[image]

livebelive, 2023-11-04:
ok 我试试看

dalianfei, 2023-11-04:
> @dalianfei 你可以先用旧版ARPL做嘛 等更新

谢谢哈，最近在搞混合存储，等不及了，先把数据迁移好。后面大不了再安装下呗

dalianfei, 2023-11-04:
> [image]

谢谢大佬，该睡觉了~

livebelive, 2023-11-05:
@wjz304 嘿，你猜怎么着？可神奇了，我只要添加了串口进去，直通的网卡也能启动了，能进web，并且可以直接进系统，硬盘也都能识别。。。去掉串口就不行。，。。

koreakid, 2023-11-05:
pve也是一样的情况。神奇

koreakid, 2023-11-05:
> @wjz304 嘿，你猜怎么着？可神奇了，我只要添加了串口进去，直通的网卡也能启动了，能进web，并且可以直接进系统，硬盘也都能识别。。。去掉串口就不行。。。这是什么原理啊？ 那么目前来说，加了串口就是为了看错误日志的，但因为加了他系统变得正常了。。。这。。。

直通显卡后就无法识别，也没有办法找到IP了

livebelive, 2023-11-05:
@koreakid 我现在就这么讲究用了 哈哈 等修复了 再删那个串口 好在生成的文件并不大

koreakid, 2023-11-05:
> @koreakid 我现在就这么讲究用了 哈哈 等修复了 再删那个串口 好在生成的文件并不大

你有直通核显吗？我直通后不生效

wjz304, 2023-11-06:
> @koreakid 我现在就这么讲究用了 哈哈 等修复了 再删那个串口 好在生成的文件并不大

可以改成其他

koreakid, 2023-11-06:
> > @koreakid 我现在就这么讲究用了 哈哈 等修复了 再删那个串口 好在生成的文件并不大
> 
> 可以改成其他

请问改成其他是什么改其他

wjz304, 2023-11-06:
> > > @koreakid 我现在就这么讲究用了 哈哈 等修复了 再删那个串口 好在生成的文件并不大
> > 
> > 
> > 可以改成其他
> 
> 请问改成其他是什么改其他

串口啊，改其他类型就不生成日志文件了

dalianfei, 2023-11-06:
哈 exsi  sa6400 加了串口还是不行~  我再看看模块的选择

dalianfei, 2023-11-06:
[Uploading serial.log…]()

dalianfei, 2023-11-06:
哦对，我是用了 dell 的h730阵列卡，但是设置了直通，ds918下正常使用的

koreakid, 2023-11-06:
现在的问题是无法核显硬解

wjz304, 2023-11-06:
> 现在的问题是无法核显硬解

核显 我不管，找jim大

wjz304, 2023-11-06:
https://github.com/wjz304/rr/issues/310

wjz304, 2023-11-06:
https://github.com/wjz304/rr/issues/329

koreakid, 2023-11-06:
> > 现在的问题是无法核显硬解
> 
> 核显 我不管，找jim大

已跟jim大对线成功。给他找个个bug

livebelive, 2023-11-06:
@koreakid 我是直通12代核显的 添加串口后一切正常 之前用就的arpl已经配置好系统了，稳定使用，现在准备升级到最新的rr。
@wjz304 好的 那我就把串口类型换一个，先这样用着了。

studiociodo, 2023-11-06:
> esxi 的问题晓得了，待解决

similar problem, workaround: prepare VM on VirtualBox, export in OVA and import on Vmware. DSM boot regular, you can add new disks directly from Esxi

jim3ma, 2023-11-08:
For some devices, /dev/console does not exist or is slow to ready
need two steps to fix:
1. change linuxrc.syno output to /var/log/rc
2. fake a new available /dev/console by "mknod -m 0666 /dev/console c 1 3"

Update ramdisk-patch.sh:

```
sed -i 's#/dev/console#/var/log/rc #g' ${RAMDISK_PATH}/usr/bin/busybox
sed -i '/^echo "START/a \\nmknod -m 0666 /dev/console c 1 3' ${RAMDISK_PATH}/linuxrc.syno
```

koreakid, 2023-11-08:
Jim佬出现了

wjz304, 2023-11-12:
> For some devices like VMs in ESXi, /dev/console does not exist or is slow to ready, this issue will cause `linuxrc.syno.impl` failed to execute.
> 
> We need two steps to fix:
> 
> 1. change linuxrc.syno output to /var/log/rc
> 2. fake a new available /dev/console by "mknod -m 0666 /dev/console c 1 3"
> 
> Update ramdisk-patch.sh:
> 
> ```
> sed -i 's#/dev/console#/var/log/rc #g' ${RAMDISK_PATH}/usr/bin/busybox
> sed -i '/^echo "START/a \\nmknod -m 0666 /dev/console c 1 3' ${RAMDISK_PATH}/linuxrc.syno
> ```

提了 https://github.com/wjz304/rr/releases/tag/23.11.4

koreakid, 2023-11-12:
> 23.11.4

[23.11.4]新版本虚拟机还是得加串口，是吗？刚试了下，不加串口好像也还是不行

livebelive, 2023-11-12:
@

> > 23.11.4
> 
> [23.11.4]新版本虚拟机还是得加串口，是吗？刚试了下，不加串口好像也还是不行

我看22.11.4的说明应该是解决了才对呀。
fix sa6400 no-Serial port.
fix possible disk loss issue.
这两项应该就是针对这个问题的吧，亲测还是不行吗？

koreakid, 2023-11-12:
> @
> 
> > > 23.11.4
> > 
> > 
> > [23.11.4]新版本虚拟机还是得加串口，是吗？刚试了下，不加串口好像也还是不行
> 
> 我看22.11.4的说明应该是解决了才对呀。 fix sa6400 no-Serial port. fix possible disk loss issue. 这两项应该就是针对这个问题的吧，亲测还是不行吗？

exsi 测试了还是不行

livebelive, 2023-11-12:
晚点我也试一下，对了，为什么@不到你的ID？

koreakid, 2023-11-12:
> 晚点我也试一下，对了，为什么@不到你的ID？

不清楚…晚点回复下！我是从.1升级update.zip到4的

koreakid, 2023-11-12:
> 晚点我也试一下，对了，为什么@不到你的ID？

我错了。已经修复了。我重新整个下载下来重新弄了下，确实不需要串口了

livebelive, 2023-11-12:
好的 谢啦 昨天睡太早了 哈哈

livebelive, 2023-11-13:
测试通过，楼上兄弟是全新部署，我这边是23.11.2在线升级，问题已得到解决！可以关了这个问题！

wjz304, 2023-11-14:
https://github.com/wjz304/rr/issues/348

closed 2023-11-23 by wjz304