RROrg/rr #314: HP MicroServer Gen8安装SA34xx/SA3610，dsm提示未找到硬盘。

kt212878, 2023-10-31:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  HP ProLiant MicroServer Gen8
 * CPU:  E3-1265L V2
 * NIC:  HP 546SFP+

RR version: (You can find it in the update menu.)  
 * RR:  23.10.8
 * addons:  默认
 * modules:  默认
 * lkms:  默认

DSM: 
 * model:  SA3400/SA3410/SA3610
 * version:  7.2/7.1

Issue:  
HP MicroServer Gen8 第一sata口接ssd，引导盘型号选择了SA3400/SA3410/SA3610，dsm都提示未找到硬盘。
型号更换DS3622xs+后顺利安装，sata2-5挂多块hdd也能顺利添加。
引导盘重写，型号变更SA3410，dsm顺利安装并迁移完成，重启进系统后发现第一sata口的ssd消失，dsm安装在了sata2-5 hdd。
关机取下hdd，引导盘重写sa3410 安装dsm，继续提示未找到硬盘。

logs:  

(Plz review the content of #173, #175, #226 first)  
...

wjz304, 2023-10-31:
23.10.9

kt212878, 2023-11-01:
> 23.10.9

引导盘更新到23.11.0，选择sa3400dsm还是未检测到硬盘(sata0口单一ssd引导盘)。

kt212878, 2023-11-01:
引导型号选择SA3610（dsm7.1）后可以找到引导盘，用7.1 dsm恢复文件安装提示版本有误，要用7.2dsm文件恢复。

用7.2dsm恢复后系统正常开机。

closed 2023-11-04 by kt212878