RROrg/rr #315: Advanced Media Extensions 提示：无法执行此操作

wu0lei, 2023-11-01:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  HPE Gen10plus  Esxi8.0u2
 * CPU:  E2146G,Wmware
 * NIC:  i350.X710-T4L

RR version: (You can find it in the update menu.)  
 * RR:  23.10.9
 * addons:   23.10.7
 * modules:   23.10.2
 * lkms:   23.10.0

DSM: 
 * model:  DS918+
 * version:  Version: 7.2.1-69057 Update 1

Issue:  
Advanced Media Extensions 打开提示：无法执行此操作。请联系Synology技术支持。原引导23.10.4,之前一直正常，rr是新下载的引导，用了同样的账号和洗白码。
logs:  

(Plz review the content of #173, #175, #226 first)  
...

KleinerSource, 2023-11-01:
同样的问题  23.10.4正常  升级到 最新 RR 23.10.9 就不行了

KleinerSource, 2023-11-01:
个人认为是 引导命令里 少了 mac2 mac3....等参数 rr不能添加这些了 但是 arpl是可以添加的

wu0lei, 2023-11-01:
MAC2能配置成网卡实际的MAC地址可以吗？还是必须用洗白的？以前MAC2我没配置，是引导直接暴露的物理网卡MAC

KleinerSource, 2023-11-01:
> MAC2能配置成网卡实际的MAC地址可以吗？还是必须用洗白的？以前MAC2我没配置，是引导直接暴露的物理网卡MAC

ame直接激活必须是 全白的 sn和mac 否则去用脚本解锁

wu0lei, 2023-11-01:
我用过脚本，还是不行

KleinerSource, 2023-11-01:
> 我用过脚本，还是不行

那我就不知道了 我没用过脚本 我一直是全白

wu0lei, 2023-11-01:
mac2参数添加不进去，等待大佬修复

wjz304, 2023-11-01:
从 9.7 之后就没mac2了，所以你说 10.4 OK，10.9不OK时缺少mac2 的依据是什么？

wu0lei, 2023-11-01:
有没有一种可能，大家都是在9.7之前安装激活AME的，之后版本升级也没注意到，你说对吧？ 然后我换了RR之后碰巧打开AME看一下 发现激活不了，具体原因麻烦大佬帮忙看一下

KleinerSource, 2023-11-01:
> 从 9.7 之后就没mac2了，所以你说 10.4 OK，10.9不OK时缺少mac2 的依据是什么？

10.4 可以手动加进去 保存后 可以生效的  但是 10.9手动加进去不生效  
[image]

KleinerSource, 2023-11-01:
我是从 10.4升级上来的不知道是不是升级造成的问题. 我目前的解决办法就是 删掉一张网卡(虚拟机) 就行了

KleinerSource, 2023-11-01:
[image]
[image]
是 10.9的bug 改了新ui 导致添加不生效  10.8是正常的

KleinerSource, 2023-11-01:
[image]
手动添加配置文件 没问题 
[image]
双网卡添加 AME也正常了

azchoi, 2023-11-01:
In case of RR, I cannot get AME licence due to mac2 issue. (maybe...)
I had try to add white code mac2 address to cmdline, but Failed. 
Finally, rollback to 23.10.4..... do you have any solution for me?

wu0lei, 2023-11-01:
更新23.11.0修复了 感谢大佬

wjz304, 2023-11-05:
https://github.com/wjz304/rr/issues/322

wjz304, 2023-11-05:
用户 cmdline 优先级最高，
如果要设置1个网卡：mac2保持空 cmdline里加 netif_num=1 就好了

wjz304, 2023-11-12:
The problem has been resolved,

The issue is closed

closed 2023-11-12 by wjz304

TheAutomatic, 2024-06-28:
装sa6400 7.2，已经在cmdline里添加了参数netif_num和值为1，洗白mac填了第一个空着第二个，开机后还是有2个网卡，不太明白

wjz304, 2024-06-28:
> 装sa6400 7.2，已经在cmdline里添加了参数netif_num和值为1，洗白mac填了第一个空着第二个，开机后还是有2个网卡，不太明白

netif_num 和 mac 用于 洗白，你机器有几个网卡 就有几个网卡，咋还不明白？

TheAutomatic, 2024-06-28:
> > 装sa6400 7.2，已经在cmdline里添加了参数netif_num和值为1，洗白mac填了第一个空着第二个，开机后还是有2个网卡，不太明白
> 
> netif_num 和 mac 用于 洗白，你机器有几个网卡 就有几个网卡，咋还不明白？

好像明白了，就是说无论引导页面读到几个ethx、开机后有几个网卡，都不影响洗白

wjz304, 2024-06-28:
> > > 装sa6400 7.2，已经在cmdline里添加了参数netif_num和值为1，洗白mac填了第一个空着第二个，开机后还是有2个网卡，不太明白
> > 
> > 
> > netif_num 和 mac 用于 洗白，你机器有几个网卡 就有几个网卡，咋还不明白？
> 
> 好像明白了，就是说无论引导页面读到几个ethx、开机后有几个网卡，都不影响洗白

嗯，