RROrg/rr #316: 有没有可能支持型号EDS14

keenghost, 2023-11-01:
&emsp;&emsp;虚拟机非直通硬盘黑群最大的问题就是Cloud Sync没法选择NFS文件夹，另一条路就是挂载宿主机提供的目录，后来我找到个自带9p的镜像，虽然不会阻止选择9p挂载的目录，但文件改动Cloud Sync并不会响应，折中的做法就是通过设置计划任务来让Cloud Sync得以重新遍历文件夹，但这种做法有个最大的缺点就是本地已删除的目录(单向同步)不会把远端的也删了，只会增量或修改同步，设置双向又怕百度云突然抽风(Cloud Sync是我目前唯一能稳定向百度云上传超过20G单文件的工具，所以非常依赖Cloud Sync)。

&emsp;&emsp;最近看Cloud Sync文档时发现虽然所有机型的群晖都不支持选择NFS，但有一款古老的存在，群晖EDS14，它没有放硬盘的空间，所以它数据的来源就是网络硬盘，Cloud Sync为此特赦这个型号可以选择NFS(也不知道我理解的对不对，总觉得不厚道)。

* 不支持同步 Synology NAS 上的已装载和外部共享文件夹（不包括 EDS14）

&emsp;&emsp;Cloud Sync 技术规范: https://www.synology.cn/zh-cn/dsm/7.2/software_spec/cloud_sync

stkroot, 2023-11-01:
特赦，(*^_^*)

楼主是想单独虚拟一个 EDS14，来同步另一台群晖的 NFS 文件夹和百度云？

wjz304, 2023-11-01:
[image]
拉倒吧

wjz304, 2023-11-01:
先不是能不能黑，< 6.2.4的固件 官方都不提供下载了，这最终版才6.0.3

closed 2023-11-01 by wjz304