RROrg/rr #317: 天钡R1 N00无法安装黑群晖

cuitzeno, 2023-11-01:
Install ENV: (You can find it in the boot interface.)  
 * DMI:  TB WTR R1/WTR R1, BIOS HCN100-MI2 10/19/2023
 * CPU:  Intel(R) N100
 * NIC:  i226

RR version: (You can find it in the update menu.)  
 * RR:  v23.11.0
 * addons:  
 * modules:  
 * lkms:  dev / prod （都试过）

DSM: 
 * model:  DS920+
 * version:  7.2(69057)

Issue:  无法启动黑群晖

logs:  

(Plz review the content of #173, #175, #226 first)  
...

andy094, 2023-11-01:
装SA6400试试

cuitzeno, 2023-11-02:
> 装SA6400试试

试过了，也不行。。。

jim3ma, 2023-11-02:
安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？

cuitzeno, 2023-11-02:
> 安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？

全部都安装完了，启动群晖之后IP Ping不通，也找不到群晖

yangyi860, 2023-11-02:
上传pat文件后8%跳出无法格式化硬盘，我是在vmware workstation中安装。。

LD166, 2023-11-02:
我也是试了多好次了，编译完引导，起不来

jim3ma, 2023-11-02:
> > 安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？
> 
> 配置引导全部都弄完了，引导启动群晖之后IP Ping不通，也找不到群晖

连群晖的安装界面都没有进入么？路由器没有新 IP ？

LD166, 2023-11-02:
> > > 安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？
> > 
> > 
> > 配置引导全部都弄完了，引导启动群晖之后IP Ping不通，也找不到群晖
> 
> 连群晖的安装界面都没有进入么？路由器没有新 IP ？

是的连群晖的安装界面都没有进入，路由器后台没有黑群 IP。

jim3ma, 2023-11-02:
> > > > 安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？
> > > 
> > > 
> > > 配置引导全部都弄完了，引导启动群晖之后IP Ping不通，也找不到群晖
> > 
> > 
> > 连群晖的安装界面都没有进入么？路由器没有新 IP ？
> 
> 是的连群晖的安装界面都没有进入，路由器后台没有黑群 IP。

按操作，获取日志，https://github.com/wjz304/rr/issues/173

jim3ma, 2023-11-02:
> > > > > > 安装到哪一步了？找不到 IP，还是有 IP，只是上传 PAT 后安装失败？
> > > > > 
> > > > > 
> > > > > 配置引导全部都弄完了，引导启动群晖之后IP Ping不通，也找不到群晖
> > > > 
> > > > 
> > > > 连群晖的安装界面都没有进入么？路由器没有新 IP ？
> > > 
> > > 
> > > 是的连群晖的安装界面都没有进入，路由器后台没有黑群 IP。
> > 
> > 
> > 按操作，获取日志，#173
> 
> 是我启用arpl中的dbgutils插件方法不对么？我进入p1 里边没有 logs 文件。。。。

再看看机器有没有串口吧，串口也可以看日志

如果 dbgutils 插件确认启用了，没有 logs 日志的话，那就是 BIOS 或者主板不支持 SA6400 了，换其他机型或者使用虚拟黑群晖吧

LD166, 2023-11-03:
> 再看看机器有没有串口吧，串口也可以看日志
>
> 如果 dbgutils 插件确认启用了，没有 logs 日志的话，那就是 BIOS 或者主板不支持 SA6400 了，换其他机型或者使用虚拟黑群晖吧

> 再看看机器有没有串口吧，串口也可以看日志
>
> 如果 dbgutils 插件确认启用了，没有 logs 日志的话，那就是 BIOS 或者主板不支持 SA6400 了，换其他机型或者使用虚拟黑群晖吧

LD166, 2023-11-03:
> 再看看机器有没有串口吧，串口也可以看日志
>
> 如果 dbgutils 插件确认启用了，没有 logs 日志的话，那就是 BIOS 或者主板不支持 SA6400 了，换其他机型或者使用虚拟黑群晖吧

一共有两个问题：
1. 在arpl 里选择完型号和系统后，进入 高级设置 里把第一项改为 test 后编译系统，编译完进入启动选项，这时并不能顺利进入系统，需要重启后才选择u盘启动第一项才能进入系统。
2. 如果用两个盘的话，就会没ip地址。。
3.晚会我把log 发出来。

LD166, 2023-11-03:
[Uploading logs.zip…]()

LD166, 2023-11-03:
> 再看看机器有没有串口吧，串口也可以看日志
>
> 如果 dbgutils 插件确认启用了，没有 logs 日志的话，那就是 BIOS 或者主板不支持 SA6400 了，换其他机型或者使用虚拟黑群晖吧

[logs.zip](https://github.com/wjz304/rr/files/13246425/logs.zip)

jim3ma, 2023-11-03:
> [logs.zip](https://github.com/wjz304/rr/files/13246425/logs.zip)

你上传的日志，并不是 SA6400 的，是 918 的

LD166, 2023-11-03:
> 你上传的日志，并不是 SA6400 的，是 918 的

SA6400 根本启动不起来，我没法给你发日志啊。

LD166, 2023-11-03:
/

LD166, 2023-11-03:
> 你上传的日志，并不是 SA6400 的，是 918 的

测试了好多个型号，只有918+，改 test 选项才能勉强启动起来。
SA6400 改了也启动不起来，我没法给你发日志啊。

jim3ma, 2023-11-03:
没有日志的话，很难解了，靠猜是不行的

wsmayi, 2023-11-04:
我的畅网P5 N100目前在用23.9.0版本号的，选择SA6400+7.2默认配置可以正常用，其他型号都不行。后面试过编译rr版本好像也是不行，现在都不敢升级到最新的7.2.1版本，怕起不来了。

wukongdaily, 2023-11-04:
[logs.tar.gz](https://github.com/wjz304/rr/files/13256198/logs.tar.gz)
您好～以上附件是日志，机型和楼主一样，天钡N100。用u盘引导。机型无论选哪个都无法引导。如果有空可以帮忙看看哈～谢谢啦！🙏🙏🙏

LD166, 2023-11-04:
> [logs.tar.gz](https://github.com/wjz304/rr/files/13256198/logs.tar.gz) 您好～以上附件是日志，机型和楼主一样，天钡N100。用u盘引导。机型无论选哪个都无法引导。如果有空可以帮忙看看哈～谢谢啦！🙏🙏🙏 目前的现象是：已经出现ip和端口号5000，但用浏览器访问是不通的。等待10分钟也无法进入 群晖的安装界面。下面截取了log里的一小段片段。不知道和此处是否有关系 [ 0.630434] WARNING: CPU: 0 PID: 836 at /source/input/shim/block_fw_update_shim.c:53 register_fw_update_shim.cold+0x9f/0xa4 redpill [ 0.632543] BUG log triggered [ 0.633023] Modules linked in: redpill(OE+) [ 0.633784] CPU: 0 PID: 836 Comm: insmod Tainted: G OE 4.4.302+ #69057 [ 0.635087] Hardware name: TB WTR R1/WTR R1, BIOS HCN100-MI2 10/19/2023 [ 0.636224] 0000000000000000 ffff880274273c18 ffffffff8157dae1 ffff880274273c60 [ 0.637511] ffffffffa0017c60 ffff880274273c50 ffffffff8104e223 0000000000000000 [ 0.638794] ffffffff81a6e044 0000000000000006 0000000000000000 ffff880274273eb0 [ 0.640079] Call Trace: [ 0.640503] [] dump_stack+0x57/0x6d [ 0.641387] [] warn_slowpath_common+0x83/0xd0 [ 0.642426] [] warn_slowpath_fmt+0x49/0x50 [ 0.643420] [] register_fw_update_shim.cold+0x9f/0xa4 [redpill] [ 0.644726] [] ? 0xffffffffa0021000 [ 0.645613] [] init_+0x9e/0x11c [redpill] [ 0.646586] [] do_one_initcall+0x88/0x130 [ 0.647559] [] do_init_module+0x5b/0x200 [ 0.648518] [] load_module+0x1df4/0x2300 [ 0.649475] [] ? symbol_put_addr+0x30/0x30 [ 0.650467] [] ? kernel_read+0x3d/0x50 [ 0.651395] [] SYSC_finit_module+0x79/0x90 [ 0.652388] [] SyS_finit_module+0x9/0x10 [ 0.653352] [] entry_SYSCALL_64_fastpath+0x1e/0x93 [ 0.654463] ---[ end trace 5fb3b416c64439b1 ]--- [ 0.655264] PCI host bridge to bus 0001:01 [ 0.655979] pci_bus 0001:01: root bus resource [io 0x0000-0xffff] [ 0.657052] pci_bus 0001:01: root bus resource [mem 0x00000000-0x7fffffffff] [ 0.658269] pci_bus 0001:01: root bus resource [bus 00-ff] [ 0.659223] pci 0001:01:00.0: [1b4b:9215] type 00 class 0x010601 [ 0.660273] pci 0001:01:00.0: Can't map mv9235 registers

试下usb转千兆网口装 sa6400，有大佬已经装上了

wjz304, 2023-11-04:
https://github.com/wjz304/rr/issues/324

LD166, 2023-11-05:
> [logs.tar.gz](https://github.com/wjz304/rr/files/13256198/logs.tar.gz) 您好～以上附件是日志，机型和楼主一样，天钡N100。用u盘引导。机型无论选哪个都无法引导。如果有空可以帮忙看看哈～谢谢啦！🙏🙏🙏 目前的现象是：已经出现ip和端口号5000，但用浏览器访问是不通的。等待10分钟也无法进入 [...]

最新11.1 引导已经正常引导 918+了，但是不能在系统里点重启，我重启后，进不去系统了。需要强制关机，再开机选引导的第一个选项进。也就是每次遇到开机后进不去系统，或者显示器显示在引导的几个选项界面，需要手动选择引导的第一个选项。

LD166, 2023-11-05:
> 我的畅网P5 N100目前在用23.9.0版本号的，选择SA6400+7.2默认配置可以正常用，其他型号都不行。后面试过编译rr版本好像也是不行，现在都不敢升级到最新的7.2.1版本，怕起不来了。

你可以把主板上的串口接出来，然后用新的系统启动，如果跑不起来，把串口的打印信息，保存成文件，发上来，让大佬看下打印的信息。

BenBenPeng, 2023-11-05:
我是用的零克N100

单网口的好好的，换了双网口的版本，坏菜了 无论选择什么引导模式都是重启后就失恋了。

每个版本都试过了 真是想把机器都砸了 只要重启，就啥都没了！

BenBenPeng, 2023-11-05:
零克小主机N100
单网口各种引导都能用 双网口同样配置就是编译后一直重启，网卡ping上几秒钟就断了。

LD166, 2023-11-05:
> 零克小主机N100 单网口各种引导都能用 双网口同样配置就是编译后一直重启，网卡ping上几秒钟就断了。 重起几次以后就又进了7681 ，进不了安装界面。 没法说23.10以后得版本我都试了 都不行

有debug接口么？接出来把打印信息保存下，发上来。

LD166, 2023-11-05:
> 了好多次了，编译完引导，起不来，感觉是网卡驱动的问题。
#

> 没有日志的话，很难解了，靠猜是不行的

[image]
问下大佬机器上是2个226网卡，但是系统里查到还有2个 211，会影响启动么？

jim3ma, 2023-11-05:
> > 了好多次了，编译完引导，起不来，感觉是网卡驱动的问题。
> 
> > 没有日志的话，很难解了，靠猜是不行的
> 
> [image] 问下大佬机器上是2个226网卡，但是系统里查到还有2个 211，会影响启动么？

一般情况没有啥影响，网卡多而已，4 个网卡都不能正常使用么？

BenBenPeng, 2023-11-05:
23.11.1 选SA6400 全部默认 直接引导起来了。。。

本着破罐子破摔的精神 神奇的引导起来了

wjz304, 2023-11-05:
https://github.com/wjz304/rr/issues/327

LD166, 2023-11-05:
> > > 了好多次了，编译完引导，起不来，感觉是网卡驱动的问题。
> > 
> > 
> > > 没有日志的话，很难解了，靠猜是不行的
> > 
> > 
> > [image] 问下大佬机器上是2个226网卡，但是系统里查到还有2个 211，会影响启动么？
> 
> 一般情况没有啥影响，网卡多而已，4 个网卡都不能正常使用么？

2个网口用的226，211不知道哪里出来的。

BenBenPeng, 2023-11-07:
理论上不可能，什么硬件专门不让装物理黑裙 我觉得还是引导可以解决的

我买过之前的版本天呗  amd也用的好好的

LD166, 2023-11-12:
> [第324章](https://github.com/wjz304/rr/issues/324)

最新的11.4 引导，能进系统安装界面了，但是装不上系统，10分钟后，点再次连接，又开始10分钟倒计时了。这时候路由器后台有设备ip ，但是就是装不好系统。
[image]
[image]

LD166, 2023-11-13:
> telnet 到机器上，看内核日志

进来了，需要哪些文件？

LD166, 2023-11-13:
> telnet 到机器上，看内核日志

进来了。是需要这个 /var/log 下的文件么？
[image]

LD166, 2023-11-13:
> telnet 到机器上，看内核日志

[image]
进来了。是需要这个 /var/log 下的文件么？

jim3ma, 2023-11-13:
再次安装，等失败后，/var/log 目录日志上传
另外就是执行 poweroff，看看能不能正常关机

LD166, 2023-11-13:
> 再次安装，等失败后，/var/log 目录日志上传 另外就是执行 poweroff，看看能不能正常关机

poweroff 命令关机不了 ，怎么重新安装，是在5000 端口页面点再次连接，还是在u盘引导页面，选择重新安装DSM？下边是messages 文件的截图。

jim3ma, 2023-11-13:
poweroff 不能关机是其他主板已知问题，去 BIOS 里关闭串口有关的功能应该就好了

LD166, 2023-11-13:
> 去 BIOS 里关闭串口有关的功能应该就好了

意思是主板里那个super io 里的配置给关闭了么？如果主板的debug口不是uart，而是spi 会有影响么？

jim3ma, 2023-11-13:
> > 去 BIOS 里关闭串口有关的功能应该就好了
> 
> 意思是主板里那个super io 里的配置给关闭了么？如果主板的debug口不是uart，而是spi 会有影响么？

挨个关闭试试，常见的是串口

LD166, 2023-11-13:
> 常见的是串口
这块主板都没有做串口。。。。。。

wjz304, 2023-11-13:
看不见不代表没有

LD166, 2023-11-13:
> 看不见不代表没有

> > > 去 BIOS 里关闭串口有关的功能应该就好了
> > 
> > 
> > 意思是主板里那个super io 里的配置给关闭了么？如果主板的debug口不是uart，而是spi 会有影响么？
> 
> 挨个关闭试试，常见的是串口

谢谢大佬，确实是串口问题，硬件上r1 群主说了确实没有串口，用他给的新bios ，关闭一些端口，确实好了。

LD166, 2023-11-14:
R1 N100 已经正常引导并使用了，不能正常使用的罪魁祸首是 BIOS 里没有串口配置选项，目前已经正常使用，不存在重启问题了。

zhangdayu16, 2023-11-15:
> R1 N100 已经正常引导并使用了，不能正常使用的罪魁祸首是 BIOS 里没有串口配置选项，目前已经正常使用，不存在重启问题了。

请问用的那个群晖版本，我之前23.11.02可以引导进去系统，今天想升级下11.6可以硬解，但试了918和sa6400，但编译完：5000进不到设置页面。

LD166, 2023-11-15:
> > R1 N100 已经正常引导并使用了，不能正常使用的罪魁祸首是 BIOS 里没有串口配置选项，目前已经正常使用，不存在重启问题了。
> 
> 请问用的那个群晖版本，我之前23.11.02可以引导进去系统，今天想升级下11.6可以硬解，但试了918和sa6400，但编译完：5000进不到设置页面。

你需要新bios，等老板发新的bios

CarlosLucky, 2023-11-16:
畅网 n100 P5 还可以装ds3622，运行正常，启动也很快。

dreamfly-jiefa, 2023-11-18:
> 我的畅网P5 N100目前在用23.9.0版本号的，选择SA6400+7.2默认配置可以正常用，其他型号都不行。后面试过编译rr版本好像也是不行，现在都不敢升级到最新的7.2.1版本，怕起不来了。

装好之后，用SHR组存储池一直在卡着，都跑了2天了，还没有后台优化完成，你会出现这个情况吗？

LD166, 2023-11-20:
你们的 SA6400 系统硬盘能休眠么？我在系统里选择硬盘休眠时间，但是不起作用，是不是 SA6400 系统本来不允许硬盘休眠？

dreamfly-jiefa, 2023-11-20:
> 我的畅网P5 N100目前在用23.9.0版本号的，选择SA6400+7.2默认配置可以正常用，其他型号都不行。后面试过编译rr版本好像也是不行，现在都不敢升级到最新的7.2.1版本，怕起不来了。

你是直接装在物理机上，还是装在虚拟化平台上的？

LD166, 2023-11-22:
系统里的定时开关机功能只有关机可用，开机不能用是咋回事？我把 bios 时间和系统时间改一致了，也还是不行，只能关机不能开机。

closed 2023-11-23 by cuitzeno

LD166, 2023-11-23:
早知道了，问题还是我发现的。