RROrg/rr #323: 重置密码能否加个小功能？关闭两步验证，我测试过重置密码功能，只改了密码，两步验证没动

stkroot, 2023-11-03:
需要修改 method.config 文件，把 status=on 改为 off
路径 /usr/syno/etc/packages/SecureSignIn/preference/账号名称

具体可参考：
https://wp.gxnas.com/13689.html

wjz304, 2023-11-03:
赞👍

wjz304, 2023-11-04:
https://github.com/wjz304/rr/releases/tag/23.11.1

wjz304, 2023-11-04:
试试好不好使，因为 我没有洗白的环境开不了验证，我也一直都在不停的格盘安装。
不好使反馈下

stkroot, 2023-11-05:
不需要洗白，不要用官方的登录器，用第三方的还好用一些。
我用的 bitwarden 和 AuthenticatorPro，都支持两步验证TOTP验证码

stkroot, 2023-11-05:
> 试试好不好使，因为 我没有洗白的环境开不了验证，我也一直都在不停的格盘安装。 不好使反馈下

测试了，不行，重置密码功能都失效了，没改成功。

stkroot, 2023-11-05:
我以前测试的修改密码，是在 arpl-i18n 下，rr没试过。所以不确定rr之前重置密码是否有效。

其实可以考虑把重置密码和关闭验证登录功能分开

wjz304, 2023-11-05:
一开始确实考虑分开，但是吧，两个流程 98%是一样的

wjz304, 2023-11-05:
好吧，buildroot升级了python的版本，导致 一些依赖库报错了

stkroot, 2023-11-05:
是的，流程基本一样，不应该有问题。所以你改好了后，我都没想过再测试一下😀

wjz304, 2023-11-06:
https://github.com/wjz304/rr/releases/tag/23.11.2
我试了

closed 2023-11-06 by wjz304